MarlinFirmware/Marlin #7807: Malyan M150 blank screen after flashing example configuration

nunofgs, 2017-10-01:
### Bug Report
- Description: Flashed the configuration from the `examples/Malyan/M150` folder in the 1.1.x branch.
- Expected behaviour: Printer shows blank screen on boot.
- Actual behaviour: The printer does not display anything on the LCD screen.
- Steps to reproduce:
  - Just flashed with: `avrdude -v -p atmega1284p -c arduino -P /dev/cu.wchusbserial1410 -b 57600 -V -U flash:w:myfile.hex`

> Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

I'm using the examples from https://github.com/MarlinFirmware/Marlin/tree/1.1.x/Marlin/example_configurations/Malyan/M150

@thinkyhead are you able to flash the example config on a Malyan M150?

basdelfos, 2017-10-20:
I have the same problem with building and flashing the source from release 1.1.6.
How can I solve the black screen?

basdelfos, 2017-11-07:
I have tried building 1.1.0RC8 and with that version the screen works fine.

CutterSlade001, 2017-11-18:
Same issue here also tried on 1.1.5. The newest version I have that works is the 1.1.3 from https://www.thingiverse.com/thing:2376302  .

thinkyhead, 2017-11-18:
@CutterSlade001 Please post your Configurations in a reply, either as a ZIP or by adding ".txt" to the filenames.

CutterSlade001, 2017-11-20:
This is modified for autoleveling (compared to the Thingiverse version) but it doesn't autolevel properly which is why I'd like to try 1.1.6 to see if autoleveling works there.

[Malyan-M150-firmware-Marlin-1.1.3.zip](https://github.com/MarlinFirmware/Marlin/files/1489007/Malyan-M150-firmware-Marlin-1.1.3.zip)

s0ftcorn, 2017-12-07:
Same here, i also looked at changes others made to ultralcd_st7920_u8glib_rrd.h and pins_SANGUINOLOLU_11.h but the issue persists.
Attached is my config.

[Malyan-M150-Marlin-1.1.6.zip](https://github.com/MarlinFirmware/Marlin/files/1537626/Malyan-M150-Marlin-1.1.6.zip)

ocarlier, 2017-12-20:
Howdy, I'm pretty confident your problem exists in 'pins_SANGUINOLOLU_11.h'. 
I'm about to publish 1.1.7 binaries on Thingiverse, they are working fine.

closed 2017-12-20 by thinkyhead

CutterSlade001, 2017-12-20:
@ocarlier Sadly I am unable to compile your 1.1.7 soucre, it says:  HEATER_0_PIN not defined for this board.

basdelfos, 2017-12-21:
@ocarlier Same problem here, not able to compile your 1.1.7 source (HEATER_0_PIN not defined).
Did some research and found that some files in the `marlin-firmware.zip` from  thingiverse are corrupted. They are empty. One of the files is `pins_MELZI.h` and `pins_SANGUINOLOLU_11.h`.

Could you upload a new zip and inform us.

Thanks.

ocarlier, 2017-12-22:
Sorry about that! The virtual machine I was using crashed and corrupted more of the files than I was aware. I've restored the source files, built and tested both 1.1.3 and 1.1.7.

[Download from Thingiverse](https://www.thingiverse.com/thing:2376302/zip)