MarlinFirmware/Marlin #7808: Bug Report re: auto bed leveling

HappySock, 2017-10-01:
### Bug Report
- Description: --- unable to properly level the bed using the autoleveling feature on a 'CR-10s S5'
- Expected behaviour: --- I expect the head to move to the four corners then end in the middle.
- Actual behavior: --- head moves through the proper motions but **only within the quadrant closest to the home position**. It seems to think that there is only a small 250mm2 bed instead of the 500mm2 bed
- Steps to reproduce:
  - run auto leveling feature, step through process

Tannoo, 2017-10-01:
Properly configure your bed size then try again.

HappySock, 2017-10-02:
I am going to get with Sainsmart on this as they are shipping it this way!

closed 2017-10-02 by HappySock