MarlinFirmware/Marlin #7810: [FR] configure bed size

HappySock, 2017-10-01:
I ran into an issue today with a bad gcode that was included with my creality CR-10 S5.

The gcode specified a Y position way outside the bed area. This led to two problems.

1. the printer reached the position limit and attempted to go beyond it due to the bad gcode. It skipped quite a few teeth on the belt before it thought it had reached the intended position. upon proceeding to the next specified location, it no longer knew where it was and began printing in mid-air, away from where it should have been.
2. skipping teeth is really a bad thing regardless of the cause, and it can be avoided.
3. the print failed where, had the firmware realized that the specified location was out of range and simply skipped the command and proceeded to the next location, the print would have finished fine with  hardly an issue. It would not have lost its position.

**So, please add a feature where we can specify the bed area in the code and have the code skip a command that would take you outside of the bed area.**

fiveangle, 2017-10-01:
it already exists, but you need to configure correctly to honor the limits (*_BED_SIZE, probing limits, safe homing, software endstops, etc, etc, etc).  Start looking through your config around those areas, google to find some examples of how they work, and once you have researched to where you think you understand most of it, come back here if you still have questions and we can help fill in the missing bits.

fiveangle, 2017-10-01:
Also, check the CR-10 configuration recently uplaoded to compare.  I'm not saying the example is the end-all, be-all, but good to compare and investigate the differences so you might understand how you want to configure yours.

HappySock, 2017-10-02:
all this is beyond my current abilities. I have a stock board with no bootloader on it so far. I was told by sainsmart that it had the latest marlin firmware. that is all I know. so I reported my issues here.

Tannoo, 2017-10-03:
You could download a new copy from here, but you need your printer info to set it up correctly.
Ask them for the source code.

closed 2017-11-15 by thinkyhead