MarlinFirmware/Marlin #7813: bugfix-2.0.x Anet SD/Fan Issues

Phr3d13, 2017-10-02:
### Bug Report
- Description: SD Card Won't mount
- Expected behaviour: Insert SD Card, select Init SD Card, SD is read, profit
- Actual behaviour: Insert SD, select Init SD Card, Part cooling fan spin, SD not mounted.
- Steps to reproduce:
  - Compile/Upload bugfix-2.0.x for Anet board using example config
  - Try to init SD Card
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1347675/Marlin.zip)

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

Phr3d13, 2017-10-04:
I'll add that the SD works in 1.1.5 and bugfix-1.1.x

Phr3d13, 2017-10-09:
Got rid of the serial doubling so I could finally send commands to the printer, and this is still an issue, I've looked through SD related files, ANET specific stuff, and SPI related stuff, and I still can't figure out what is broken. SD card works fine with 1.1.5, bugfix-1.1.x, etc, just not bugfix-2.0.x.

MasterPIC, 2017-10-09:
I confirm the issue is limited to bugfix-2.0.x
I didn't check the code. Anyway since cooling fan is not related to SD, maybe something related to pin assignment is wrong...

Phr3d13, 2017-10-09:
The part cooling fan has always had issues, even with stock firmware. Scrolling to the bottom of a long list of files will make the fan spin. I'm not sure we can use it as a clue.

Phr3d13, 2017-10-11:
I think it has something to do with either that the sd is on the printer board rather than the lcd, or that the anet board requires a board added to arduino

Phr3d13, 2017-10-14:
Ok, comparing pins_ANET.h from Skynet to what we currently have, I see two differences that jump out at me. from Skynet pins_ANET.h
```
#define ST7920_DELAY_1 DELAY_0_NOP
#define ST7920_DELAY_2 DELAY_1_NOP
#define ST7920_DELAY_3 DELAY_2_NOP```
these 3 lines aren't in ours. and 
``` #define ADC_KEYPAD_PIN      1```
is in ours, but not the skynet version, I'll test as soon as I get a chance to, but does anyone know why these are different? Also, would either of these differences account for SD issues?

Phr3d13, 2017-10-14:
@Bob-the-Kuhn got any ideas?

Phr3d13, 2017-10-16:
@thinkyhead Do you have any ideas why SD won't work with bugfix-2.0.x on an anet board?

Phr3d13, 2017-11-03:
Is there anything else I can do to get more information to someone to help?

Phr3d13, 2017-11-11:
Tried a build of bugfix-2.0.x from today (11/11/2017) and this issue is still happening.

RalphSch, 2017-11-15:
The problem with the Anet 1.0x controller board and the lcd/fan/sd issue is probably because of the wiring/routing of the FAN signal and the SPI bus. The FAN signal goes from the Atmega (pin 44) to the mosfet and also to connector J3 (pin 9) building an antenna. The SPI bus (Atmega pin 1,2 and 3) is also splittet up to the onboard sd-card reader and J3. I think that this situation with parallel routed and open ended wires, with high frequency signals running through them, is generating a lot of noise and interference which are causing the problems. If you are using the A6 lcd display you are extending all those 4 signal lines from J3 (miso/mosi/sck and fan) by the length of the ribbon cable (all four lines are then open ended on the 12864 lcd pcb) which changes the behavior, you can easily get data corruption on the A6 lcd if you are routing/mounting the ribbon cables in a wrong way. That's probably the point were the ST7920_DELAY came in (st7920 is the lcd controller of the A6 lcd display). I also used these ST7920 delays to get the A6 display working with Marlin 2.0 on RADDS/Due.

closed 2018-02-09 by thinkyhead

wazzzap, 2018-02-19:
For me its not working on Marlin v1.1.8. When i try to init SD card - fan starts to spin.

thinkyhead, 2018-02-19:
> For me its not working on Marlin v1.1.8. When i try to init SD card - fan starts to spin.

You should consider posting a new issue about that, with more details. But only if you find that the problem still exists in `bugfix-1.1.x`.