MarlinFirmware/Marlin #7815: Hbot print is tilted 45 degrees

EricKabir, 2017-10-02:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

EricKabir, 2017-10-02:
I am not sure about the issue. I doubt if it might be posted here or no!
The Hbot is printing quite well. But XY keep shifting each and every layer. At the end I got 45 degree tilted printed 10X10 cube. 
Any help would be appreciated.

Tannoo, 2017-10-03:
So, you end up with some type of trapizoid instead of a cube?
Fyi, this also not an [FR].

EricKabir, 2017-10-03:
Yes. Exactely. The problem is, I changed the XY axis, the same result. Only trapezoid axis changes. Sort of parallelogram/rhombus.  I tried the Repetier FW,working fine. Using external drivers with MKS m/b. Not sure about the equations. Maybe some modifications might help. 
Regarding [FR], my fault. I came to know just now.

Tannoo, 2017-10-03:
Pictures?

EricKabir, 2017-10-03:
XY axis both behave same. I use normal corexy XY design, single belt H-Bot. Then I swap the XY to YX. same behaviour. Little bit fuzzy, I switch to Repetier, it sounds ok, prints almost 10X10X10 (not precise but sure cube!) . Switch back to Marlin FW,  flip all motor wires, inverting directions, all the endstops, still the same problem. 
[1.zip](https://github.com/MarlinFirmware/Marlin/files/1351457/1.zip)

Tannoo, 2017-10-03:
Why are you flipping motor wires?

EricKabir, 2017-10-03:
Just to be sure XY swap will behave same.

EricKabir, 2017-10-03:
But same mechanics and wiring will be ok with repetier.

EricKabir, 2017-10-03:
As per issue : Support for reverse x-axis direction of H-bot is needed.  #2587 ,
I use setup#2 in hbot design. Following [https://github.com/MarlinFirmware/MarlinDev/pull/55] , no hope.
In Repetier configuration tool I do select as attach. 
[3.zip](https://github.com/MarlinFirmware/Marlin/files/1351569/3.zip)

Tannoo, 2017-10-03:
Have you tried bugfix-1.1.x?

EricKabir, 2017-10-03:
Yes. I did. Now I double checked again. Same problem exist.

Tannoo, 2017-10-03:
Attach your sliced file for me to test. I don't have a coreXY machine though.

Tannoo, 2017-10-03:
I will first test with bugfix-2.0.x. If that fails, I will create and slice my own cube.

EricKabir, 2017-10-03:
Same file also I print with Repetier. Sounds ok. Also I tried several engines. 
Thanks. I appreciate. 
[test 10x10x10.zip](https://github.com/MarlinFirmware/Marlin/files/1351976/test.10x10x10.zip)

p3p, 2017-10-03:
I have no experience with corexy designs, but in general if there are skipped steps and offset layers I would tell you to lower the jerk setting by half and try again (can tune it later) same for acceleration, hardware wise the current setting on the stepper drivers or a "sticky" belt can cause the same effect but if Repetier FW works then its more likely the config.

EricKabir, 2017-10-04:
Thanks for suggestion. I even play around with jerk and acceleration. All ended up the same results. Meanwhile the Repetier FW works well with the same structure.
While Googling, I came across the doubled steps per mm for Repetier H-Bot calculation. 
The way they calculate the hbot movements are a bit different from Marlin.
Anyhow I still doubt about my hbot equation calculation as mentioned earlier by others, the issue # 2587 with reverse mechanism. 
Previously I used several Corexy machines under Marlin. Work like charm.

EricKabir, 2017-10-11:
not sure. consider a bug.

closed 2017-10-11 by EricKabir