MarlinFirmware/Marlin #7817: Z Axis smashing / stuttering issue.

eseglem, 2017-10-02:
I have been trying to debug this issue for most of the last day, but I am out of ideas, and hope someone might be able to help me out. I know its more of a config issue, and not a true bug, but couldn't find anywhere else to ask. 

I have a Lulzbot Taz 5 I want to update to a more recent version of Marlin, so I was attempting to migrate their settings over. When using the 1.1.x branch of Marlin, the Z axis ends up in a state where it stutters, and does not travel as much as it should. It seems to start off OK, but at some point fairly quickly fails to move as far as it should up and end up with a smashed print. The last calibration cube I tried started fine for 3mm, but was short by ~1.5mm over the ~12mm it should have printed. Whenever I stop the print and manually move it, it stutters and acts weird. But then a bit later I can move it again and it acts fine. I am guessing a setting is causing some sort of heat issue? The motors themselves are not hot, but I did not crack open the case to see about the drivers themselves yet. 

I know the steps are correct, on both the Z and extruder, I have verified them over and over again. And it would be more consistent if it were a step issue. I also know its not a complete hardware issue, as every time I put the default firmware back on it it prints fine. It has to be something in the translation of old settings to new settings.

It appears they based their last firmware on 1.0.0 and settings have changed a lot since then, but I did my best. I have attached my config files, as well as the working older ones, hopefully its something simple that I missed. 

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/1347984/Configs.zip)

fiveangle, 2017-10-02:
Looks like possible HW issue with your controller/steppers overheating due to case fan not running because USE_CONTROLLER_FAN is not enabled (old left, new right):

[image]

Roxy-3D, 2017-10-02:
The first thing I would do is change the max feed rate for the Z Axis down to a small number.  That way, if something bad happens, I have time to press the Reset button.

eseglem, 2017-10-02:
@fiveangle Thank you so much, I am going to bet that is the cause of the issue. Updating firmware now, and will update in a bit with hopefully a successful print. Its funny, I had thought about the fan last night when I noticed it was off. I just figured it had been more than 60 seconds, as I knew I had set the speed. I did not realize that it was in an if statement in the newer settings.

closed 2017-10-02 by eseglem