MarlinFirmware/Marlin #7822: UBL Fine Tuning “P4”

empakoso, 2017-10-02:
### Bug Report
- Description: ---
Executing a fine tune operation at the nearest point to the nozzle.
Command: G29 P4 H0.09 R9
Command breakdown:
P4: phase 4-fine tune the closest mesh point to the nozzles (X,Y) position.
H0.09: height of the shim/feeler gauge being used to feel the gap between the nozzle and bed (this value is subtracted from the encoder height “felt”, and saved as the mesh point height)
R9: 9 different points to measure around the first point 
- Expected behaviour: ---
Nozzle moves to the nearest probe point.
Nozzle moves done to a height of 0.09
I place my 0.09mm shim under the nozzle
I rotate my encoder ‘clockwise’ to lower it
Click the knob for the second point
I can now rotate my encoder ‘counter clockwise’
(the way it’s set up in my firmware)
- Actual behaviour: ---
Expect the encoder to be rotated in one direction every time.
- Steps to reproduce:
  - Do this
Fine tune the mesh
  - Do that
After every execution of the G29 P4 H0.09 R9 the first point need to have the encoder turned the opposite direction.

I do have my encoder configured to turn in reverse.
Clockwise is up the menu list
Counterclockwise is down the menu list

Roxy-3D, 2017-10-02:
I'm trying to understand the issue.   Is the entire problem you are seeing that the first point being edited responds in the opposite direction, but the rest of the points being edited behave correctly?

empakoso, 2017-10-02:
You nailed it. The first point the encoder turns in the opposite direction.  The rest of the 8 points turn in the configured direction.

closed 2017-10-20 by empakoso

Roxy-3D, 2017-10-20:
I just re-wrote the G29 P1 U option.   I'll take a 2nd look at the G29 P4 R stuff soon...  We need G26 to work on 32-bit and when that happens, I'll be editing the mesh points on 32-bit soon.

empakoso, 2017-10-20:
No problem.  I figured there was enough on your plate that this little cosmetic issue can be dealt with at an appropriate time.  Till then I didn’t want this to become stagnant. Thank you.

Roxy-3D, 2017-10-20:
> I'm trying to understand the issue. Is the entire problem you are seeing that the first point being edited responds in the opposite direction, but the rest of the points being edited behave correctly?

One question...   Can you include your Configuration.h and Configuration_adv.h files?   I'm thinking this only happens under certain settings for the encoder direction.   When I get to the G29 P4 R stuff, I'll run with you encoder wheel directions just to be sure I can duplicate the problem.

empakoso, 2017-10-20:
You’re timing is excellent, I just loaded 1.1.6 bug fix yesterday and will provide the config files shortly. Thanks

empakoso, 2017-10-21:
OK, loaded the latest 1.1.6 bugfix.  I can no longer reproduce the problem using the exact same encoder reversed settings.  We can forget this non-problem now.

Roxy-3D, 2017-10-21:
OK!   Thanks for the update!