MarlinFirmware/Marlin #7825: Reboot loop

kruuth, 2017-10-02:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---  Anet A8 motherboard reboots every 5 seconds.  System otherwise seems to be working fine.  However, during uploading firmware system does not reboot.
- Expected behaviour: ---  Marlin doesn't reboot
- Actual behaviour: ---  Marlin reboots every 5 seconds
- Steps to reproduce:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1350656/Marlin.zip)

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

Tannoo, 2017-10-03:
It you thinks it's firmware related, diable all options except for MB, thermo and movement settings, and see what happens.

kruuth, 2017-10-03:
Ok.  Flashed with nothing but the MB, thermo, and movement stuff.  Getting same loop.  I'm guessing this is a HW issue.  Again.

Tannoo, 2017-10-03:
Look for shorts on the 5v stuff... that can overload the regulator causing reboots.

Tannoo, 2017-10-03:
Unplug all limits and see if that's it.

kruuth, 2017-10-03:
I've unplugged everything and it's still rebooting.  I'm guessing it's the board.  Again.

Tannoo, 2017-10-03:
Well... that sux.

Tannoo, 2017-10-04:
Can you remove the drivers?

Phr3d13, 2017-10-09:
@Tannoo Built-in drivers. @kruuth I can confirm that bugfix-1.1.x works on an anet board.

closed 2017-11-14 by thinkyhead