MarlinFirmware/Marlin #7833: FeedRate can't be finely tuned by LCD (44780 - Anet A8) (on Marlin bugfix 1.1.x)

MasterPIC, 2017-10-03:
### Bug Report
- Description: by LCD (44780 on Anet A8) I'm no more able to **finely** tune FeedRate.
- Expected behaviour: FeedRate should increment/decrement by **1 unit at time**.
- Actual behaviour: FeedRate increments/decrements by **3-5 units at time**.

Roxy-3D, 2017-10-04:
You probably have the number of encoder steps per click set pretty high.

Phr3d13, 2017-10-06:
Where does one find "encoder steps per click" defined?

AnHardt, 2017-10-06:
Maybe
https://github.com/MarlinFirmware/Marlin/blob/9c00cff89836706776f6353635368a47cc5bf881/Marlin/example_configurations/Anet/A8/Configuration.h#L1258
in combination with
https://github.com/MarlinFirmware/Marlin/blob/e68b7afc52792ef7e97dbefc19c82d5b4902744e/Marlin/pins_ANET_10.h#L188

Phr3d13, 2017-10-06:
So what combination will net me with a 1% or 1 degree change with each button press?

Phr3d13, 2017-10-06:
Also, that copy/paste is from the full graphic (Anet A6 Display) portion of the Anet pins.h

Phr3d13, 2017-10-06:
I have the 5 button lcd, with no rotary encoder

fiveangle, 2017-10-06:
how the heck do we know ?  experiment and find out :)

I would try "1" for starters *shrug*

Phr3d13, 2017-10-06:
```/**
 * LCD / Controller
 *
 * Only the following displays are supported:
 *  ANET_KEYPAD_LCD
 *  ANET_FULL_GRAPHICS_LCD
 *  REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
*/

#if ENABLED(ULTRA_LCD) && ENABLED(NEWPANEL)
  #define LCD_SDSS           28
  #if ENABLED(ADC_KEYPAD)
    #define SERVO0_PIN         27 // free for BLTouch/3D-Touch
    #define LCD_PINS_RS        28
    #define LCD_PINS_ENABLE    29
    #define LCD_PINS_D4        10
    #define LCD_PINS_D5        11
    #define LCD_PINS_D6        16
    #define LCD_PINS_D7        17
    #define BTN_EN1            -1
    #define BTN_EN2            -1
    #define BTN_ENC            -1
    #define ADC_KEYPAD_PIN      1
    #define ENCODER_FEEDRATE_DEADZONE 2
  #elif ENABLED(REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER) || ENABLED(ANET_FULL_GRAPHICS_LCD)
    // Pin definitions for the Anet A6 Full Graphics display and the RepRapDiscount Full Graphics
    // display using an adapter board  // https://go.aisler.net/benlye/anet-lcd-adapter/pcb
    // See below for alternative pin definitions for use with https://www.thingiverse.com/thing:2103748
    #define SERVO0_PIN         29 // free for BLTouch/3D-Touch
    #define BEEPER_PIN         17
    #define LCD_PINS_RS        27
    #define LCD_PINS_ENABLE    28
    #define LCD_PINS_D4        30
    #define BTN_EN1            11
    #define BTN_EN2            10
    #define BTN_ENC            16
    #define ST7920_DELAY_1 DELAY_0_NOP
    #define ST7920_DELAY_2 DELAY_1_NOP
    #define ST7920_DELAY_3 DELAY_2_NOP
    #define STD_ENCODER_PULSES_PER_STEP 1
    #define STD_ENCODER_STEPS_PER_MENU_ITEM 1
  #endif
#endif  // ULTRA_LCD && NEWPANEL
```
Looking at this from pins_ANET_10.h, the top section is for the ANET keypad LCD, and the bottom is for the two Graphic LCDs.

I had to uncomment both of these and set them to 1:
```
//
// ENCODER SETTINGS
//
// This option overrides the default number of encoder pulses needed to
// produce one step. Should be increased for high-resolution encoders.
//
#define ENCODER_PULSES_PER_STEP 1

//
// Use this option to override the number of step signals required to
// move between next/prev menu items.
//
#define ENCODER_STEPS_PER_MENU_ITEM 1
```
Can I put both of those defines in the top section of the pins file:
```
#if ENABLED(ADC_KEYPAD)
    #define SERVO0_PIN         27 // free for BLTouch/3D-Touch
    #define LCD_PINS_RS        28
    #define LCD_PINS_ENABLE    29
    #define LCD_PINS_D4        10
    #define LCD_PINS_D5        11
    #define LCD_PINS_D6        16
    #define LCD_PINS_D7        17
    #define BTN_EN1            -1
    #define BTN_EN2            -1
    #define BTN_ENC            -1
    #define ADC_KEYPAD_PIN      1
    #define ENCODER_FEEDRATE_DEADZONE 2
    #define ENCODER_PULSES_PER_STEP 1
    #define ENCODER_STEPS_PER_MENU_ITEM 1
```

benlye, 2017-10-06:
If you have `ANET_KEYPAD_LCD` enabled, which you should, then there are already values [configured in Conditionals_lcd.h](https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/Conditionals_LCD.h#L49):

```
  #elif ENABLED(ANET_KEYPAD_LCD)

    #define REPRAPWORLD_KEYPAD
    #define REPRAPWORLD_KEYPAD_MOVE_STEP 10.0
    #define ADC_KEYPAD
    #define ADC_KEY_NUM 8
    #define ULTIPANEL
    // this helps to implement ADC_KEYPAD menus
    #define ENCODER_STEPS_PER_MENU_ITEM 1
    #define REVERSE_MENU_DIRECTION
```

Perhaps `#define ENCODER_PULSES_PER_STEP 1` ought to go there as well?  If I get a chance I'll resurrect my Anet board and 5-button keypad and play around with it.

Phr3d13, 2017-10-06:
Ok, I'll look in there, test out that define in that location, and if it works, make a PR.

Phr3d13, 2017-10-09:
@MasterPIC Can this be closed now?

MasterPIC, 2017-10-09:
Unfortunately I can't check. I also own an Anet A8  with 5 buttons lcd and no rotary encoder. So did you find it working?

Phr3d13, 2017-10-09:
It should, but could you scoop up the latest bugfix-1.1.x and verify that one button press equals an increment of 1

MasterPIC, 2017-10-09:
OK, I just checked and it works as expected.

closed 2017-10-09 by MasterPIC