MarlinFirmware/Marlin #7835: Need help setting probe points for UBL

totalitarian, 2017-10-03:
I have a 300 x 300 bed, my probe has offsets of

X -29 
Y -51  

I need around 10mm clearance around the edges of the bed to avoid my binder clips. Due to the position of the probe, I can reach the front and left edges of the bed but not the rear and right edges.

What values do I need to set for the following and how do I calculate them?

  
  [code]
  #define UBL_MESH_INSET 30          // Mesh inset margin on print area
  #define GRID_MAX_POINTS_X 15      // Don't use more than 15 points per axis, implementation limited.
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X
  
  #define UBL_PROBE_PT_1_X 50       // Probing points for 3-Point leveling of the mesh
  #define UBL_PROBE_PT_1_Y 220
  #define UBL_PROBE_PT_2_X 50
  #define UBL_PROBE_PT_2_Y 90
  #define UBL_PROBE_PT_3_X 240
  #define UBL_PROBE_PT_3_Y 90
[/code]

thinkyhead, 2017-11-15:
Did you figure it out?

closed 2017-11-15 by thinkyhead

ferrocam, 2018-03-26:
No I need points for a 220 by 220 bed with 10mm all around probe is -33 offset

thinkyhead, 2018-03-28:
With the current Marlin ("nightly" bugfix-x.x.x branches) there are simpler options. Just set your probe offsets, then set `MIN_PROBE_EDGE` to 10 to avoid the binder clips. Everything else will be calculated automatically.

Roxy-3D, 2018-04-19:
MIN_PROBE_EDGE should behave the same across all bed leveling systems.   It should restrict the nozzle's movement such that it stays that far away from the edge of the bed on all sides and for all bed leveling systems.

(If it doesn't...   We need to know so we can fix it.)