MarlinFirmware/Marlin #7836: Confused about UBL

totalitarian, 2017-10-03:
Hi,

I have read the excellent documentation on UBL and created mesh, done a test print and everything looks perfect!

My mesh is now loaded but this is where I get confused. Should UBL do a test probe in 3 places to verify the mesh tilt? I have added G28 to my startup script but it just homes all axis and goes on to print.

Am I missing something? It's nice having a topology of my bed but not very practical if it can't tilt this mesh if my bed gets knocked slightly. 

Apologies if I have missed something....

Tannoo, 2017-10-04:
`G29 Jn`  (n = number of side points)

Roxy-3D, 2017-10-04:
> My mesh is now loaded but this is where I get confused. Should UBL do a test probe in 3 places to verify the mesh tilt? I have added G28 to my startup script but it just homes all axis and goes on to print.

Don't over think it.   You don't need every bed leveling option that was ever invented.   UBL lets you have what every want or need at the same time.

The original bed leveling was 3-Point.   You can do that if you want.  You can combine it with a finely tuned mesh.   Probably you don't need it.   But if you are careless and often times let some little plastic threads get under the glass when you put it back on the printer....  combining 3-point with the Mesh is going to work.

totalitarian, 2017-10-04:
@Roxy-3D ideally I want a topology of my bed and then it verifying for tilting before every print as sometimes a filament change or a stubbon print that refused to be removed from the bed can cause the bed to move. You mention I can combine 3-point with the mesh, how would I do that? I did try G28/G29 J but my nozzle ended up about 1mm from the bed.

BTW the mesh test pattern was the most beautiful thing I had ever seen off my bed. It's quite warped in the middle but the pattern printed on bare glass perfectly 👍

totalitarian, 2017-10-04:
I think the penny might have dropped. So with UBL mesh active, it affects everything. So I can do a G29 J before each print to tilt the active mesh?

Tannoo, 2017-10-04:
G29 J3 to start with.

Roxy-3D, 2017-10-04:
>  So I can do a G29 J before each print to tilt the active mesh?

Yes.   The first thing to do is get the mesh well tuned.   That is always used.   But then you can add a G29 J type command to tilt the mesh.

studiodyne, 2017-10-04:
[UBL ALL SECRETS](https://github.com/MarlinFirmware/Marlin/issues/7562)
I'm a newbie and i have made a special UBL UNDERSTANDING for newbie
If i can , you can , of course ! 😅

studiodyne, 2017-10-04:
Some users test UBL , with lcd moves , and it doesn't works , you have to make a G1 XY move to see appears the Z axis correction
Look In my tuto 👍

totalitarian, 2017-10-04:
@Roxy-3D Thanks I think i've got it now. Mesh came out perfect 1st time. I then lowered one side of the bed, ran G29 J3 and re ran the test print. Still came out fine :)

Roxy-3D, 2017-10-04:
Great!

closed 2017-11-15 by thinkyhead