MarlinFirmware/Marlin #7838: autobed level manual with lcd

Radiu, 2017-10-04:
sorry to ask this question but im realy confused in how to set auto bed level manual with lcd
im expecting something to happen but cant figuer out what im missing

the problem is that when i send to do the auto level from lcd  it seems like the is all off set
like will start probe from front left end 10 cm off set from bed conner then does 2 more points to the right and last point does outside of bed, i dont see no place where to put the coords of the points of front,back , left,right for the mesh manual bed to set positions

i have enable
#define mesh_bed_leveling
#define (lcd_bed_leveling)

if i play with the #define mesh_inset i see there is some spacings that gets bigger or small; same if i define the bed size bigger then it is

could someone point me or guide me in how to set points where is going to do the auto level

Radiu, 2017-10-04:
when i home my Y will be on top bed on back with position 0 but my X stays offset from the bed top 24mm
is this why bed not center?
if so how to make off sets for when i home machine will go to bed position and count has 0 home?

Tannoo, 2017-10-04:
In `Configuration.h`:

Set these:
```
// The size of the print bed 
#define X_BED_SIZE 200 
#define Y_BED_SIZE 200 

// Travel limits (mm) after homing, corresponding to endstop positions. 
#define X_MIN_POS 0 
#define Y_MIN_POS 0 
#define Z_MIN_POS 0 
#define X_MAX_POS X_BED_SIZE 
#define Y_MAX_POS Y_BED_SIZE 
#define Z_MAX_POS 200 
```

Then these:
```
#define MESH_INSET 10          // Mesh inset margin on print area 
#define GRID_MAX_POINTS_X 3    // Don't use more than 7 points per axis, implementation limited. 
#define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X 
```

Radiu, 2017-10-04:
@Tannoo  got it to work

sense im using MAX end stops and my travel to MIN left side to stay inside the bed is 198 
and my X axis is offset like 23 steps from bed 

i had to disable

//#define MIN_SOFTWARE_ENDSTOPS

and set bed size to
#define X_BED_SIZE 198

and then apply the manual offset to
#define MANUAL_X_HOME_POS 221

i hope this is how is done, is defenetly working has expected now

Thanks alot for your great help

closed 2017-10-04 by Radiu