MarlinFirmware/Marlin #7839: bugfix-2.0.x Teensy3.6 TEMP_PIN Issues

JXPA, 2017-10-04:
### Bug Report

Hi,
I connected Teensy3.6 and Ramps1.4 and tested it.

Teensy3.6	　　Ramps1.4
　　A11　　　　　A13　　　　　100k beta 3950 1% thermistor
　　3.3V　　　　　VCC
　AGND　　　　　GND
minimum connection

I checked the temperature using Repetier-Host software.
But I could not get the correct temperature.

I tested whether all the analog input pins can measure the voltage correctly.
There was no problem with all analog input pins.
*Teensy3.6 analog input pins of number 14 or higher do not work unless "A" is attached.
```
analogRead( 14 );Do not work
analogRead( A14 );it work
```

Next, I changed the setting of TEMP_0_PIN and tested it.

pin number　/roomtemperature : 25°C
　　　A2　　　　: 25.01°C(correct)
　　　A8　　　　: 25.2°C(correct)
　　　A9　　　　: 25.0°C(correct)
　　　A10　　　: 25.3°C(correct)
　　　A11　　　: 165°C (incorrect)
　　　A13　　　: 958.8°C(incorrect)
　　　A17　　　: 148.81°C(incorrect)
　　　A21　　　: 25.01°C(correct)
　　　A22　　　: 160.53°C(incorrect)
　　　A23　　　: 139.08°C(incorrect)
　　　A24　　　: 161.90°C(incorrect)

**src/HAL/HAL_TEENSY35_36/HAL_Teensy.cpp**
```cpp
static const uint8_t pin2sc1a[] = {
  5, 14, 8, 9, 13, 12, 6, 7, 15, 4, 3, 19+128, 14+128, 15+128, // 0-13 -> A0-A13
  5, 14, 8, 9, 13, 12, 6, 7, 15, 4, // 14-23 are A0-A9
  255, 255, 255, 255, 255, 255, 255, // 24-30 are digital only
  14+128, 15+128, 17, 18, 4+128, 5+128, 6+128, 7+128, 17+128,  // 31-39 are A12-A20
  255, 255, 255, 255, 255, 255, 255, 255, 255,  // 40-48 are digital only
  10+128, 11+128, // 49-50 are A23-A24
  255, 255, 255, 255, 255, 255, 255, // 51-57 are digital only
  255, 255, 255, 255, 255, 255, // 58-63 (sd card pins) are digital only
  3, 19+128, // 64-65 are A10-A11
  23, 23+128,// 66-67 are A21-A22 (DAC pins)
  1, 1+128,  // 68-69 are A25-A26 (unused USB host port on Teensy 3.5)
  26,        // 70 is Temperature Sensor
  18+128     // 71 is Vref
};

```
It seems that pins processed for "+128" are not working properly.

- Expected behaviour: Correct temperature is displayed.
- Actual behaviour: Incorrect temperature is displayed.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1354362/Marlin.zip)

teemuatlut, 2017-10-04:
Well it's good to know that at least some of them work =)
It's likely that when I was testing whether or not the temperature readings work, I just got lucky and picked a pin that is OK to use.

I believe the problem can be found from [this](https://github.com/PaulStoffregen/cores/blob/master/teensy3/analog.c#L449). Basically any time the channel is over 128, then we should be using ADC1 instead of ADC0. This likely requires a small change into the HAL.

Donziboy2, 2017-10-04:
The really sad part is Teensy 3.1-3.6 have hardware DMA that can run the ADC's independent of the processor.  Meaning you can sample at up to 800Ksps with each of the 2 ADC's onboard.

teemuatlut, 2017-10-04:
Not sad, just room for future development. Let's first try to get the basic functionality, testing and hardware done before we move on to more advanced features.

Also I have very little knowledge of the DMA and how to use it. Can we use it to read three or four pins? Are there any simple examples that demonstrate how to set it up? What are the actual advantages compared to normal sampling? I believe the temperature ISR runs at around 1KHz and taking turns on which pin to read.

Donziboy2, 2017-10-04:
With DMA the CPU does nothing, its separate hardware that sets up the ADC's, selects the channel to use and initiates the Read.  Then it takes the returned value and sticks it in a buffer, once the buffer is full it signals an interrupt and we normally move the data to another buffer.  Normally when you call an ADC read the CPU does all of the above, meaning its idling during the read and waits for the values to come back then it goes back to work.  If your sampling alot of ADC's or just a few very often it eats up alot of CPU time just idling. By default at 96Mhz the Teensy samples the ADC 4 times and gives you the average.  Which takes around 4.6uS to complete all 4 reads and return the value.

One example can be found [here ](https://forum.pjrc.com/threads/30171-Reconfigure-ADC-via-a-DMA-transfer-to-allow-multiple-Channel-Acquisition) it uses the ADC Library to allow us to setup a few additional things but those items can be done manually also if you know the register calls for the Teensy Core.

teemuatlut, 2017-10-04:
Interesting. I'll definitely take a look at that thread.
Fortunately Marlin is not quite inefficient like that. The way it's set up is once the temperature ISR (1kHz) is triggered, it sets up the ADC and then proceeds with other tasks. The next time the ISR triggers, it reads the value. Then the next another pin case and so on. It's certainly not as efficient as offloading it to DMA but not as bad as just calling analogRead().

Donziboy2, 2017-10-05:
Could you point me towards that fancy code because i've never heard of ADC's being used like that.  Ive read the Teensy bibles a few times for both the 3.2 and 3.6 ADC sections and never saw mention of that.  Other then continuous conversion with polling.

AnHardt, 2017-10-05:
https://github.com/MarlinFirmware/Marlin/blob/9c00cff89836706776f6353635368a47cc5bf881/Marlin/temperature.cpp#L1610
espetially the state machine from
https://github.com/MarlinFirmware/Marlin/blob/9c00cff89836706776f6353635368a47cc5bf881/Marlin/temperature.cpp#L1924
on.
The ultimate sources are the processors datasheets.

boelle, 2019-02-19:
@JXPA can you confirm that the problem has been fixed in the latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

Phr3d13, 2019-06-10:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

boelle, 2019-06-22:
yep can be closed still

boelle, 2019-07-21:
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

PR #12258 (2018-10-28): [2.0.x] Fix for Teensy 3.5/3.6 ADC access for ADC1
Ive changed the ADC function to check for a ADC1 pin and uses ADC1 instead of ADC0, allowing all of the analog pins to be used. Not tested on a 3.5 but should be the same as 3.6

Possibly fixes #7839
closed 2019-07-21 by boelle