MarlinFirmware/Marlin #7848: Reglage marlin 1.1.5 mini kossel capteur inductif

petitbrochet, 2017-10-05:
bonjour
Après avoir installé marlin 1.1.5, fait l'autocalibration et avoir renté le décalage de la buse par rapport au capteur.
L'impression débute toujours a 2.48mm du plateau.
J'ai beau faire différent réglage mais toujours le même problème.
Qui pour m'aider?
Merci d'avance

teemuatlut, 2017-10-05:
In English? It'll be hard to help you if we don't know what you're saying.

petitbrochet, 2017-10-05:
Hello
After installing marlin 1.1.5, the autocalibration is performed and the nozzle offset relative to the sensor.
Printing always starts at 2.48mm from the tray.
Although I do different setting but still the same problem.
Who to help me?
thank you in advance

petitbrochet, 2017-10-06:
Good evening
person to help me?
Please

thinkyhead, 2017-11-15:
Sorry this issue was missed. It has been a very busy Autumn.

boelle, 2019-02-19:
@petitbrochet did you try the latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

Phr3d13, 2019-06-10:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

boelle, 2019-07-21:
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle