MarlinFirmware/Marlin #7850: When i compile fresh git checkout with DEFAULT_MAX_ACCELERATION i get error:

Lostish, 2017-10-05:
Standard clone of a fresh new marlin on master.
Cant get it compile with the standard values.

The error when compiling:
SanityCheck.h:1307:14: error: expected constructor, destructor, or type conversion before ‘(’ token
 static_assert(COUNT(sanity_arr_3) <= XYZE_N, "DEFAULT_MAX_ACCELERATION has too many elements.");
              ^

What can be done here?

jlucguillen, 2017-10-08:
Hello there,
I have the same issue
trying to upload marlin in my Infitary M508
I replace the configuration.h and configuration_adv.h by the files from /example_configurations/infitary
in Arduino IDE I set up with 
board : Arduino mega 2560 or Mega ADK
port : /dev/ttyACMo  ( i 'm using UBUNTU) 

and the result is the same message
I have seen than in the original firmware of the infitary there is this line
`#define DEFAULT_MAX_ACCELERATION      {9000,9000,100,10000}    // X, Y, Z, E maximum start speed for accelerated moves. E default values are good `
and in the file form marlin
`#define DEFAULT_MAX_ACCELERATION      { 3000, 3000, 100, 10000 }`

please can you help ?

fiveangle, 2017-10-08:
Works fine here.  You need to upload your config*s

fiveangle, 2017-10-08:
Works fine here:

<details><summary>build log</summary>

```
/Applications/Arduino 1.8.3-teensy.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino 1.8.3-teensy.app/Contents/Java/hardware -hardware /Users/speedster/Library/Arduino15/packages -hardware /Users/speedster/Documents/Arduino/hardware -tools /Applications/Arduino 1.8.3-teensy.app/Contents/Java/tools-builder -tools /Applications/Arduino 1.8.3-teensy.app/Contents/Java/hardware/tools/avr -tools /Users/speedster/Library/Arduino15/packages -built-in-libraries /Applications/Arduino 1.8.3-teensy.app/Contents/Java/libraries -libraries /Users/speedster/Documents/Arduino/libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10803 -build-path /var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_482709 -warnings=default -build-cache /var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_cache_498663 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/Users/speedster/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2 -prefs=runtime.tools.arduinoOTA.path=/Users/speedster/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.1.1 -prefs=runtime.tools.avrdude.path=/Users/speedster/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino9 -verbose /Users/speedster/dev/github/Marlin/Marlin/Marlin.ino
/Applications/Arduino 1.8.3-teensy.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino 1.8.3-teensy.app/Contents/Java/hardware -hardware /Users/speedster/Library/Arduino15/packages -hardware /Users/speedster/Documents/Arduino/hardware -tools /Applications/Arduino 1.8.3-teensy.app/Contents/Java/tools-builder -tools /Applications/Arduino 1.8.3-teensy.app/Contents/Java/hardware/tools/avr -tools /Users/speedster/Library/Arduino15/packages -built-in-libraries /Applications/Arduino 1.8.3-teensy.app/Contents/Java/libraries -libraries /Users/speedster/Documents/Arduino/libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10803 -build-path /var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_482709 -warnings=default -build-cache /var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_cache_498663 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/Users/speedster/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2 -prefs=runtime.tools.arduinoOTA.path=/Users/speedster/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.1.1 -prefs=runtime.tools.avrdude.path=/Users/speedster/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino9 -verbose /Users/speedster/dev/github/Marlin/Marlin/Marlin.ino
Using board 'mega' from platform in folder: /Users/speedster/Library/Arduino15/packages/arduino/hardware/avr/1.6.20
Using core 'arduino' from platform in folder: /Users/speedster/Library/Arduino15/packages/arduino/hardware/avr/1.6.20
Detecting libraries used...
"/Users/speedster/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Users/speedster/Library/Arduino15/packages/arduino/hardware/avr/1.6.20/cores/arduino" "-I/Users/speedster/Library/Arduino15/packages/arduino/hardware/avr/1.6.20/variants/mega" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_482709/sketch/Marlin.ino.cpp" -o "/dev/null"
"/Users/speedster/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Users/speedster/Library/Arduino15/packages/arduino/hardware/avr/1.6.20/cores/arduino" "-I/Users/speedster/Library/Arduino15/packages/arduino/hardware/avr/1.6.20/variants/mega" "-I/Applications/Arduino 1.8.3-teensy.app/Contents/Java/libraries/LiquidCrystal/src" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_482709/sketch/Marlin.ino.cpp" -o "/dev/null"
"/Users/speedster/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Users/speedster/Library/Arduino15/packages/arduino/hardware/avr/1.6.20/cores/arduino" "-I/Users/speedster/Library/Arduino15/packages/arduino/hardware/avr/1.6.20/variants/mega" "-I/Applications/Arduino 1.8.3-teensy.app/Contents/Java/libraries/LiquidCrystal/src" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_482709/sketch/G26_Mesh_Validation_Tool.cpp" -o "/dev/null"
"/Users/speedster/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Users/speedster/Library/Arduino15/packages/arduino/hardware/avr/1.6.20/cores/arduino" "-I/Users/speedster/Library/Arduino15/packages/arduino/ [... cut ...]
Global variables use 4158 bytes (50%) of dynamic memory, leaving 4034 bytes for local variables. Maximum is 8192 bytes.
```

</details>

```
"/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_482709/Marlin.ino.hex"
Using library LiquidCrystal at version 1.0.5 in folder: /Applications/Arduino 1.8.3-teensy.app/Contents/Java/libraries/LiquidCrystal 
Sketch uses 102480 bytes (40%) of program storage space. Maximum is 253952 bytes.
Global variables use 4158 bytes (50%) of dynamic memory, leaving 4034 bytes for local variables. Maximum is 8192 bytes.
```

jlucguillen, 2017-10-09:
Hello
thx for the reply
I'm  using this branch 1.1.x - 
i'm glad to read that it's working for you and I'm sur that it's working for every body... that's why i'm trying to use it.. but it's not working for me....

Please tell me how can i get this "build log" , then I'll post it here for you can read it and lead me to the solution
Ask me for all the information you need but tell me how I can reach them (command line , path,..) 
thx in advance

jlucguillen, 2017-10-09:
@Lostish - thx to read my post,
 I'm using the branch 1.1.x
I'm french and I don't really understand the sentence "above is bf1 and 1.1.5 with those example configs dropped in"
can you please tell me which informations you need and how can I get them (command line,  file's path,...) then I'll post them here for you to read and lead me to the solution
Thx a lot

fiveangle, 2017-10-09:
That message was for @Lostish so it was not really intended for you.

Summary: I can not reproduce your problem with the limited information you provided.
[Je ne peux pas reproduire votre problème avec les informations limitées que vous avez fournies.]

fiveangle, 2017-10-09:
Configuration.h, Configuration_adv.h

Lostish, 2017-10-09:
I have tried some things now.
This error i get on linux.
And i may miss libraries for the lcd that does it.
Dont know if they are in linux by standard or not.
As they are not in wondows. I assume they are not in linux either.

Running windows runs great.
u8lib for lcd was required for compiling in windows.
And windows works. So i need to recheck why its not working in linux.

So the problem in this case is Arduino IDE  with linux

jlucguillen, 2017-10-09:
@fiveangle Hello
Thx a lot
as the forum doesn't support .h extention, I add .txt extension to upload those files here
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/1367791/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/1367790/Configuration_adv.h.txt)

those files are in Marlin-1.1.x\Marlin\example_configurations\Infitary\i3-M508

If you need other informations, please tell me

jlucguillen, 2017-10-09:
@fiveangle , @Lostish 
thx a lot for your cooperation
reading  @Lostish  last comment about the u8lib for LCD i have made some research and find this 
http://forums.reprap.org/read.php?110,554303 which is about this missing librarie creating issue with LCD screen and marlin firmware
I have tested the  Marlin-1.1.x\Marlin\ in the arduino IDE on my windows PC ( at hte office)  ( including liquidChrystal librarie)  and it's compiling well...
So this evening( at home)  I'll check on my linux if all the libraries are include, and I'll also try to compile with my windows PC and upload from windows to my print INFITARY

I will post the result of my tests here

jlucguillen, 2017-10-09:
@fiveangle , @Lostish
hi,
From my PC windows
arduino IDE 1.8.5
 compilition has worked and i have been able to upload in my INFITARY..  yes !!!
I don't know why it didn't work from ubuntu 16.04. perhaps I'll check it when i had more time ;) or if you asked me to do it...
Now, for the screen my button  is inverted : when I turn it to the right, on the the screen the cursor is going down when it should go up... but i really don't care ( I'm to happy to have marlin on my printer !! )
And I have an other question :  on the screen the letters X, Y Z are blinkink and switching to "?"
do you know why ?
thx

fiveangle, 2017-10-09:
Search your config files for "REVERSE"... to see where to change direction, either entire encoder, or only numerical values.

Close this issue if you're all good :+1: 

-=dave

jlucguillen, 2017-10-10:
@fiveangle   OK I found it thx a lot

> // This option reverses the encoder direction for navigating LCD menus.
> //
> //  If CLOCKWISE normally moves DOWN this makes it go UP.
> //  If CLOCKWISE normally moves UP this makes it go DOWN.
> //
> //#define REVERSE_MENU_DIRECTION
> 
> 

thx a lot... I ll do it later 
WE ARE ALL GOOD

jlucguillen, 2017-10-10:
@Lostish Hi, 
for me it's OK thx a lot
Close this issue if you're all good

closed 2017-11-15 by thinkyhead

didtoknan, 2018-07-09:
I know it has been closed but it should not have been. It has not been really resolved.
I have the same error when I compile on raspbian. On Windows I have no error.
I really want to be able to compile with raspbian because my Pi is next to my printer (Octopi) and my desktop is not.