MarlinFirmware/Marlin #7856: GPL Violation

KangDroid, 2017-10-06:
Hi, I would like to get help/support from Marlin developers, 
Since I bought CORE200 3D Printer from Korea(Made in Korea as well), I was able to sign up its forum.(Its all in Korean)

On October 3th,(GMT 9+), I was asking the forum moderator about : what is the firmware base, and moderator said, its "Marlin"-based. It basically means CORE200's firmware affected by GPL V3. 
Also, I have seen that moderator released its firmware by "hex", not source. So I was asking to release the source since I want to modify and improve itself.

First reply from moderator was: "Right now, I am not thinking about releasing code because I am keep editing, and If it gets more stable, then I will think about it."
As I am also Opensource developer(AndroidDev) I also knew that if someone released its binary based on GPL V3 affected source, that someone have to release its source code as well.
So, I was linking GPL Site to moderator, and saying you need to open its source by user's request.(For released binary one.) and now, the moderator is keep ignoring open-source request, even I sent him a private mail about this, and he is keep ignoring. 

I know its quite complicated and my English is really bad. Please understand it, and if there is more information needed, I will provide it as much as I can, AND,
I would like to post a link about forum, OR a screenshot of its forum, but its forum is in all Korean, and basically, its private who didn't signed-up so the link would not work. I was thinking about report to http://gpl-violations.org ,but it would be right to report main author first.
Thanks for reading long-way post.

Roxy-3D, 2017-10-06:
Thank You for the information.  If they are shipping the Marlin Firmware on the printers they are selling, they need to make to source (especially the Configuration.h and Configuration_adv.h files) available to the customers.

Please send them the link to this thread.   And in the mean time, we can probably help you regenerate the Configuration.h and Configuration_adv.h. file so you can use the very latest version of the firmware.

Assuming you are running a recent (in the last 6 months) version of Marlin, M503 will give you most of the Configuration.h settings.   If you save the Boot Splash information and do the M503....   Probably you will have enough to reconstruct the Configuration.h file.   At that point you will be free from the vendor and able to support your self.

KangDroid, 2017-10-06:
Thanks for the clear-reply and some useful information about configuration!.

At this point right now, I am at long holiday so I couldn't test M503 since printers are in laboratory.

Anyway, Thanks in advanced.(Will contact you again if M503 is not working - can't sure whether its running a recent version of marlin.)

Roxy-3D, 2017-10-06:
Just so you understand...   The main options to setup the printer are in Configuration.h.   If you get those setup correctly, the printer will pretty much work.   But the fancier settings are in Configuration_adv.h.   Probably, you don't need much to be changed in that file.   Things like Z-BabyStepping and Double Clicking the encoder wheel to activate it are in that file.   Anything in that file you can change over time to match your preferences.

KangDroid, 2017-10-06:
Thanks for the info dev.
I've been touching configuration.h for delta, prusa i3, but I never touched CoreXY type, I will try best as I can after the holiday is over.

Respectfully,
KangDroid.

KangDroid, 2017-10-06:
@Roxy-3D
I posted another post with the link, 
Moderator said: "If there is a problem with law, I will think about it, and  most commercial 3D-Printer is based on Marlin, and I heard that not everyone is opening there source"

I know that you are the copyright(or left) holder, if you know any knowledge about GPL law(like international..) please tell me about it, and I apologize for keep bothering you about this complicated problem. I bet we need more specific proof about law so the moderator would understand.

Tannoo, 2017-10-06:
>They are legally required to do that.

Umm.. doesn't that only apply to the U.S.A.?

coldtobi, 2017-10-09:
> Umm..  doesn't that only apply to the U.S.A.?

No.

TiGa-RCE, 2017-10-11:
Same thing for the Anycubic i3 Mega. It uses the firmware but distributes it only in hex format.
I created a new pins file for their board but I haven't been able to make the TFT screen work, I had to replace it with a cheap LCD.

closed 2017-11-15 by thinkyhead

chrisalbertson, 2018-01-12:
Many companies are violating the GPL.   It might be worth seeing what can be done.   First explain it to them.  Point out the terms they accepted.

I might simply be that these companies don't even know what "source code" is.  They could be hired out the work and only have access to the binary files.

I don't know the simplest next step.     Possibly work  with ebay and Amazon to get their product off of those sites.

thinkyhead, 2018-01-12:
I've never heard of a product getting pulled due to GPL violation, at least not without first getting sued. But I will look into it and see if there's something like the EFF that would back such a suit.

MisterPrinterProblems, 2019-03-12:
Any updates here other than the Creality release?  I am now stuck here not able to add a bl touch or different extruder.