MarlinFirmware/Marlin #7860: Error Auto Bed Level on Marlin 1.1.5 in Geeetech Prusa i3 Pro B

boludiko, 2017-10-06:
Hello, I'm trying to get work a BLTouch sensor in a Geeetech Prusa I3 Pro B. All the hardware is working and I can get the grid from the sensor with the different values.

Bilinear Leveling Grid:
      0      1      2
 0 -0.355 -0.130 -0.250
 1 -0.165 +0.105 -0.080
 2 -0.175 +0.060 +0.162

But when I print this grid is not applied so the leveling is not working. Here is an photograph.

[image]

My Cura starting code is this.

M851 Z-0.7 ;
G21 ;
G90 ;
M82 ;
M104 S{material_print_temperature} ;
M140 S{material_bed_temperature} ;
M190 S{material_bed_temperature} ;
M109 S{material_print_temperature} ;
G28 ;
G29 ;
M420 S1 ;
G1 X0 F12000.0 ;
G1 Y0.0 F12000.0 ;
G1 Z0.1 F12000.0 ;
G92 E0.0 ;
G1 X60.0 E9.0 F1000.0 ;
G1 X100.0 E21.5 F1000.0 ;
G92 E0.0 ;

I try with and without M420 S1 and always the same. The system is configured with the Bilineal method.

Here are the configuration and configuration_adv files.

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/1363311/Configurations.zip)

I don't know what more I can try. The Lineal system do exactly the same. RCbugfix doesn't work too.

I anybody can help me.

Thanks for your time and sorry for my English.

guiseco, 2017-10-06:
Same problem every where. Bilinear do not work for me. I'm using LINEAR.

guiseco, 2017-10-06:
https://github.com/MarlinFirmware/Marlin/issues/7278

boludiko, 2017-10-07:
Thanks for answer. If you have the same printer and your board is gt2560 a+ please can you send me your firmware and conf files. I try linear but it doesnt work.

fiveangle, 2017-10-09:
Have you both tried these options ?

//#define PROBING_HEATERS_OFF       // Turn heaters off when probing
//#define PROBING_FANS_OFF          // Turn fans off when probing
//#define DELAY_BEFORE_PROBING 200  // (ms) To prevent vibrations from triggering piezo sensors

luismi7777, 2017-10-17:
Im  the same problem have you solved?

boludiko, 2017-11-13:
Hi everybody. Recently I change the firmware from 1.1.5 to 1.1.6 and I make some test from different methods. Lineal and BiLineal. The results are the same.

@fiveangle yes I tried these options.

@guiseco The lineal method doest work for me.

On this video you can see the BiLineal method

<iframe width="560" height="315" src="https://www.youtube.com/embed/hHqLBEmPVUc?rel=0" frameborder="0" allowfullscreen></iframe>

and on this video you can see the Lineal method.

<iframe width="560" height="315" src="https://www.youtube.com/embed/S5EU1_-PV3I?rel=0" frameborder="0" allowfullscreen></iframe>

I don't know how to fix it.

boludiko, 2017-11-14:
@guiseco @fiveangle @thinkyhead This is my last test.

I just performed a manual calibration for the bed and everything it’s ok. 
After the calibration then i perform a Z-Probe Repeatability Test and this are the results:

SENDING:M48 P20 V4
M48 Z-Probe Repeatability Test
Positioning the probe...
Bed X: 90.000 Y: 95.000 Z: -0.003
1 of 20: z: -0.008 mean: -0.0075 sigma: 0.000000 min: -0.008 max: -0.008 range: 0.000
2 of 20: z: -0.011 mean: -0.0094 sigma: 0.001875 min: -0.011 max: -0.008 range: 0.004
3 of 20: z: -0.008 mean: -0.0088 sigma: 0.001768 min: -0.011 max: -0.008 range: 0.004
4 of 20: z: -0.011 mean: -0.0094 sigma: 0.001875 min: -0.011 max: -0.008 range: 0.004
5 of 20: z: -0.011 mean: -0.0098 sigma: 0.001837 min: -0.011 max: -0.008 range: 0.004
6 of 20: z: -0.011 mean: -0.0100 sigma: 0.001768 min: -0.011 max: -0.008 range: 0.004
7 of 20: z: -0.014 mean: -0.0105 sigma: 0.002098 min: -0.014 max: -0.008 range: 0.006
8 of 20: z: -0.009 mean: -0.0103 sigma: 0.002049 min: -0.014 max: -0.008 range: 0.006
9 of 20: z: -0.013 mean: -0.0106 sigma: 0.002051 min: -0.014 max: -0.008 range: 0.006
10 of 20: z: -0.014 mean: -0.0109 sigma: 0.002169 min: -0.014 max: -0.008 range: 0.006
11 of 20: z: -0.013 mean: -0.0110 sigma: 0.002120 min: -0.014 max: -0.008 range: 0.006
12 of 20: z: -0.014 mean: -0.0113 sigma: 0.002165 min: -0.014 max: -0.008 range: 0.006
13 of 20: z: -0.014 mean: -0.0114 sigma: 0.002184 min: -0.014 max: -0.008 range: 0.006
14 of 20: z: -0.013 mean: -0.0115 sigma: 0.002122 min: -0.014 max: -0.008 range: 0.006
15 of 20: z: -0.016 mean: -0.0118 sigma: 0.002366 min: -0.016 max: -0.008 range: 0.009
16 of 20: z: -0.010 mean: -0.0117 sigma: 0.002333 min: -0.016 max: -0.008 range: 0.009
17 of 20: z: -0.014 mean: -0.0118 sigma: 0.002314 min: -0.016 max: -0.008 range: 0.009
18 of 20: z: -0.014 mean: -0.0119 sigma: 0.002291 min: -0.016 max: -0.008 range: 0.009
19 of 20: z: -0.010 mean: -0.0118 sigma: 0.002271 min: -0.016 max: -0.008 range: 0.009
20 of 20: z: -0.014 mean: -0.0119 sigma: 0.002253 min: -0.016 max: -0.008 range: 0.009
Finished!
Mean: -0.011938 Min: -0.016 Max: -0.008 Range: 0.009
Standard Deviation: 0.002253

With this results i send a G29 command and there is the results:

SENDING:G29

Bilinear Leveling Grid:
      0      1      2
 0 +0.211 -0.011 -0.690
 1 +0.199 +0.039 -0.606
 2 +0.110 -0.053 -0.686
Subdivided with CATMULL ROM Leveling Grid:
        0        1        2        3        4        5        6
 0 +0.21125 +0.15398 +0.09671 -0.01125 -0.20370 -0.44685 -0.69000
 1 +0.20991 +0.16080 +0.11169 +0.01065 -0.17695 -0.41648 -0.65602
 2 +0.20856 +0.16762 +0.12667 +0.03255 -0.15020 -0.38612 -0.62204
 3 +0.19875 +0.16338 +0.12801 +0.03875 -0.14032 -0.37329 -0.60625
 4 +0.17481 +0.14072 +0.10662 +0.01880 -0.15858 -0.38968 -0.62079
 5 +0.14241 +0.10700 +0.07160 -0.01685 -0.19371 -0.42361 -0.65352
 6 +0.11000 +0.07329 +0.03657 -0.05250 -0.22884 -0.45755 -0.68625

Now, if i send a G1 Z0 on one corner (X186 Y181) the nozzle is touching the bed without level. 
So i calibrate the bed without doing G28, moving the nozzle with G1. (G1 X0 Y0 G1 X0 Y181, G1 X186 Y181, G1 X186 Y0)
After check that everything it’s ok i repeat the G29 command with this results:

Bilinear Leveling Grid:
      0      1      2
 0 +0.186 -0.341 -1.336
 1 +0.177 -0.343 -1.301
 2 +0.066 -0.450 -1.405
Subdivided with CATMULL ROM Leveling Grid:
        0        1        2        3        4        5        6
 0 +0.18625 +0.02773 -0.13079 -0.34125 -0.63829 -0.98727 -1.33625
 1 +0.18713 +0.02910 -0.12894 -0.33773 -0.63113 -0.97529 -1.31944
 2 +0.18801 +0.03046 -0.12709 -0.33421 -0.62397 -0.96330 -1.30264
 3 +0.17750 +0.02042 -0.13667 -0.34250 -0.62958 -0.96542 -1.30125
 4 +0.14801 -0.00864 -0.16530 -0.37046 -0.65649 -0.99102 -1.32556
 5 +0.10713 -0.04911 -0.20536 -0.41023 -0.69616 -1.03072 -1.36528
 6 +0.06625 -0.08958 -0.24542 -0.45000 -0.73583 -1.07042 -1.40500

And now again if i send G1 Z0 on the same corner (X186 Y181) the nozzle hits the bed.

It’s anything bad with the G29 command and the mesh calculation?

boludiko, 2017-12-04:
@thinkyhead hello. Any update on this?

iz3man, 2017-12-04:
would be interested to see some update as well.
i see that with bilinear abl the z-axis is moving up and down to compensate, but it seems it's undercompensating on one side, and overcompensation on another.
no idea what's going wrong here. 
using marlin 1.1.6 with a DC42 ir sensor.

thinkyhead, 2017-12-05:
Please test the current `bugfix-1.1.x` branch and report back. Issues might be fixed.

boludiko, 2017-12-11:
I will try it now. Later I come back with feedback. Thanks.

boludiko, 2017-12-12:
Hello. @thinkyhead the results are awesome. Now the printer makes the bed level okey. I have to make some changes on the pins configuration for the GT2560 because the pin for the BLTouch is not defined well. And I have to increase the time for the bed heater because the systems halt. But for the rest, I can't be more happiest.

Thanks for the support.

closed 2017-12-12 by boludiko