MarlinFirmware/Marlin #7861: Unable to compile bugfix-2.0.x branch for RADDS

CReimer, 2017-10-06:
When I compile the bugfix-2.0.x branch for RADDS I get this error:
```
In file included from sketch/src/lcd/ultralcd.cpp:87:0:
sketch/src/lcd/ultralcd_impl_DOGM.h:436:13: warning: unused parameter 'blink' [-Wunused-parameter]
 inline void lcd_implementation_status_message(const bool blink) {
             ^
sketch/src/module/stepper.cpp: In static member function 'static void Stepper::isr()':
sketch/src/module/stepper.cpp:342:72: error: invalid operands of types 'uint32_t {aka long unsigned int}' and 'double' to binary 'operator%'
       const uint32_t remainder = (uint32_t)L % (ENDSTOP_NOMINAL_OCR_VAL); \
                                                                        ^
sketch/src/module/stepper.cpp:693:5: note: in expansion of macro 'SPLIT'
     SPLIT(timer);  // split step into multiple ISRs if larger than  ENDSTOP_NOMINAL_OCR_VAL
     ^
sketch/src/module/stepper.cpp:342:72: error: invalid operands of types 'uint32_t {aka long unsigned int}' and 'double' to binary 'operator%'
       const uint32_t remainder = (uint32_t)L % (ENDSTOP_NOMINAL_OCR_VAL); \
                                                                        ^
sketch/src/module/stepper.cpp:754:5: note: in expansion of macro 'SPLIT'
     SPLIT(timer);  // split step into multiple ISRs if larger than  ENDSTOP_NOMINAL_OCR_VAL
     ^
sketch/src/module/stepper.cpp:342:72: error: invalid operands of types 'uint32_t {aka long unsigned int}' and 'double' to binary 'operator%'
       const uint32_t remainder = (uint32_t)L % (ENDSTOP_NOMINAL_OCR_VAL); \
                                                                        ^
sketch/src/module/stepper.cpp:804:5: note: in expansion of macro 'SPLIT'
     SPLIT(OCR1A_nominal);  // split step into multiple ISRs if larger than  ENDSTOP_NOMINAL_OCR_VAL
     ^
exit status 1
```

p3p, 2017-10-06:
This [line](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/HAL/HAL_DUE/HAL_timers_Due.h#L52) needs to be changed to make sure its an integer.
`#define HAL_TICKS_PER_US       (((F_CPU) / 2) / 1000000) // stepper timer ticks per us`

We need to get at least 1 basic test for each platform added to Travis sorry about that.

CReimer, 2017-10-06:
Thanks. That fixed it.

closed 2017-10-06 by CReimer

p3p, 2017-10-06:
@thinkyhead could you change this directly, can't make a PR right now. 

May need to change how the macro is used eventually, don't like the repeated definition, also the fact they are double constants ..