MarlinFirmware/Marlin #7868: Unable to compile bugfix-2.0.x branch with USE_CONTROLLER_FAN

CReimer, 2017-10-06:
When I compile with `USE_CONTROLLER_FAN` I get this error:
```
sketch/src/feature/controllerfan.cpp: In function 'void controllerfan_update()':
sketch/src/feature/controllerfan.cpp:33:9: error: 'X_ENABLE_READ' was not declared in this scope
     if (X_ENABLE_READ == X_ENABLE_ON || Y_ENABLE_READ == Y_ENABLE_ON || Z_ENABLE_READ == Z_ENABLE_ON || thermalManager.soft_pwm_amount_bed > 0
         ^
sketch/src/feature/controllerfan.cpp:33:41: error: 'Y_ENABLE_READ' was not declared in this scope
     if (X_ENABLE_READ == X_ENABLE_ON || Y_ENABLE_READ == Y_ENABLE_ON || Z_ENABLE_READ == Z_ENABLE_ON || thermalManager.soft_pwm_amount_bed > 0
                                         ^
sketch/src/feature/controllerfan.cpp:33:73: error: 'Z_ENABLE_READ' was not declared in this scope
     if (X_ENABLE_READ == X_ENABLE_ON || Y_ENABLE_READ == Y_ENABLE_ON || Z_ENABLE_READ == Z_ENABLE_ON || thermalManager.soft_pwm_amount_bed > 0
                                                                         ^
sketch/src/feature/controllerfan.cpp:33:105: error: 'thermalManager' was not declared in this scope
     if (X_ENABLE_READ == X_ENABLE_ON || Y_ENABLE_READ == Y_ENABLE_ON || Z_ENABLE_READ == Z_ENABLE_ON || thermalManager.soft_pwm_amount_bed > 0
                                                                                                         ^
sketch/src/feature/controllerfan.cpp:34:12: error: 'E0_ENABLE_READ' was not declared in this scope
         || E0_ENABLE_READ == E_ENABLE_ON // If any of the drivers are enabled...
            ^
exit status 1
```

Thanks

vMeph, 2017-10-07:
fix is here https://github.com/MarlinFirmware/Marlin/issues/7791
@thinkyhead   @Roxy-3D possible to fix this in main files?

fiveangle, 2017-10-07:
#7877 submitted to resolve

CReimer, 2017-10-07:
That fixed it. Thanks

closed 2017-10-07 by CReimer