MarlinFirmware/Marlin #7880: bugfix-2.0.x: RADDS and ReprapDiscount Full Graphics Controller only shows random pixels

CReimer, 2017-10-07:
### Bug Report
- Description: Unable to use Graphics Display with RADDS and bugfix-2.0.x branch
- Expected behaviour: Display shows info screen and menus
- Actual behaviour: ReprapDiscount Full Graphics Controller shows random pixels
- Steps to reproduce:
 Connect ReprapDiscount Full Graphics Controller to Pins 44, 47 and 46 (SCK, MOSI, CS)
 
Successfully tested with example sketch from u8glib and the following settings
```U8GLIB_ST7920_128X64_4X u8g(44, 47, 46);```

LCD Enc. Pin 7 -> D46 -> LCD_PINS_RS -> ST7920_CS_PIN
LCD Enc. Pin 8 -> D47 -> LCD_PINS_ENABLE -> ST7920_DAT_PIN
LCD Enc. Pin 5 -> D44 ->LCD_PINS_D4 -> ST7920_CLK_PIN

I think it's also worth mentioning that

LCD Enc. Pin 7 is in the official documentation described as LCD 6. LCD 4 should actually be LCD Enc. Pin 5
LCD_PINS_ENABLE should be Pin 4 not Pin 8
LCD_PINS_RS should be Pin 3 not Pin 7

BEEPER_PIN, BTN_EN1, BTN_EN2 BTN_ENC are correct. I haven't checked all pins on this port yet, but until now these 4 pins are the only ones on this port which are 3.3V logic. The display pins are connected to a level shifter.
I'll check the rest when I try to connect my ReprapDiscount Smart Controller to it, which needs all LCD Pins.

AnHardt, 2017-10-07:
Obviously our highly AVR 8bit optimized `U8GLIB_ST7920_128X64_RRD` device is not suitable for other platforms.
For a quick (and dirty) fix it only needs a `#if ENABLED(***AVR_8BIT***) ` to use our device else use the stock library device - about here.
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/ultralcd_impl_DOGM.h#L155-L160

CReimer, 2017-10-07:
I expected something like that. But didn't look in the right place :disappointed: Now it works fine.

rafaljot, 2017-10-10:
So. Bye, bye ReprapDiscount Smart Controller.  What is the best alternative?

AnHardt, 2017-10-10:
@rafaljot  You couldn't be wronger. `ReprapDiscount Smart Controller` is a character based display -
while we talked about `ReprapDiscount Full Graphics Controller` - a graphical one.
The main difference is the amount of data what has to be produced, and worse, has to be transferred to the display ( 80 vs 1000 Bytes ) . The problem with the transfer to the graphical display is - its already at the maximum speed the display can handel (and sometimes above - for that the *waits* are for). That means even a much faster processor will need the same time - not the same amount of processor cycles. Only a strategy where we can actively use the cycles we waste in the currently active *waits* will give us a win (RTOS or similar). Even hardware based SPI is not a solution when we actively wait for the byte to be transferred, like U8g currently does.  
So despite `ReprapDiscount Smart Controller` displays a bit less of information, it's the faster (better) alternative (when working).

AnHardt, 2017-10-10:
Studying the *current* `ultralcd_st7920_u8glib_rrd.h` i realize - it's already modified to handle the Re-ARM-SPI-transfer partially in the HAL.
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd.h#L91-L98

Alone the reasoning (// Re-ARM requires that the LCD and the SD card share a single SPI) makes me nervous. Isn't it the other way around? Isn't REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER (and similar ) sharing the SPI-lines on that board? Wouldn't it be possible to connect a pure ST7920 and a separated SD-Card socket to different pins on the Re-ARM (and similar)?  So `SHARED_SPI` seems to be a property of `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` - not `IS_REARM`. `...\Marlin\src\HAL\HAL_LPC1768\spi_pins.h` again offers an other perspective - using the Re-ARM on-board SD- socket with hardware SPI. Even here `SHARED_SPI` has the limited view of a `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` (and similar ) in mind.

`SHARED_SPI` is not a good name to signal the use of hardware supported SPI!

rafaljot, 2017-10-10:
Sorrrrry. I wanted to say "So. Bye, bye ReprapDiscount Full Graphics Controller. What is the best alternative?"

CReimer, 2017-10-10:
People told me about the "sluggish" behaviour of the Full Graphics LCD before. I didn't want to hear it.
I looks better, but usability-wise it's much much worse. I'll go back to the character display soon.

closed 2017-11-15 by thinkyhead