MarlinFirmware/Marlin #7887: ### Compile Error FILAMENT_RUNOUT_SENSOR

android444, 2017-10-07:
### Compile Error
When I compile with `FILAMENT_RUNOUT_SENSOR` I get an error:

k:\Temp\arduino_build_680100\sketch\src\feature\pause.cpp: In function 'void resume_print(const float&, const float&, int8_t)':

k:\Temp\arduino_build_680100\sketch\src\feature\pause.cpp:344:5: error: 'filament_ran_out' was not declared in this scope

     filament_ran_out = false;

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1366300/Marlin.zip)

Tannoo, 2017-10-07:
I ran into the same issue. I couldn't find a use for it. I commented it out so I could print. I have not tested the runout sensor lately.

Roxy-3D, 2017-10-09:
I believe we need to remove the 'static' from its declaration.  Otherwise, nothing else can see it.  I'll merge a fix in the next day or two with this cleaned up.
```cpp
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  bool filament_ran_out = false;
#endif
```

android444, 2017-10-11:
great.

closed 2017-10-11 by android444