MarlinFirmware/Marlin #7888: Unable to Compile for Dual Extruders (Single hot end)

Joshj5hawk, 2017-10-07:
### Compile Error
In file included from C:\Users\SMITHC~1\AppData\Local\Temp\arduino_build_930147\sketch\MarlinConfig.h:30:0,

                 from C:\Users\SMITHC~1\AppData\Local\Temp\arduino_build_930147\sketch\G26_Mesh_Validation_Tool.cpp:27:

Configuration.h:527: error: expected primary-expression before ',' token

 #define DEFAULT_AXIS_STEPS_PER_UNIT   { 100, 100, 400, 95 [,95] }

                                                            ^

C:\Users\SMITHC~1\AppData\Local\Temp\arduino_build_930147\sketch\SanityCheck.h:1299:34: note: in expansion of macro 'DEFAULT_AXIS_STEPS_PER_UNIT'

 constexpr float sanity_arr_1[] = DEFAULT_AXIS_STEPS_PER_UNIT,

                                  ^

Configuration.h:534: error: expected primary-expression before ',' token

 #define DEFAULT_MAX_FEEDRATE          { 400, 400, 8, 50 [, 50] }

                                                          ^

C:\Users\SMITHC~1\AppData\Local\Temp\arduino_build_930147\sketch\SanityCheck.h:1300:34: note: in expansion of macro 'DEFAULT_MAX_FEEDRATE'

                 sanity_arr_2[] = DEFAULT_MAX_FEEDRATE,

                                  ^

Configuration.h:542: error: expected primary-expression before ',' token

 #define DEFAULT_MAX_ACCELERATION      { 2000, 2000, 100, 10000 [,10000] }

                                                                 ^

C:\Users\SMITHC~1\AppData\Local\Temp\arduino_build_930147\sketch\SanityCheck.h:1301:34: note: in expansion of macro 'DEFAULT_MAX_ACCELERATION'

                 sanity_arr_3[] = DEFAULT_MAX_ACCELERATION;

                                  ^

In file included from C:\Users\SMITHC~1\AppData\Local\Temp\arduino_build_930147\sketch\MarlinConfig.h:39:0,

                 from C:\Users\SMITHC~1\AppData\Local\Temp\arduino_build_930147\sketch\G26_Mesh_Validation_Tool.cpp:27:

SanityCheck.h:1302: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT requires 4 (or more) elements.

 static_assert(COUNT(sanity_arr_1) >= XYZE, "DEFAULT_AXIS_STEPS_PER_UNIT requires 4 (or more) elements.");

 ^

SanityCheck.h:1303: error: static assertion failed: DEFAULT_MAX_FEEDRATE requires 4 (or more) elements.

 static_assert(COUNT(sanity_arr_2) >= XYZE, "DEFAULT_MAX_FEEDRATE requires 4 (or more) elements.");

 ^

SanityCheck.h:1304: error: static assertion failed: DEFAULT_MAX_ACCELERATION requires 4 (or more) elements.

 static_assert(COUNT(sanity_arr_3) >= XYZE, "DEFAULT_MAX_ACCELERATION requires 4 (or more) elements.");

 ^

Using library U8glib at version 1.19.1 in folder: C:\Users\SmithCompuptersLLC\Documents\Arduino\libraries\U8glib 
exit status 1
expected primary-expression before ',' token

------------------------------------------------------------
### Bug Report
- Description: Unable to compile for Dual Extruders, single hot end, when attempting to set movement settings for second extruder.
- Expected behaviour: by adding movement settings for second extruder with format X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]] (I'm using 2000, 2000, 100, 10000 [,10000]) I would expect Marlin to compile properly
- Actual behaviour: Marlin does not compile with with the above error.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1366321/Configuration.zip)

AnHardt, 2017-10-07:
Use C/C++ syntax like
```
#define DEFAULT_MAX_ACCELERATION { 2000, 2000, 100, 10000 ,10000 }
```
Omit the [].

Joshj5hawk, 2017-10-07:
I thought that as well, but when I do that, I get a new compile error stating that I have too many elements.

`anityCheck.h:1305: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements.

 static_assert(COUNT(sanity_arr_1) <= XYZE_N, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements.");

 ^

SanityCheck.h:1306: error: static assertion failed: DEFAULT_MAX_FEEDRATE has too many elements.

 static_assert(COUNT(sanity_arr_2) <= XYZE_N, "DEFAULT_MAX_FEEDRATE has too many elements.");

 ^

SanityCheck.h:1307: error: static assertion failed: DEFAULT_MAX_ACCELERATION has too many elements.

 static_assert(COUNT(sanity_arr_3) <= XYZE_N, "DEFAULT_MAX_ACCELERATION has too many elements.");

 ^

Using library U8glib at version 1.19.1 in folder: C:\Users\SmithCompuptersLLC\Documents\Arduino\libraries\U8glib 
exit status 1
static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements.`

`/**
 * Default Axis Steps Per Unit (steps/mm)
 * Override with M92
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 100, 100, 400, 95, 95 }

/**
 * Default Max Feed Rate (mm/s)
 * Override with M203
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_MAX_FEEDRATE          { 400, 400, 8, 50, 50 }

/**
 * Default Max Acceleration (change/s) change = mm/s
 * (Maximum start speed for accelerated moves)
 * Override with M201
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_MAX_ACCELERATION      { 2000, 2000, 100, 10000, 10000 }`

Feels like I'm stuck in a loop here lol

Joshj5hawk, 2017-10-07:
Alright guys figured it out, I'm a goof sometimes.
For those in the future, make sure to define  DISTINCT_E_FACTORS or you'll have a bad time.

AnHardt, 2017-10-07:
If you don't need to use *different* values for the extruders, you can omit `DISTINCT_E_FACTORS`. Then the values for `E0` are used for all extruders.

Joshj5hawk, 2017-10-07:
Yepp I'm learning as I go! I can't get my second extruder motor going so I thought that was what I was missing. Turns out my problem lies somewhere else and I can't quite figure it out yet

closed 2018-02-09 by thinkyhead