MarlinFirmware/Marlin #7891: Support for smoothieware like boards

alexxy, 2017-10-07:
Are there plans to add support for smoothieware like boards?

Or are there some how to to add it? Since LPC1768 already there so seem only pinmap needed?

Roxy-3D, 2017-10-07:
There is nobody doing the work right now to accomplish that.  But it should happen as soon as somebody produces a pin map for it.   I suspect there are going to be a fair number of people with Smoothieware boards that will want to run Marlin.

p3p, 2017-10-07:
Re-ARM is a Smoothieware like board (it runs Smoothieware), there may be missing pins in the current pin map but in theory for any LPC1768 board it should be doable in a normal Marlin pins file (within the constrains of features that currently work with LPC1768), I don't have an LPC1769 board to test with to set the 120MHz clock correctly, (though I could just copy over the smoothieware startup code) .. they may work but we have had reports that LPC1769 has timing issues with the current 100MHz clock.

Spawn32, 2017-10-08:
@alexxy i have a MKS Sbase board that runs smoothie, it's almost identical to the smoothieboard, have hotends / bed heating and temp , TFT 3.2, endstops, etc..working, only thing missing to start testing it is a i2c lib for adjusting the digipots for the stepper drivers.. :)

alexxy, 2017-10-10:
Ok. I currently have [azteeg x5 gt](https://www.panucatt.com/azteeg_X5_GT_reprap_3d_printer_controller_p/ax5gt.htm) board with some set of drivers. And would like to add support for it with Marlin. What is the best way to start? It seesm that this board has Smoothiware pinout.

Spawn32, 2017-10-10:
@alexxy 
I have everything working except the stepper motors on a MKS Sbase, as soon as i can fix the last fault i will send a "pull request" MKS Sbase should work on smoothie and azteeg x5 ...

Roxy-3D, 2017-10-10:
There are two or three things that need to happen.   First, the clock speeds are different on the LCP1769 as compared to the LCP1768.   People are tweaking that right now and that probably will be done before you get the rest of this done.

Next, you probably want to copy the pinmap_re_arm.h file (and call it pinmap_azteeg_x5_gt.h).  In that, you want to to change each entry of the pin_map[] array to match what the azteeg_x5_gt board has.   This array currently maps the atmega-2560's pins to the LPC1768's.  For example, pin4 on the atmega-2560 is SERVO3 and on the Re-ARM it is on port1 - pin18.
```cpp
const pin_data pin_map[] = { // pin map for variable pin function
  {0,3},        //  DIO0   RXD0             A6               J4-4                           AUX-1
  {0,2},        //  DIO1   TXD0             A7               J4-5                           AUX-1
  {1,25},       //  DIO2   X_MAX_PIN                          10K PULLUP TO 3.3v, 1K SERIES
  {1,24},       //  DIO3   X_MIN_PIN                          10K PULLUP TO 3.3v, 1K SERIES
  {1,18},       //  DIO4   SERVO3_PIN        FIL_RUNOUT_PIN   5V output, PWM
  {1,19},       //  DIO5   SERVO2_PIN
  {1,21},       //  DIO6   SERVO1_PIN       J5-1
  {0xFF,0xFF},  //  DIO7   N/C
  {2,7},        //  DIO8   RAMPS_D8_PIN
  {2,4},        //  DIO9   RAMPS_D9_PIN     PWM
  {2,5},        //  DIO10  RAMPS_D10_PIN    PWM
  {1,20},       //  DIO11  SERVO0_PIN
  {2,12},       //  DIO12  PS_ON_PIN
  {4,28},       //  DIO13  LED_PIN
  {1,26},       //  DIO14  Y_MIN_PIN                          10K PULLUP TO 3.3v, 1K SERIES
  {1,27},       //  DIO15  Y_MAX_PIN                          10K PULLUP TO 3.3v, 1K SERIES
  {0,16},       //  DIO16  LCD_PINS_RS      J3-7
  {0xFF,0xFF},  //  DIO17  LCD_PINS_ENABLE   MOSI_PIN(MOSI0)  J3-10                          AUX-3
  {1,29},       //  DIO18  Z_MIN_PIN                          10K PULLUP TO 3.3v, 1K SERIES
  {1,28},       //  DIO19  Z_MAX_PIN                          10K PULLUP TO 3.3v, 1K SERIES
  {0,0},        //  DIO20  SCA
  {0,1},        //  DIO21  SCL
  {0xFF,0xFF},  //  DIO22  N/C
  {0xFF,0xFF},  //  DIO23  LCD_PINS_D4       SCK_PIN(SCLK0)   J3-9                           AUX-3
  {0,4},        //  DIO24  E0_ENABLE_PIN
  {0xFF,0xFF},  //  DIO25  N/C
  {2,0},        //  DIO26  E0_STEP_PIN
  {0xFF,0xFF},  //  DIO27  N/C
  {0,5},        //  DIO28  E0_DIR_PIN
  {0xFF,0xFF},  //  DIO29  N/C
  {4,29},       //  DIO30  E1_ENABLE_PIN
  {3,26},       //  DIO31  BTN_EN1
  {0xFF,0xFF},  //  DIO32  N/C
  {3,25},       //  DIO33  BTN_EN2          J3-4
  {2,13},       //  DIO34  E1_DIR_PIN
  {2,11},       //  DIO35  BTN_ENC          J3-3
  {2,8},        //  DIO36  E1_STEP_PIN
  {1,30},       //  DIO37  BEEPER_PIN       A4                                              not 5V tolerant
  {0,10},       //  DIO38  X_ENABLE_PIN
  {0xFF,0xFF},  //  DIO39  N/C
  {0xFF,0xFF},  //  DIO40  N/C
  {1,22},       //  DIO41  KILL_PIN         J5-4
  {0xFF,0xFF},  //  DIO42  N/C
  {0xFF,0xFF},  //  DIO43  N/C
  {0xFF,0xFF},  //  DIO44  N/C
  {0xFF,0xFF},  //  DIO45  N/C
  {2,3},        //  DIO46  Z_STEP_PIN
  {0xFF,0xFF},  //  DIO47  N/C
  {0,22},       //  DIO48  Z_DIR_PIN
  {1,31},       //  DIO49  SD_DETECT_PIN    A5               J3-1                           not 5V tolerant
  {0,17},       //  DIO50  MISO_PIN(MISO0)                                                   AUX-3
  {0,18},       //  DIO51  MOSI_PIN(MOSI0)   LCD_PINS_ENABLE  J3-10                          AUX-3
  {0,15},       //  DIO52  SCK_PIN(SCLK0)    LCD_PINS_D4      J3-9                           AUX-3
  {1,23},       //  DIO53  SDSS(SSEL0)      J3-5                                             AUX-3
  {2,1},        //  DIO54  X_STEP_PIN
  {0,11},       //  DIO55  X_DIR_PIN
  {0,19},       //  DIO56  Y_ENABLE_PIN
  {0,27},       //  DIO57                   AUX-1             open collector
  {0,28},       //  DIO58                   AUX-1             open collector
  {2,6},        //  DIO59  LCD_A0           J3-8                                             AUX-2
  {2,2},        //  DIO60  Y_STEP_PIN
  {0,20},       //  DIO61  Y_DIR_PIN
  {0,21},       //  DIO62  Z_ENABLE_PIN
  {0,26},       //  DIO63  AUX-2             A3               J5-3                           AUX-2
  {0xFF,0xFF},  //  DIO64  N/C
  {0xFF,0xFF},  //  DIO65  N/C
  {0xFF,0xFF},  //  DIO66  N/C
  {0,23},       //  DIO67  TEMP_0_PIN       A0
  {0,24},       //  DIO68  TEMP_BED_PIN     A1
  {0,25},       //  DIO69  TEMP_1_PIN       A2
  {1,16},       //  DIO70                   J12-3             ENET_MOC
  {1,17},       //  DIO71                   J12-4             ENET_MDIO
  {1,15},       //  DIO72                   J12-5             REF_CLK
  {1,14},       //  DIO73                   J12-6             ENET_RX_ER
  {1,9},        //  DIO74                   J12-7             ENET_RXD0
  {1,10},       //  DIO75                   J12-8             ENET_RXD1
  {1,8},        //  DIO76                   J12-9             ENET_CRS
  {1,4},        //  DIO77                   J12-10            ENET_TX_EN
  {1,0},        //  DIO78                   J12-11            ENET_TXD0
  {1,1},        //  DIO79                   J12-12            ENET_TXD1
//  {0,6},        //  DIO80  SD-CS            on board SD card
//  {0,7},        //  DIO81  SD-SCK           on board SD card
//  {0,8},        //  DIO82  SD-MISO          on board SD card
//  {0,9},        //  DIO83  SD-MOSI          on board SD card
};
```
And then lastly...  Get the compiler to pull in your table instead of the re-arm table.

I'm sure there will be something else that causes problems, but that is the big stuff.

alexxy, 2017-10-10:
Ghm... How this table do mapping? For example there is pin table for smoothie like boards
http://smoothieware.org/lpc1769-pin-usage

p3p, 2017-10-10:
@alexxy You shouldn't need to change much if anything (like I said Re-ARM is a Smoothieware board), I think a few minor things may need added (@Spawn32 is looking into it), the configuration for your board pinout would be done in the pins files like normal just use the DIO## pin number associated the the LPC1768 pin. I do plan to simplify this (if I'm not beaten to it) allowing you to use LPC1768 pin numbering directly from the pins file.

Roxy-3D, 2017-10-10:
Well...   Really, that pin table should go away.   It is there to simplify the migration from an AVR based RAMPS board to ARM.   Pin 0 is the first entry.  That maps to port 0 pin 3 on the LPC.   Pin 1 in the next entry.   That maps to port 0 pin 2 on the LPC.   And so on...

We have been having some discussions on how to change things for generic LPC processors.  You can read some of the thoughts here:  https://github.com/MarlinFirmware/Marlin/issues/7076#issuecomment-334883022

That table adds an extra level of indirection.   For Re-ARM having a RAMPS board it made sense to do that.   It kept things simple.   But a Smoothieboard isn't going to need to map RAMPS locations to the LPC.  It should just use the LPC port and pin numbers.

But for now...   The simplest path is to look up each important pin (X_MAX_PIN for example) and fill in the Smoothieboard port and pin numbers (in the 3rd table entry in this example) and pull in the RAMPS 1.4 pins file.   In the RAMPS 1.4 pin file it will have X_MAX_PIN defined as 2 which is going to be the 3rd entry into the table (starting from zero).

alexxy, 2017-10-10:
Ok. using LPC port and pin numbers looks better to me. But is it possible now? or its better to use DIO pins?

So can this pin map be avoided?

Roxy-3D, 2017-10-10:
Yes the pin map can be avoided.   But we don't have consensus on the correct path forward.   

But to p3p's point....   The Re-ARM setup is pretty close to the SmoothieBoard mapping.   It maybe you need to add some extra definations for items the SmoothieBoard has that Re-ARM & RAMPS do not have.

closed 2017-11-15 by thinkyhead