MarlinFirmware/Marlin #7895: Extruder Nozzle Switching on Z-axis? (Watch Videos)

kcheeeung, 2017-10-08:
Hey guys, can you all look at these videos? I am trying to build a 3D printer than can extrude chocolate of multiple types. I want to have a syringe mechanism similar to these videos to prevent oozing.

Please look carefully at the syringe heads and each one independently moves up/down when the extruder is chosen. What kind of feature is this? How is this achieved in the firmware? It is very likely they use some variant of the Marlin firmware. If not, what firmware does have this mechanism.

https://youtu.be/xeHbGsw00-g?t=3s
https://youtu.be/KuO0tdF3ySA?t=43s
https://youtu.be/W1G12ZQo_O4?t=23s

empakoso, 2017-10-08:
Don’t see anything that can’t be done with Marlin.  I’ve been looking at making silicon gaskets using this:
https://www.structur3d.io/discov3ry-products/discov3ry
Or making my own using this:
http://www.appropedia.org/Open-source_syringe_pump

kcheeeung, 2017-10-08:
I know it is very well possible. I am more interested in how each "extruder" is independently moving up and down on the Z axis on on tool change.

Do you have any ideas?

empakoso, 2017-10-08:
Control using a servo mechanism.  Rack and pinion design.

kcheeeung, 2017-10-09:
Very interesting. Would you happen to know about the firmware side of things? Not sure if I've ever understood the switching_extruder / switching_nozzle correctly. Would it be implemented with this?

```
// A dual extruder that uses a single stepper motor
//#define SWITCHING_EXTRUDER
#if ENABLED(SWITCHING_EXTRUDER)
  #define SWITCHING_EXTRUDER_SERVO_NR 0
  #define SWITCHING_EXTRUDER_SERVO_ANGLES { 0, 90 } // Angles for E0, E1[, E2, E3]
  #if EXTRUDERS > 3
    #define SWITCHING_EXTRUDER_E23_SERVO_NR 1
  #endif
#endif

// A dual-nozzle that uses a servomotor to raise/lower one of the nozzles
//#define SWITCHING_NOZZLE
#if ENABLED(SWITCHING_NOZZLE)
  #define SWITCHING_NOZZLE_SERVO_NR 0
  #define SWITCHING_NOZZLE_SERVO_ANGLES { 0, 90 }   // Angles for E0, E1
  //#define HOTEND_OFFSET_Z { 0.0, 0.0 }
#endif
```

kcheeeung, 2017-10-15:
@Bob-the-Kuhn I just recently saw you made some posts and are active again. I hope you're doing well. Would you have any ideas about this?

For the first video, I confirmed that the Biobot1 is using the Marlin firmware, and what puzzles me is what mechanism they use do achieve that independent "extruder" switching. Same for the other videos demonstrating Aether1 and Cellink.

closed 2017-11-08 by kcheeeung

Roxy-3D, 2017-11-08:
> For the first video, I confirmed that the Biobot1 is using the Marlin firmware, and what puzzles me is what mechanism they use do achieve that independent "extruder" switching.

I didn't watch the video's.  But the independent extruder switching might be handled with a T0 or T1 or T2 command.