MarlinFirmware/Marlin #7897: [BR] LCD not working after firmware flash/update

Tininno, 2017-10-08:
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/1366662/Configuration_adv.zip)

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1366661/Configuration.zip)

- Description: Firmware was working, flashed new copy of firmware, LCD no longer works, and uncertain whether or not printer itself is functioning properly
- Expected behaviour: LCD powering on properly, printer running as normal
- Actual behaviour: LCD has backlight, no text displayed (was working previously)
- Steps to reproduce:
---------
  (The board is an older RAMBo.)
Downloaded legacy version of Arduino (1.0.5)
Downloaded legacy version of Marlin (1.0 I believe, this board is 4+ years old mind you)
Downloaded U8Glib (Using reprap discount Full Graphics LCD)
Downloaded RAMBo hardware file
Moved RAMBo hardware file into hardware folder
Moved U8Glib into libraries folder
Launch Arduino
Uncomment the #define for the discount Full graphics LCD
Verify the file (mine came up fine)
Make sure RAMBo board is selected
Make sure COM is selected
Select AVRISP MKII programmer
Upload the file
-------

Keep in mind, could be something I did wrong here. I've only been printing about 6 months, this printer I got used, and it's a great printer (Mendelmax 2.0 beta). But, the downside being I do not have the firmware the seller originally had installed, I've spent sometime digging the archives of the web trying to collect everything I need to get the firmware set up, but this issue I'm having is the culmination of 8 or so hours of research and digging. 

Below is a copy of my config.h and config_adv.h files, I haven't checked at this point if these copies actually contain the line I uncommented, however as far as changes go, that's the only thing I've modified aside from adding the hardware as well as the U8Glib library.

Any help, advice, insight, experience, insults, criticism, is appreciated at this point. I feel like a clueless idiot, I'm trying my best to learn but as you can see it's not going without hitches. Thank you :)

Roxy-3D, 2017-10-08:
If it was working....  And you flashed a new version of the firmware and it quit working...  My bet is this is wrong:

Uncomment the #define for the discount Full graphics LCD

Do you have your old configuration.h file so we can take a look?

Tininno, 2017-10-08:
@Roxy-3D thank you for the quick reply, it was definitely unexpected. If I had a copy of it (I really wish I did) I'd be so happy to provide it, but being that I picked it up used, it was already flashed and setup, but I have reason to believe something was done incorrectly, as none of the endstops were working, which was the reason for reflashing in the first place. Basically all I've done is removed the // in front of #define REPRAP_FULL_GRAPHIC_SMART_CONTROLLER. The U8Glib libraries are showing in Arduino, so it's in there, and I was able to select the RAMBo as the board as well. So I'm thinking that possibly there's something within U8Glib I've done wrong here, or maybe there's something else that I've overlooked within my config file. I've reached out to the person I purchased it from, however I've yet to hear anything back on their config file, or whether or not they even still have it. Any advice regarding what I may have missed?

Roxy-3D, 2017-10-08:
If the firmware compiles OK and uploads...   Probably the u8Glib library is OK.  (Not guarinteed, but most likely.)   What makes you think you have a REPRAP_FULL_GRAPHIC_SMART_CONTROLLER?     If it used to work, and you haven't moved any of the cables, it would seem the LCD Panel should light up and work if you have the LCD Panel type properly defined.

Can you post a detailed picture of your LCD Panel board?

Tininno, 2017-10-08:
@Roxy-3D YES I most definitely will :). Also what leads me to believe that I have that one is due to it having the signature white back, it matches the general look of that display. I get home this evening and will definitely post pics!

fiveangle, 2017-10-08:
Since there's no guarentee the old Marlin code you're trying to compile is the same base as the FW that was loaded, I would ditch the old sofware and try to bring it up on current marlin/arduino IDE since you've got nothing to loose.  If it doens't work, then continue your current troublshooting path.  RAMBo is even a supported board in PlatformIO so whatever older-than-dirt RAMBo HW library you have has surely been superceeded with something more robust, or at least identical functionality.

Tininno, 2017-10-08:
@fiveangle thank you much for the reply. I think you might be right about trying a new version. Legacy Arduino is testy anyway and I know a small amount more about newer versions of Marlin, it's also a bit more forgiving. I guess at this point I do not even know which library I'd use for this board, it's actually pretty difficult finding any current hardware libraries for RAMBo. I think it may be easier to try using the RAMBo hardware library I have now, with a newer version of Marlin, which U8Glib isn't even needed on any longer. My biggest concern of all is that using a newer version of Marlin, that it would have the appropriate pins to correspond to this older board. Because I believe pins 70 and 71 other boards such as the RAMPS 1.4 is lacking. Do you happen to have any insight or reassurances going forward into something like this? Do you know if there is a version of Marlin currently that is gauged for the RAMBo, or is it as easy as commenting in a #define for the proper board and // all the others?

Thanks again for taking the time to reply, both of you.

fiveangle, 2017-10-09:
"reassurances" ?  You might want to [look here](http://www.prusa3d.com/) ;)

One thing I can assure you of: there will be more people here willing to help you resolve any incorrect pins in current marlin for your board, than there will be those willing to help you debug a random collection of 4 year old software.  The former is useful for others besides you, the latter is a zero-sum game. * *shrug* *

I suggest (again) you download the current Marlin, configure for your board as best you can, and whatever problems you can't resolve on your own or via google, post back here and we can problem help narrow in on the problem.

Tannoo, 2017-10-09:
Also, if you know the make and model of your printer, there might be an example config set ready to try.

Tininno, 2017-10-09:
@fiveangle I definitely agree. I'll work on it and post back if there's any problems

@tannoo it's an older printer, mendelmax 2.0 beta, that's kinda where the struggle comes in. I've got a robo r1+ and a few a8's. I've messed with firmware on a duet wifi, working on some dual extruder setups, but working with older printers is a little out my league hahaha. If you have any insight on that printer, let me know :)

Tannoo, 2017-10-09:
Google found this:

https://github.com/UBCRapid/MendelMax2.0/tree/master/Firmware/Marlin%20MendelMax%202.0/Marlin/example_configurations/delta

Tannoo, 2017-10-09:
That's not going to be a drand-n-drop solution. But, can get some machine data from it.

Tininno, 2017-10-09:
@tannoo I think that's a good start. I think I may have been looking at that repository previously. As you can see though, most of the files we're working with here are over 3 years old, so that's kinda the catch 22. I'm gonna give this a shot. Possibly use the old configs with a new version of Marlin? Flash using the RAMBo without U8Glib? Not sure if possible but we'll see. If not tbh I'll probably just pick up a new board and mess with the motor calibration, most of that is likely done in the config anyway, I remember most of the values for the steps so that part shouldn't be too difficult, plus I know how to calibrate. But as you can tell firmware isn't a strong point for me. Thanks for the help and the insight, I'm gonna get to work on this.

Roxy-3D, 2017-10-09:
> Possibly use the old configs with a new version of Marlin? 

Yes...  Pretty much, that will work.   But there is one extra step.   You need to get a visual diff program like ExamDiff Pro or NotePad++ so you can compare the old Configuration.h and Configuration_adv.h file to the new ones.   Some of the options have changed.   Just dropping an old configuration.h file set into the code won't work.

But if you have a visual diff program to compare things...  You should be able to get your settings moved over to the new files in 15 or 20 minutes even if you move very slowly and  carefully.

Tininno, 2017-10-09:
@roxy-3d That makes sense. It'll take some time I'm sure but it'd probably be worth it. I'll mess around with it a bit and hopefully with enough time and effort I'll be able to get it going again :) thank you

closed 2017-11-15 by thinkyhead