MarlinFirmware/Marlin #7901: Thermistor table 13

iain1965, 2017-10-08:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

iain1965, 2017-10-08:
When trying to use high temperature thermistor with thermistor table 13 get the following compile errors:

In file included from sketch\thermistortables.h:80:0,

                 from sketch\temperature.h:30,

                 from sketch\Marlin_main.cpp:250:

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '3.20640015e+2' from 'double' to 'const short int' inside { } [-Wnarrowing]

 };

 ^

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '3.71040009e+2' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '4.27359985e+2' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '4.99679993e+2' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '5.84320007e+2' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '6.84e+2' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '8.10880005e+2' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '9.6352002e+2' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '1.15247998e+3' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '1.38943994e+3' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '1.64464001e+3' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '1.99135999e+3' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '2.41632007e+3' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '2.92576001e+3' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '3.53152002e+3' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '5.07135986e+3' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '7.15472021e+3' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '9.44975977e+3' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '1.179696e+4' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '1.37243203e+4' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '1.50323203e+4' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '1.57764805e+4' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '1.61392002e+4' from 'double' to 'const short int' inside { } [-Wnarrowing]

In file included from sketch\thermistortables.h:80:0,

                 from sketch\temperature.h:30,

                 from sketch\configuration_store.cpp:183:

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '3.20640015e+2' from 'double' to 'const short int' inside { } [-Wnarrowing]

 };

 ^

[... 357 lines cut ...]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '2.41632007e+3' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '2.92576001e+3' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '3.53152002e+3' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '5.07135986e+3' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '7.15472021e+3' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '9.44975977e+3' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '1.179696e+4' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '1.37243203e+4' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '1.50323203e+4' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '1.57764805e+4' from 'double' to 'const short int' inside { } [-Wnarrowing]

sketch\thermistortable_13.h:48:1: warning: narrowing conversion of '1.61392002e+4' from 'double' to 'const short int' inside { } [-Wnarrowing]

Sketch uses 137,940 bytes (54%) of program storage space. Maximum is 253,952 bytes.
Global variables use 4,768 bytes (58%) of dynamic memory, leaving 3,424 bytes for local variables. Maximum is 8,192 bytes.

GMagician, 2017-10-08:
That's normal. first field in thermistor table is an integer value, not a float. Table is wrong.
A simple fix can be round values to nearest integer but I can't assure that current values are correct

GMagician, 2017-10-09:
Maybe, to remove all warnings, you can try to write (short)(value * OVERSAMPLENR) for all "first" column entries
Eg. for first entry is: (short)(20.04 * OVERSAMPLENR)

But I don't know if 20.04 is a correct value

boelle, 2019-02-19:
@iain1965 did you solve your "issue" ?

closed 2019-02-20 by thinkyhead