MarlinFirmware/Marlin #7902: [bf2.0.x] BILINEAR issues double-touch on some probe points, normal on others

fiveangle, 2017-10-08:
### Bug Report
- Description: During G29 BILINEAR issues double-touch on some probe points, normal on others
- Expected behaviour: Normal single probes
- Actual behaviour: Issues double-touch probes on some points:
https://youtu.be/bpuzHJtRfeo
- Steps to reproduce:
  - Compile with BILINEAR and do not enable DOUBLETOUCH
  - G8;G29
  - Observe double-touch on some points

https://github.com/fiveangle/Marlin/blob/bf20-pb_simple_metal/Marlin/Configuration.h
https://github.com/fiveangle/Marlin/blob/bf20-pb_simple_metal/Marlin/Configuration_adv.h

```Send: G28
Recv: echo:G28
Recv: echo:busy: processing
Recv: X:51.00 Y:76.00 Z:3.00 E:0.00 Count X:1020 Y:1520 Z:1521
Recv: ok P15 B3
[...]
Recv: echo:M105
[...]
Send: G29
Recv: echo:G29
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +0.138 +0.099 +0.067 +0.067 +0.075
Recv:  1 +0.028 -0.028 -0.071 -0.047 -0.039
Recv:  2 -0.006 -0.024 -0.049 -0.053 -0.069
Recv:  3 +0.059 +0.032 -0.004 -0.030 -0.075
Recv:  4 +0.036 -0.004 -0.020 -0.041 -0.061
Recv: 
Recv: X:124.00 Y:152.00 Z:3.05 E:0.00 Count X:2480 Y:3040 Z:1521
Recv: ok P15 B3
```
No time now, but will enable debug leveling and try to dig into this later in the week.

-=dave

thinkyhead, 2017-11-15:
Has this problem remained in the latest `bugfix-1.1.x` branch? If so, please re-open this issue.

fiveangle, 2017-12-03:
No change as of https://github.com/MarlinFirmware/Marlin/commit/05f8e98c4ab1f96ddb0af0dcc51af838ec9b0c30

Can't re-open as I did not close it.

Roxy-3D, 2017-12-03:
It's reopened for you...

fiveangle, 2017-12-03:
Thx.  Tried bisecting to narrow it down, but too many compile errors on Printrboard.  Have a feeling this will take a while 😝

Roxy-3D, 2017-12-03:
Does anybody want to hear my feelings towards the Printrboard people?    If so...  Let's start a new thread so we don't screw this one up.

fiveangle, 2017-12-03:
I thought you liked me @Roxy-3D  ? 💋 😉 

Unfortunately, I see little evidence to support this being Printrboard-specific.  I it's triggered via configuration-specific (it's just that I have so much custom config, even that is difficult to narrow down).

Roxy-3D, 2017-12-03:
> Does anybody want to hear my feelings towards the Printrboard people? 

I guess I was sloppy with my English.   I should have worded that more like "Does anybody want to hear my feelings towards the Printrbot company and its lack of support for the Printrboard?"

samibaskir, 2017-12-13:
It is not related to Printrboard. Also it is not a bug either. Probably expected behaviour. Let me explain;

@fiveangle did you set your "#define Z_CLEARANCE_DEPLOY_PROBE" to 0?

Set it to 1 and it will return to normal.

I did try to set it to 0 because I have a fixed probe, but later realized that it says only integer values >= 1 are applicable. This is probably required due to some equation about Z clearance calculation. Can this behaviour be improved? Probably yes, but I don't think this counts as a bug since there is a word of caution already in place.

fiveangle, 2017-12-13:
I think the only person who thinks this might be related to Printrboard is Roxy, and that's more a running joke tongue-in-cheek than the truth :)

From my first post I state my suspicion of the `Z_CLEARANCE_DEPLOY_PROBE = 0 `being the issue.  But there is no reason it should not be supported and for behavior to change.  I do see the detail on in the Config.h, but from SanityCheck.h:

https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/SanityCheck.h#L693-L701 : 
``` C++
  #ifndef Z_CLEARANCE_DEPLOY_PROBE
    #error "You must define Z_CLEARANCE_DEPLOY_PROBE in your configuration."
  #elif !defined(Z_CLEARANCE_BETWEEN_PROBES)
    #error "You must define Z_CLEARANCE_BETWEEN_PROBES in your configuration."
  #elif Z_CLEARANCE_DEPLOY_PROBE < 0
    #error "Probes need Z_CLEARANCE_DEPLOY_PROBE >= 0."
  #elif Z_CLEARANCE_BETWEEN_PROBES < 0
    #error "Probes need Z_CLEARANCE_BETWEEN_PROBES >= 0."
  #endif
```

## "Probes need Z_CLEARANCE_DEPLOY_PROBE >= 0."

I will confirm again that this behavior was not present in 1.1.4 and is present in 1.1.5 (since managing checkout, compile, shuffling Marlin.hex / firmware.hex to/from octoprint, dfu-util, etc there is always the possibility for human error).

But it's definitely not WAD.  The issue must be fixed, either by forcing `Z_CLEARANCE_DEPLOY_PROBE ` to be within the required range via SantityCheck.h (the hack solution) or determining where the indeterminant behavior is being injected when `Z_CLEARANCE_DEPLOY_PROBE` becomes 0.

Most of the commits leading up to 1.1.5 fail to compile for Printrboard due to unrelated issues and I'm not looking too forward to figuring out which ones require backporting just so I can get it to compile to then do a bisect search through the 114-115 changes to find the culprit.

But an easy immediate solution would be to submit the "hack" solution for now, and keep this open for the long-term "todo" fix.

-=dave

fiveangle, 2018-01-24:
No longer interested in figuring out where it broke.  Closing...

closed 2018-01-24 by fiveangle