MarlinFirmware/Marlin #7905: Marlin 1.1.5 Does not respond to feedrate change or acceleration in Gcode

1Ghasthunter1, 2017-10-09:
- Description: The printer is stuck at the default Feedrate, 100%. You can turn the knob on the LCD display and also(in pronterface) set the speed, and the feedrate on the number on the LCD will increase to whatever you set it to, however the actual print job is not affected, and nothing speeds up. This also happens with accelaration; as when using Cura, set the Outer and Inner wall accelaration to about 350, and try to print the accelaration is the same as the default accelaration, 3000. Also do recall that I cannot change the accelaration in the Tune Menu during a print, as it has the same issue with the feedrate(the numbers change, accel does not).
- Expected behaviour: The print will slow down and increase with the feedrate, and it will respond to accelaration in the gcode 
- Actual behaviour: ^
- Steps to reproduce:
Use my firmware that I will attach(I downloaded this from marlinfw.org, and the feedrate never worked on marlin 1.1.5. Then start a print, get it going(as in it begins the print correctly, the filament is flowing, the print is sticking, and try to increase and decrease the feedrate. Nothing SHOULD happen. Then, try making the Gcode with accelaration that is not 3000, such as 350 or 500, something that is easily noticable. Nothing should change

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/1367547/Configs.zip)

psavva, 2017-10-12:
Ensure your maximum velocity for your extruder is set high enough in order to allow the feedrate to have an effect.

vilsed, 2017-10-25:
I'm having the exact same problem with 1.1.6 on a delta machine. Previously I had 1.1.0 and everything worked fine

vilsed, 2017-10-25:
#7948

closed 2018-02-09 by thinkyhead