MarlinFirmware/Marlin #7906: UBL G29 P1 U does full bed probe

Maglin, 2017-10-09:
### Bug Report
- Description: --- `G29 P1 U` not working
- Expected behaviour: --- Probe perimeter of the bed
- Actual behaviour: --- Probes the entire bed
- Steps to reproduce: execute a `G29 P1 U` or `G29 P1 T U`

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1367814/Marlin.zip)

Roxy-3D, 2017-10-09:
It will still probe the entire bed if you let it go long enough.  But the U option should force it to start at the outside perimeter and work its way inward.

Maglin, 2017-10-09:
Ahhh..  I just misunderstood the functionality of the U function.  Thank you.

Roxy-3D, 2017-10-09:
Yeah...  Pretty much...   The 'U' option is just a quick and dirty way to get map numbers that are suitable for the physical bed leveling.   And the user gets to control how many points are acquired (by stopping the probing when there are enough).

closed 2017-10-09 by Roxy-3D