MarlinFirmware/Marlin #7907: G38.2 does not stop on Contact with Z_Min

Edgar48, 2017-10-09:
### Bug Report
- Description: Im using Marlin 1.1.3 also tried Marlin1.1.5 ; G38.2 Z does not stop on Contact with Z_MIN-G28 and G30 are working correct--
- Expected behaviour: G38.2 would stop on Contact
- Actual behaviour: >>>M502
SENDING:M502
echo:Hardcoded Default Settings Loaded
>>>M500
SENDING:M500
echo:Settings Stored (481 bytes; crc -31085)
>>>M111 S247
SENDING:M111 S247
echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:G91
echo:G1 Z10.0 F3000
echo:G90
echo:M114
echo:M105
echo:M105
>>>G38.2 Z
SENDING:G38.2 Z
echo:G38.2 Z
  current_position=(0.00, 0.00, 10.00) : setup_for_endstop_or_probe_move
echo:busy: processing
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
Error:Failed to reach target

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1367847/Configuration.zip)

[ERROR] Error:Failed to reach target

  current_position=(0.00, 0.00, 0.00) : clean_up_after_endstop_or_probe_move
echo:M105

Tried to follow simelar Problemrepoert but it seem that all these bugfixes already have been implemented.

Find ZIP of `Configuration.h` and `Configuration_adv.h`  here.

fiveangle, 2017-10-11:
I'm not sure but are endstops required to be enabled for G38 to work ?

Have you tried after issuing M120 immediately prior ?

Also, once you reach target, endstop should be already triggered, so what is output of M119 at that point ?

fiveangle, 2017-10-11:
Also, this wording seems like it can be slightly confusing if, like me, one isn't familar with the feature. 
G38.2 attempts to move to the position entered as parameters to the command and expects to trigger an endstop before reaching that end position.  However, the error message presented if it reaches end-position without triggering the endstop is:

`SERIAL_ERRORLNPGM("Failed to reach target");`

A clearer error might be "_Target not detected_" since with "_Failed to reach target_" once could conflate end-position (known and entered) with the target location (unknown which we are hoping to find), or otherwise question if they understand how the feature works (which is what I did).

But I digress...   :smile:

Edgar48, 2017-10-12:
Thank you for your reply fiveangle. I tried M120 prio to G38.2 but didnt help. The M119 shows that Z_Min is trigered on contact, but z axis continues to move after contact  until Z=0 which is the target.
>>>M120
SENDING:M120
>>>G38.2 Z
SENDING:G38.2 Z
echo:busy: processing
Error:Failed to reach target
[ERROR] Error:Failed to reach target

>>>M119
SENDING:M119
Reporting endstop status
x_min: open
y_min: open
z_min: TRIGGERED

Edgar48, 2017-10-12:
I'm currently trying to dig my way trough the code but I cant find the definition of the funktion  endstops.hit_on_purpose(). so i'm stuck

AnHardt, 2017-10-12:
https://github.com/MarlinFirmware/Marlin/blob/99d18ecd0cc9be27cfc32f37cc7d1b476b4139a1/Marlin/endstops.h#L78

Edgar48, 2017-10-13:
Thank you AnHardt.

I now found following:
In Routine
// Check endstops - Called from ISR!
void Endstops::update() {
    
 
is a fork G38  and I found, that in my case the expression PIN_EXISTS(Z_MN_PROBE) is not giving True
  
  //hier Test 
  #if ENABLED(G38_PROBE_TARGET)    && !(CORE_IS_XY || CORE_IS_XZ)  //&& PIN_EXISTS(Z_MIN_PROBE)
    // If G38 command is active check Z_MIN_PROBE for ALL movement
    
    if (G38_move) {
      UPDATE_ENDSTOP_BIT(Z, MIN_PROBE);
      
   SERIAL_ERRORLNPGM("update: G38_move reached"); 

      if (TEST_ENDSTOP(_ENDSTOP(Z, MIN_PROBE))) {
        if      (stepper.current_block->steps[_AXIS(X)] > 0) { _ENDSTOP_HIT(X, MIN); stepper.endstop_triggered(_AXIS(X)); }
        else if (stepper.current_block->steps[_AXIS(Y)] > 0) { _ENDSTOP_HIT(Y, MIN); stepper.endstop_triggered(_AXIS(Y)); }
        else if (stepper.current_block->steps[_AXIS(Z)] > 0) { _ENDSTOP_HIT(Z, MIN); stepper.endstop_triggered(_AXIS(Z)); }
        G38_endstop_hit = true;
      }
    }
  #endif    ........

I also found PIN_EXISTS(Z_MN_PROBE) means by this definition:

#define PIN_EXISTS(PN) (defined(PN ##_PIN) && PN ##_PIN >= 0)

Z_MIN_PROBE_PIN has to be defined and should be positiv or zero
There are a lot of definitions for Z_MIN_PROBE_PIN obviously depending on the Hardware used

My Motherboard is 
 #define MOTHERBOARD BOARD_RAMPS_14_EFB

My Procesor is Arduino Mega2560

Trying to find out how where "my" #define Z_MIN_PROBE_PIN is to be found.

Bob-the-Kuhn, 2017-10-14:
Sorry - been busy in the 32 bit branch.

This is the third or fourth time that the G38 logic has been broken. 

I spent a couple of hours looking at all the logic and ended up with a headache.

Hopefully I'll have a fix tested tomorrow.

Bob-the-Kuhn, 2017-10-14:
Question - why do you have both UBL enabled and G38 enabled?

My understanding is that the bed leveling systems don't take into account the offsets that are usually generated when using G38.  

I could be out of date.  I haven't been paying much attention to Marlin 1.x.

fiveangle, 2017-10-14:
Yeah, I looked at was like 🤷‍♂️

Edgar48, 2017-10-14:
Im using my self-designed 3D Printer also as pcb board cnc and CNC for foam. I have to touch the surface of the workpice and I thought g38.2 z would be helpfull. maybe I did not understand UBL completely but I find it very complicatated to define the UBL testspots just to touch the surface. Is there a way of using UBL defining manualy the spots to test by going there in xy direction?

Bob-the-Kuhn, 2017-10-14:
Take a look at Issue #7972.  Fixing the G38 logic the correct way is a lot of work.  Looks like it'll be a while before I get comments on my proposal and have it ready for prime time.

I think your change is a good interim solution as long as the Z_MIN endstop is setup as ENDSTOP_INVERTING true (the way it is in your config files).

----

Follow Issue #7970 on using bed leveling and G38.  I expect that others have already come up with ways to use G38 with bed leveling.

Edgar48, 2017-10-14:
@fiveangle : I didnt quiete understand your comment. What you are tring to say? As i'm new into Marlin and also new into Github maybe I'm not contributing the proper way. I just tried to explain my thoughts and show how I aproach the problem so that the experienced ones can give me hints where i'm thinking wrong or give me hints on the Structure of marlin. I ´f i did not behave properly I want to excuse myself and would be more than happy if you or someone could give a starter like me a hint what and how I can do better. I find it marvelous what the conributers to Marlin have created and I realy want to contribute with my small skills as good as possible.

fiveangle, 2017-10-14:
I looked at [the code]…

fiveangle, 2017-10-14:
You explained yourself just fine (which is definitely not always the norm ! : )

printme3daachen, 2020-04-12:
Hey, are there any news on this topic? I'm trying to use G38.2 as well but have the same issue in Marlin 2.0.5.2. As I'm pretty new to Marlin Code I don't fully understand what the result of this is. I've seen some old Issues here like this one but somehow it still isn't working. 
I read a lot of comments using some kind of dirty way to make it running. If G38 still is known to be broken could you give a more detailed explanation on what to do to get this running? 

I really appreciate all of the amazing work that was done to get Marlin to what it is today!

fiveangle, 2020-04-13:
This issue is closed so I'd open a new issue if you want it looked at.