MarlinFirmware/Marlin #7910: Not 5V on inductive sensor

Jean-Fred-CEC, 2017-10-09:
Hi everybody, 

I have an issue with upgrading my ZProbe : I bought an inductive LJ12A3-4-Z/BX sensor that I chequed apart. It gives 11,5v on the signal wire when I connect to my power supply and approx 0.5V when it dettects metal.
I also bought an LM7805cv voltage regulator that gives 5v when I feed it with 11,5v.
Everything is good so far, but when I connect the both, to my ramps, the LM7805cv output is only 1.35v...
I checked all the wiring ten times but everything looks good, is it normal that I have a lower voltage than 5v when I connect the sensor and voltage regulator to my ramps?

Best regards, 

Jean Fred

Spawn32, 2017-10-09:
Dont think you can "drive" a LM7805 from the signal output from the probe, but i might be wrong :)

Jean-Fred-CEC, 2017-10-09:
OK thanks, I will test with the resistors .
I saw some printers that had this modification, but  I am not an electronics specialist...
For undestanding, do you know why I have a different voltage on the LM7805 with an without the sensor? Ground is common but there is no intensity.

Thanks for your repply!

Tannoo, 2017-10-10:
LJ12A3-4-Z/BX is an NPN sensor. There is no positive V out from it.

Power it from 12v and the output is negative and goes directly to the z-axis limit input.

Tannoo, 2017-10-10:
The LM7805 is not needed for that probe.

Jean-Fred-CEC, 2017-10-10:
Are you sure Tannoo? I'm affraid it will fry my ramps with 11.5v instead of 5v...

Tannoo, 2017-10-10:
An NPN will not. Mine is the same way.
If it was a PNP, yes.

Tannoo, 2017-10-10:
If you are testing the output of the NPN sensor with a meter while it's disconnected from any load... yes, you will see a "floating" voltage.

Do some more research on NPN sensors and you will see.

Tannoo, 2017-10-10:
If it concerns you that much, you can throw in a zener diode in series to prevent overvoltage of the input.

Tannoo, 2017-10-10:
I have had a few of these sensors go bad... but input is still okay.
I will admit that when I first start doing probe leveling, I temporarily wired up a sensor and had it working. I was printing multiple things and telling myself to dress up and terminate the wires correctly. Well... I kept putting that off and one day the wires got moved and shorted the 12v+ to the output. That took out the sensor AND 2 inputs.
I used some other unused input until I replaced the board.

Jean-Fred-CEC, 2017-10-10:
OK, I checked the sensor internal resistance: 10 KOhms... may be this is why the regulator doesn't works?
So I made, with some resistances I had, a 7KOhms resistance in order to make a voltage divider, it works! (4,8v)
now when I enter M119 I have the sensor trigered.

Thank you all!

closed 2017-10-10 by Jean-Fred-CEC

Spawn32, 2017-10-10:
LJ12A3-4-Z/BX is blue tip'd ?
I have 3 of those, from 8mm to 18mm, all og them outputs 12v volt on the signal pin, be very careful, it will kill your board, wire it up an trigger it, then read the voltage before you do anything :)

Jean-Fred-CEC, 2017-10-10:
Yes, that's what I did.
With the resistor trick, the opened voltage is 4.8v, and 0.5v triggered.

Spawn32, 2017-10-10:
correct, that's what i have, without resistor you would have gotten 12v straight out on the signal wire :/

Tannoo, 2017-10-10:
I will say that mine has been working just fine for the last 2 years.

Tannoo, 2017-10-10:
No divder, no regulator. Maybe I am just one of the lucky ones.

Spawn32, 2017-10-10:
That is very strange Tanno, is this a blue tip ? but as long as it work's :)  no if we could adapt the temp compensation that Prusa has put in to their software it would be great, all we need to do is to use some kapton tap to mount a semic temp sensor to the metal housing of the probe and we could use the prusa routines, this probe drift under temperature...

Tannoo, 2017-10-10:
I have some blue tipped ones, but they don't see my bed very well. I have been using oranged tipped NPNs that are adjustable range. These see my bed much better and even 4mm away.

Tannoo, 2017-10-10:
I have also tried the IR pi sensor that worked better than the blue tipped NPNs.... but still not consistantly.

Spawn32, 2017-10-10:
the blue tip 12mm has a detection range of 4mm and seems very good, the 18mm has 8mm detection but is very big :) and the 8mm has only 2mm, to little :/

Tannoo, 2017-10-10:
I know... but it doesn't see my bed very well... even tried some steel banding strips under the glass and still no.
They will see it reliably at about .5mm from the bed. Too close for my comfort.

Spawn32, 2017-10-10:
Ahh, glass, i switched to a all in one heatbed with integrated alu plate, on top of it I use a buildtak…
On glass I used the 18mm, never tested the 12mm on it..