MarlinFirmware/Marlin #7911: [BR] 1.1.5 ABL Ignoring Z Offset - Creality CR-10S 2560 Board

houseofbugs, 2017-10-09:
### Bug Report
- Description: After running G28 G29 at random the Z Offset will be ignored.
- Expected behaviour: Printer moves head to print height after G29 completes
- Actual behaviour: Printer head is up same MM as the Z offset
- Steps to reproduce:
  - Flash CR-10S 2560 Board with the config attached.
  - At random it will ignore the Z Offset during the beginning of a print.
  - Next print may or may not have the same issue.

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.
[CR10SCFG.zip](https://github.com/MarlinFirmware/Marlin/files/1369281/CR10SCFG.zip)

RalphSch, 2017-10-20:
Can you please post the output of M503?

houseofbugs, 2017-10-25:
I am getting my customers that had these issues to send me their output. I haven't been able to replicate on my machines with the board but I have about 6 people that had the same issue. Going back to 1.1.4 fixed it.

houseofbugs, 2017-10-26:
Send: M503
Recv: echo: G21 ; Units in mm
Recv: echo: M149 C ; Units in Celsius
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo: M200 D1.75
Recv: echo: M200 D0
Recv: echo:Steps per unit:
Recv: echo: M92 X80.00 Y80.00 Z400.00 E100.01
Recv: echo:Maximum feedrates (units/s):
Recv: echo: M203 X300.00 Y300.00 Z5.00 E25.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo: M201 X1000 Y1000 Z100 E25
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo: M204 P500.00 R1000.00 T500.00
Recv: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo: M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
Recv: echo:Home offset:
Recv: echo: M206 X0.00 Y0.00 Z0.00
Recv: echo:Auto Bed Leveling:
Recv: echo: M420 S1 Z0.00
Recv: echo:Material heatup parameters:
Recv: echo: M145 S0 H200 B60 F0
Recv: M145 S1 H240 B90 F0
Recv: echo:PID settings:
Recv: echo: M301 P17.42 I1.27 D59.93
Recv: echo: M304 P690.34 I111.47 D1068.83
Recv: echo:Z-Probe Offset (mm):
Recv: echo: M851 Z-1.79
Recv: ok
[...]
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1930
Please see https://bit.ly/octoserial for possible reasons of this.
Changing monitoring state from 'Operational' to 'Offline: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1930'
Connection closed, closing down monitor
Changing monitoring state from 'Offline' to 'Detecting serial port'
Serial port list: []
Changing monitoring state from 'Detecting serial port' to 'Error: Failed to autodetect serial port, please set it manually.'
Failed to autodetect serial port, please set it manually.
Changing monitoring state from 'Offline' to 'Detecting serial port'
Serial port list: []
Changing monitoring state from 'Detecting serial port' to 'Error: Failed to autodetect serial port, please set it manually.'
Failed to autodetect serial port, please set it manually.
Changing monitoring state from 'Offline' to 'Detecting serial port'
Serial port list: []
Changing monitoring state from 'Detecting serial port' to 'Error: Failed to autodetect serial port, please set it manually.'
Failed to autodetect serial port, please set it manually.
Changing monitoring state from 'Offline' to 'Detecting serial port'
Serial port list: []
Changing monitoring state from 'Detecting serial port' to 'Error: Failed to autodetect serial port, please set it manually.'
Failed to autodetect serial port, please set it manually.
Changing monitoring state from 'Offline' to 'Detecting serial port'
Serial port list: []
Changing monitoring state from 'Detecting serial port' to 'Error: Failed to autodetect serial port, please set it manually.'
Failed to autodetect serial port, please set it manually.
Changing monitoring state from 'Offline' to 'Detecting serial port'
Serial port list: []
Changing monitoring state from 'Detecting serial port' to 'Error: Failed to autodetect serial port, please set it manually.'
Failed to autodetect serial port, please set it manually.
Changing monitoring state from 'Offline' to 'Detecting serial port'
Serial port list: ['/dev/ttyUSB0']
Connecting to: /dev/ttyUSB0
Changing monitoring state from 'Detecting serial port' to 'Opening serial port'
Connected to: Serial<id=0x6f1909b0, open=True>(port='/dev/ttyUSB0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Recv: start
Send: N0 M110 N0*125
Recv: echo: External Reset
Recv: Marlin 1.1.5
[... 71 lines cut ...]
Recv: echo: M200 D1.75
Recv: echo: M200 D0
Recv: echo:Steps per unit:
Recv: echo: M92 X80.00 Y80.00 Z400.00 E95.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo: M203 X300.00 Y300.00 Z5.00 E25.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo: M201 X1000 Y1000 Z100 E25
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo: M204 P500.00 R1000.00 T500.00
Recv: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo: M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
Recv: echo:Home offset:
Recv: echo: M206 X0.00 Y0.00 Z0.00
Recv: echo:Auto Bed Leveling:
Recv: echo: M420 S0
Recv: echo:Material heatup parameters:
Recv: echo: M145 S0 H200 B60 F0
Recv: M145 S1 H240 B90 F0
Recv: echo:PID settings:
Recv: echo: M301 P17.42 I1.27 D59.93
Recv: echo: M304 P690.34 I111.47 D1068.83
Recv: echo:Z-Probe Offset (mm):
Recv: echo: M851 Z-1.79
Recv: ok

nopick, 2017-11-12:
I am one of houseofbugs users. I am also having the problem on 1.1.6. I can provide more data if needed.

nopick, 2017-11-13:
This is also happening with 1.1.6 and 1.1.x bugfix.

thinkyhead, 2017-11-14:
Please install the latest `bugfix-1.1.x` and enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL`. Execute `M111 S247` before `G28` and `G29` to show the most detail in the leveling procedure. Then try some moves with `G1` to see that it is performing properly. Do occasional `M114` commands to get the current position output. Copy the console output from the host and put into a `.txt` file, then drop into your reply. Hopefully it will reveal where the issue is rooted.

nopick, 2017-11-14:
Here is the gcode i ran and the terminal output.
[terminal.txt](https://github.com/MarlinFirmware/Marlin/files/1472372/terminal.txt)
[Test_CR10-S5_1.75.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/1472374/Test_CR10-S5_1.75.gcode.txt)

joshfria, 2017-12-11:
Was there any further work or bug fixes on this?