MarlinFirmware/Marlin #7912: [Marlin 1.1.5] Lcd bed leveling executes 2 times in a row

Its-my-right, 2017-10-09:
Hi,

With Marlin 1.1.5 on a Prusa I3 clone, sometimes when I do a manual bed leveling via the lcd screen, the bed leveling process executes a second time just after the first one. 

In this case, the lcd displays "leveling done" only at the end of the first one.

Is it a known issue or Am I just doing something wrong ?

Don't hesitate if I can upload files or info.

Thanks

Tannoo, 2017-10-10:
Please attach your config files.

EddieFerg, 2017-10-10:
Is this the issue you are seeing?
https://github.com/MarlinFirmware/Marlin/issues/7491

If so, I've not tried going through the process twice to see if it then exits properly to the Main Menu.

Its-my-right, 2017-10-10:
Hi, you will find attached my config files.

It looks like the #7491 indeed, sorry I didn't noticed it. But the percentages are not 30% success and 70% looping but rather 50/50 for me.

[Marlin conf.zip](https://github.com/MarlinFirmware/Marlin/files/1372893/Marlin.conf.zip)

thinkyhead, 2017-11-14:
Has this ever been resolved? If not, might as well fix it now! Please test `bugfix-1.1.x` and let us know.

ir-fuel, 2017-11-16:
I have the same issue on my flashed CR-10S with Marlin 1.1.6. Sometimes it works, most of the time it keeps on running in circles.

ir-fuel, 2017-11-16:
@thinkyhead Are you asking for someone to install the bugfix branch on their printer? If so, let me know and I might have a go at it and see what happens.

ir-fuel, 2017-11-16:
In the mean time I did the test. Took the bugfix branch and compiled and deployed it to my CR-10S. Behaviour is still the same. I can't get out of the bed leveling loop. I also don't see the corner levelling anymore (although it is enabled in the configuration #define). On my printer it has basically become impossible to level the bed. It just keeps on restarting (even with the 1.1.6 build).

ir-fuel, 2017-11-16:
Is there anything I can do to help diagnose the issue? I am a developer, but I have zero experience with the Marlin codebase

thinkyhead, 2017-11-26:
@ir-fuel What form of bed leveling are you testing with? Please post your configurations.

ir-fuel, 2017-11-26:
Here are my config files.

I am using:

`#define MESH_BED_LEVELING`

[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/1503964/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1503965/Configuration.txt)

EddieFerg, 2017-11-26:
I'm using:

#define MESH_BED_LEVELING
#define LCD_BED_LEVELING

I have a 2004 LCD Smart Display Controller.

After adjusting Z on the final point and pressing the knob, it Homes XYZ, then displays "Click to Begin". That is, MESH 
[configh.txt](https://github.com/MarlinFirmware/Marlin/files/1504118/configh.txt)
Bed Leveling upon completion does not exit to the Main Menu, and you're now stuck in the Mesh creation loop and can't exit.

This issue is repeatable on my printer, although it has returned to Main Menu a couple of times.

ir-fuel, 2017-11-26:
I had it happen again to me today. I power cycled the printer and then it didn't do it anymore.

thinkyhead, 2017-11-27:
It looks like MBL is getting a little long-in-the-tooth in trying to follow old-school methods for interacting with the LCD. The bilinear `PROBE_MANUALLY` procedure doesn't suffer from the same problems. So, I will aim to rewrite the MBL code so it conforms to the way that `PROBE_MANUALLY` works.

Roxy-3D, 2017-11-27:
> The bilinear PROBE_MANUALLY procedure doesn't suffer from the same problems. So, I will aim to rewrite the MBL code so it conforms to the way that PROBE_MANUALLY works.

It is my belief Marlin will benefit if we make a 'Manual Probe' always available if there is an LCD Panel available to move the nozzle up and down (and click).    My thinking is most commands should default to using the real Z-Probe if there is one available in the system.   If not...  Those commands would revert to using the always available 'Manual Probe'.     And in some cases...  the commands should be sensitive letting the user over ride which probe is used (assuming both are available).

This school of thought would help MBL as well as UBL.   I probably helps Bi-Linear also.

thinkyhead, 2017-11-27:
> if we make a 'Manual Probe' always available if there is an LCD Panel 

Manual probing with MBL and ABL doesn't require an LCD. You can just adjust Z using G1 in-between invocations of `G29`.

thinkyhead, 2017-11-27:
Ok, so there's now a patch for the issue linked above. Since ABL with `PROBE_MANUALLY` works fine, I just decided to make MBL follow the same pattern. My own testing shows it to be working, but please let me know if there are any other anomalies.

thinkyhead, 2017-11-27:
Merged the patch. Please file a new report if the updated MBL still gives trouble.

PR #8574 (2017-11-27): [1.1.x] Make MBL work more like PROBE_MANUALLY
Fix #7912 and Fix #7491 by making MBL's `LCD_BED_LEVELING` work in the same manner as `PROBE_MANUALLY`.
PR #8575 (2017-11-27): [2.0.x] Make MBL work more like PROBE_MANUALLY
See also #8574

Fix #7912 and Fix #7491 by making MBL's `LCD_BED_LEVELING` work in the same manner as `PROBE_MANUALLY`.
closed 2017-11-27 by thinkyhead