MarlinFirmware/Marlin #7913: [Z Offset]Nozzle-Bed distance

joao-righetti, 2017-10-09:
I'm adjusting a 3D printer. I levelled all of it's axis.
When I started printing, the nozzle was too far from the bed, and so the PLA was going wasted.
As I understand I got to change the Z offset in order to the print to come out ok (starting at a reasonable distance).
But I dont know how much and in which direction.
Some useful information:

- Z coordinate after leveling: 11.74

- Z offset: -01.75

- Paper test: the paper goes through the nozzle and the bed at +000.8 bot not at +000.7

If I'm missing any important information, please let me know!

So, i got to make the Z offset more negative or more positive? And how much should I increase or decrease that number?

Thank you for your help!

P.S.:I'm new to github, so if I posted in the wrong way or in the wrong place, I'm Sorry!

Tannoo, 2017-10-10:
Negative moves closer to the bed.

Roxy-3D, 2017-10-10:
> Negative moves closer to the bed.

Assuming you are above the bed!

Tannoo, 2017-10-10:
I don't believe that would matter because if the bed were above the printing head, the steppers would be reversed. That way, all software operations remain the same.

Chaos99, 2017-10-10:
What I do (tm):

(Assuming a cold nozzle and bed, with a z probe that triggers somewhat below the nozzle.)

1. `M502` to revert to defaults (and deactivate bed leveling)
2. `M500` save defaults
3. `M503` to print out settings. Notice the `M851 Z probe offset` line. Whatever you can read here is what is set as probe offset in the configuration.h of the Marlin firmware.
4. `M851 Z0` will reset this to 0 for the following calibration
5. `G28` to let it home in all  axis. The nozzle will be above the bed by the distance of the probe trigger point to the nozzle, plus the safety margin configured in the firmware to move down after homing
6. `M211 S0` deactivate software endstops, so we can go past Z0.0
7. `G0 Z0` to go to the probe trigger point (in case there was a safety margin)
8. `M119` reports the endstops/probe status. Ensure that it reads `TRIGGERED` for the z-probe. Move up and down with `G0 Z0.05` or `G0 Z-0.05` to check that you are in deed right at the trigger point where it switches back and forth between `TRIGGERED` and `open`
9. `M114` note down the exact z height
10. Use `G0 Z-x.xx` to slowly approach the perfect nozzle height with the paper method. See notes to temperature below.
11. `M114` Note down the final height, subtract the initial height of the trigger point.
12. `M851 Z-x.yz` Set the calculated z offset (negative, if nozzle is above trigger point)
13. `M500` Save the value
14. `G29` start bed leveling (details may vary depending on your bed level algorithm)
15. `M500` save bed level data
16. `M211 S1` re-enable software endstops

Remember to re-do (G29) or re-use (M420 S1) bed leveling after each new G29. (Probably inside your startup gcode.)

You could also put the value as probe offset into the configuration.h to have it as your default value.

You are done. You have to repeat this whenever you change the distance of your nozzle in relation to the sensor (e.g. mount a different heat break or nozzle).

A note to temperatures: When your nozzle or your bed is hot, it changes dimensions and therefore the z offset. The thickness of the paper is meant to compensate for that. But of course this is not 100% precise. But beware of doing the above with hot nozzle and/or bed: The hot nozzle will heat the bed locally and deform it. Also, more severe, inductive probes (and probably others too) have a temperature drift. The probe will heat up when near the nozzle or near the heated bed and the trigger point will shift, rendering this method useless. If you absolutely MUST do, then be very quick.

If anyone has an easier way to do this, I'm happy to hear it. Also any corrections to the above.

pandel, 2017-10-10:
@Chaos99 
Just for the record, I'll use a slighlty different method. It is oderwat's solution, posted some time ago, so I only repeat his words:

```
; preheat for working temperatures
M502 ; optionally reset to firmware settings
M500; save
M851 Z0 ; reset any z-offset
G28 ; home all
G1 X110 Y110 ; move nozzle in center
M211 S0 ; disable software endstop (optionally, if not disabled via configuration.h)
; now move your nozzle manually onto the bed such that it (strongly) grips paper.
G92 Z0 ; tell the printer that this is Z = 0
G30 X110 Y110 ; move the sensor over the center and let it trigger
; note what z position G30 reports, that, negated, is your z offset
M851 Z-1.23 ; register that z offset (here 1.23 as example)
M500 ; store those values into the EEPROM
M211 S1 ; enable software endstop (optionally, if not disabled via configuration.h)
```

That has always worked for me using linear leveling.

joao-righetti, 2017-10-11:
@pandel 
I think I'll try this routine you suggested.
But this printer cant be connected to a computer. I can control its movements, temperature at it's own display. Do you think I should try this manually or generate a gcode to the SD card and do it step by step?

fiveangle, 2017-10-11:
@pandel - That is a kludge that uses workspace offsets to translate the coordinate system to be nozzle-centric (once you manually set the point).  This has been a large source of confusing for many, and I think now NO_WORKSPACE_OFFSETS is even enabled by default now, so I would move away from that method.

@Chaos99 method that summes up the entirety of the Z-offset in within the M851 adjustment value is the preferred method to adjust.

pandel, 2017-10-11:
@fiveangle - NO_WORKSPACE_OFFSET is **disabled** by default, I enabled it to circumvent the behaviour you described. But I don't really understand why you see this as a kludge. If you **disable** NO_WORKSPACE_OFFSET, then you only get the sensor misreading when doing a G30, nothing more. That makes no sense at least to me when it comes to the Z value this command spits out...

But in fact, I don't get this whole workspace offset thing regarding Z offset _in combination_ with a Z probe, but perhaps it is because I don't really understand the reason for this. I don't need a persistent Z offset if I have a Z _probe_ offset with which I can set the correct base level via M851. Sometimes, and I'm totally fine with that, I may need a Z offset _temporarily_, for special filaments or whatever, but that's another thing imho...

@Chaos99 - Perhaps you should forget my comment though ;-) Things seem to get complicated here - and if your way works for you then stay with it!

thinkyhead, 2018-02-09:
> I think now NO_WORKSPACE_OFFSETS is even enabled by default now

Nope.

closed 2018-02-09 by thinkyhead

fiveangle, 2018-02-09:
Yes- that was a typo - it was enabled be default for the longest, and then disabled.  Corrected comment for all posterity...