MarlinFirmware/Marlin #7921: [BR] Build failure for Re-ARM using bugfix-2.0.x

TGMods, 2017-10-10:
### Compile Error
When I compile with the config I've been using for this project since bugfix-2.0.x was first published I get an error:
```
> Executing task: platformio.exe run <

[10/10/17 08:24:17] Processing Re-ARM (build_unflags: -Wall; build_flags: !python Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py; src_build_flags: -Wall; lib_deps: U8glib-ARM, CMSIS-LPC1768; extra_script
s: Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py; board_f_cpu: 100000000L; platform: nxplpc; lib_extra_dirs: frameworks; lib_ldf_mode: off; src_filter: +<src/*> -<src/config>)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Warning! Library `CMSIS-LPC1768` has not been found in PlatformIO Registry.
You can ignore this message, if `CMSIS-LPC1768` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Verbose mode can be enabled via `-v, --verbose` option
Collected 6 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <U8glib-ARM> v1.18.1
|-- <CMSIS-LPC1768> v0.0.0
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_AVR\HAL_AVR.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_AVR\HAL_spi_AVR.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_AVR\MarlinSerial.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_AVR\persistent_store_impl.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_AVR\servo_AVR.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_AVR\watchdog_AVR.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_DUE\HAL_Due.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_DUE\HAL_spi_Due.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_DUE\HAL_timers_Due.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_DUE\Servo_Due.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_DUE\persistent_store_impl.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_DUE\watchdog_Due.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_LPC1768\HAL.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_LPC1768\HAL_spi.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_LPC1768\HAL_timers.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_LPC1768\HardwareSerial.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_LPC1768\LPC1768_Servo.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_LPC1768\SoftwareSerial.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_LPC1768\WInterrupts.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_LPC1768\arduino.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_LPC1768\main.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_LPC1768\persistent_store_impl.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_LPC1768\watchdog.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_STM32F1\HAL_Servo_Stm32f1.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_STM32F1\HAL_Stm32f1.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_STM32F1\HAL_spi_Stm32f1.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_STM32F1\HAL_timers_Stm32f1.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_STM32F1\persistent_store_impl.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_STM32F1\watchdog_Stm32f1.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_TEENSY35_36\HAL_Servo_Teensy.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_TEENSY35_36\HAL_Teensy.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_TEENSY35_36\HAL_spi_Teensy.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_TEENSY35_36\HAL_timers_Teensy.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_TEENSY35_36\persistent_store_impl.o
Compiling .pioenvs\Re-ARM\src\src\HAL\HAL_TEENSY35_36\watchdog_Teensy.o
Compiling .pioenvs\Re-ARM\src\src\HAL\I2cEeprom.o
Compiling .pioenvs\Re-ARM\src\src\HAL\SpiEeprom.o
Compiling .pioenvs\Re-ARM\src\src\HAL\servo.o
Compiling .pioenvs\Re-ARM\src\src\Marlin.o
Compiling .pioenvs\Re-ARM\src\src\core\serial.o
In file included from Marlin\src\Marlin.cpp:36:0:
Marlin\src\module/stepper.h: In static member function 'static uint32_t Stepper::calc_timer(uint32_t)':
Marlin\src\module/stepper.h:300:104: error: the value of 'SystemCoreClock' is not usable in a constant expression
constexpr uint32_t MIN_TIME_PER_STEP = (HAL_STEPPER_TIMER_RATE) / ((STEP_DOUBLER_FREQUENCY) * 2);
^
In file included from frameworks\CMSIS\LPC1768\include/LPC17xx.h:104:0,
from Marlin\src\inc/../HAL/HAL_LPC1768/fastio.h:37,
from Marlin\src\inc/../HAL/HAL_LPC1768/HAL.h:59,
from Marlin\src\inc/../HAL/HAL.h:47,
from Marlin\src\inc/MarlinConfig.h:33,
from Marlin\src\Marlin.h:29,
from Marlin\src\Marlin.cpp:31:
frameworks\CMSIS\LPC1768\include/system_LPC17xx.h:31:17: note: 'uint32_t SystemCoreClock' is not const
extern uint32_t SystemCoreClock;     /*!< System Clock Frequency (Core Clock)  */
^~~~~~~~~~~~~~~
*** [.pioenvs\Re-ARM\src\src\Marlin.o] Error 1
======================================================================================== [ERROR] Took 30.12 seconds ========================================================================================

================================================================================================= [SUMMARY] =================================================================================================
Environment megaatmega2560              [SKIP]
Environment megaatmega1280              [SKIP]
Environment teensy20                    [SKIP]
Environment rambo                       [SKIP]
Environment anet10                      [SKIP]
Environment sanguino_atmega644p         [SKIP]
Environment DUE                         [SKIP]
Environment teensy35                    [SKIP]
Environment Re-ARM                      [ERROR]
Environment Re-ARM_debug_and_upload     [SKIP]
======================================================================================== [ERROR] Took 30.18 seconds ========================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

```

------------------------------------------------------------
### Bug Report
- Description: Compile Failure
- Expected behaviour: **Environment Re-ARM                      [SUCCESS]**
- Actual behaviour: **Environment Re-ARM                      [ERROR]**
- Steps to reproduce:
  - Build bugfix-2.0.x for Re-ARM

This bug occured after [2.0.x] Apply fixes for DUE #7890 was merged.
------------------------------------------------------------
### Bug Report Tips
Changing **constexpr** in this code snippit from stepper.h to **const** allows bugfix-2.0.x to build for Re-ARM.

```
      #ifdef CPU_32_BIT
        // In case of high-performance processor, it is able to calculate in real-time
        **constexpr** uint32_t MIN_TIME_PER_STEP = (HAL_STEPPER_TIMER_RATE) / ((STEP_DOUBLER_FREQUENCY) * 2);
        timer = uint32_t(HAL_STEPPER_TIMER_RATE) / step_rate;
        NOLESS(timer, MIN_TIME_PER_STEP); // (STEP_DOUBLER_FREQUENCY * 2 kHz - this should never happen)
      #else
```

psavva, 2017-10-10:
I suggest you submit a pull request.

TGMods, 2017-10-10:
@psavva Like I've said many times before in other posts, I'm not a developer, I'm just a very interested user. I'm willing to test the code and pass on my observations.

closed 2017-10-11 by TGMods