MarlinFirmware/Marlin #7922: [1.1.5] Resets when activating hot end

ashthespy, 2017-10-10:
### Bug Report
- Description: Sending a `M104 Sxx` causes a reset 
- Expected behaviour: Hot end heats up
- Actual behaviour: Board resets
- Steps to reproduce:
  1. Flash latest Marlin `1.1.5` ( also present on `bugfix-1.1.x` branch)
  1. Open up repetier host and send `M104 S45` via console
  1. Marlin resets.

Some more details, I am using an Azteeg X3 board, with a generic E3D hot end. I have a single extruder/hot-end and no heated bed. Normal motion G Codes seem to work fine.

[Configuration files](https://gist.github.com/ashthespy/69d160b4fb0a8bac8942567f94808d5f)

How would one debug this?

GMagician, 2017-10-10:
Try to disconnect heat cartridge from board and test again.
This seems like a short-circuit issue

ashthespy, 2017-10-10:
Hmm, I measured the resistance of the heater cartridge, and it is about 3.6 ohm - as expected for a 40W cartridge.
But I shall try swapping out cartridges this evening and check

simon-jouet, 2017-10-10:
Is it a new build? (Did it work before?) Because it could also be your power supply that's weak and can't deliver enough power.

ashthespy, 2017-10-10:
Sorry for wasting your time, was in fact the power supply. I assumed stupidly that the beefy lab regulated DC supply had a high power output - but in fact was limited to 2 Amps :-S
Works fine now with a beefier 40A PSU, and I am no on to printing my first calibration cube!

Cheers!

closed 2017-10-10 by ashthespy