MarlinFirmware/Marlin #7925: Question: G28 Recovery

apballard, 2017-10-10:
Hi All, quick question.

I've been looking into the config and what not, and when I do a G28, because of the sensor offset from the actual nozzle, instead of probing at X150 Y100, it does more of a X150 Y88 and in that way the sensor is at the X150 Y100 location.

Is there an option to return the nozzle after the probing back to X150 Y100?

Roxy-3D, 2017-10-10:
You can put an extra line into your Slicer's start up GCode to move it there.   A G1 X150 Y100.    But you can also pick the SAFE_HOMING coordinates.   You could pick a location offset by some amount to put the nozzle at the center of the bed.

But really...  Why do you care?   It probably is best to probe the center of the bed, and when you start printing your GCode file, the nozzle will move to where it is supposed to go.

apballard, 2017-10-10:
@Roxy-3D Thanks for the reply.  It dawned on me that I did not mention that I do want it to probe the centre, but I then like to double check the z-offset at the probed point.  Not sure if this makes a difference.

Roxy-3D, 2017-10-10:
Well...   At some point, you need to trust the configuration of your machine.   And...   Really...   the proof is in the pudding.   The whole purpose of having everything configured correctly is so you can print.   After a G28, you are ready to heat things up and print.

apballard, 2017-10-11:
I agree, thanks

closed 2017-10-11 by apballard