MarlinFirmware/Marlin #7927: UBL Is not compensating enough for first layer

Festivejelly, 2017-10-10:
------------------------------------------------------------
### Bug Report
- Description: UBL Is not compensating enough for first layer
- Expected behaviour: UBL should raise or lower the Z axis sufficiently in order to get a fairly consistent first layer.
- Actual behaviour: Only half of the bed prints at the right height and the other prints slightly too high.
- Steps to reproduce:
  - Home all
  - Do G29 P1
  - Do G29 S1 
  - Do G29 A

This should create a mesh saved in slot one then activates it.

Then when I do a print my before gcode is as follows:

```
G21 ;metric values
G90 ;absolute positioning
G28 ; home all axes
G29 A; activate auto leveling
G29 L1; load auto level mesh in slot 1
G1 X5 Y0 F3000; move to front of bed
M82 ;set extruder to absolute mode
G92 E0 ;zero the extruded length
G0 Z0.25 ; Drop to bed
G1 Y3.0 F500.0 ; move out of print volume
G1 X60.0 E9 F500.0 ; start purge line
G1 X100.0 E12.5 F500.0 ; finish purge line
G92 E0 ;zero the extruded length again
M117 Printing...
```
Here is my mesh that I just did just before a failed print:

-0.072 -0.035 0.013 0.020 0.072 0.087 0.097 0.077 0.077 0.092
-0.095 -0.050 -0.023 0.013 0.050 0.053 0.072 0.053 0.038 0.053
-0.080 -0.053 -0.002 0.003 0.040 0.048 0.048 0.040 0.008 0.048
-0.077 -0.043 -0.030 0.005 0.030 0.033 0.045 0.038 0.008 0.025
-0.063 -0.043 -0.028 -0.023 0.005 0.030 0.043 0.033 0.003 0.010
-0.072 -0.038 -0.020 0.008 0.023 0.033 0.045 0.028 0.013 0.013
-0.105 -0.070 -0.033 -0.045 0.000 0.030 0.015 0.003 -0.005 -0.025
-0.095 -0.077 -0.045 -0.028 0.003 0.035 0.048 0.018 0.010 0.008

Looks fairly level to me not hugely out I dont think, at least within the tolerances for what the bed leveling can handle.

I don't understand why its not compensating sufficiently. I can see the lead screws moving up and down as I watch it but it just looks like its not doing it enough.

I've attached my configs for marlin.

Just FYI im using a Ramps board with DRV8825 drivers on 1/16 microstepping.
I'm using Marlin 1.1.5 and a genuine BL touch probe. I've measured the offset of the probe correctly too.

[MarlinBugReport.zip](https://github.com/MarlinFirmware/Marlin/files/1373374/MarlinBugReport.zip)

Roxy-3D, 2017-10-10:
So...   Let's do a G26 and see what the Mesh Validation Pattern looks like.

Festivejelly, 2017-10-10:
So I did a G26 C P T3.0
And it actually didnt look too bad. Should I do it with just G26 without params?

Roxy-3D, 2017-10-10:
Two comments...  First G26 is going to print a first layer across the entire bed.   If it can get good adhesion, your normal prints probably will too.   'T' is not a G26 parameter.   I don't know what you were trying to accomplish with that.

You can do G26 with no parameters if you have the default bed and nozzle heat specified correctly (at the top of) the G26.cpp file.

Festivejelly, 2017-10-10:
Hrm I think I may have copied it from somewhere. I shall do that now.

Festivejelly, 2017-10-11:
[image]
[image]
[image]
[image]

Ive attached some images. Sorry for the bad quality but you can see on the left side the lines are quite visibly flatter whilst to the right the lines are more cylindrical shaped. I'm surprised it even stuck to the bed, but I think thats because of the magic of the printbite tbh.

Festivejelly, 2017-10-11:
Sorry about the rotation it seems github has rotated them

AnHardt, 2017-10-11:
One of your z-axis is leaning forward/backward. For that the sled is rotating around the x-axis, when moving along the x-axis, Because the probe has a higher distance from the center of the rotation then the nozzle it's detecting higher height differences than the nozzle has in reality.

Best option is to correct the z-towers.
Second best is to hand-tune the UBL-grid.

Festivejelly, 2017-10-11:
If its learning forward wouldnt that show up on the bed as changes to the front and back? not the left and right? I've updated the pics so they are not rotated.

Festivejelly, 2017-10-11:
I will replace my Z rods to eliminate this variable and report back.

Festivejelly, 2017-10-11:
Replaced Z lead screws, checked everything is aligned and it is. No movement of the print head. But im still getting the same issue.

Roxy-3D, 2017-10-11:
I'm going to have to think about this.   The Z-Height (and correction) appear to be working.  But it is missing a few line segments of the circles.  Its almost like it didn't even try to draw them.

Festivejelly, 2017-10-11:
I will try that. But isnt that what the bed level probe is supposed to find for me? I think the cut off circles are because I inset the mesh by 10 mm, so I suppose its not trying to draw outside.

Festivejelly, 2017-10-12:
Okay so I actually watched the whole thing because I was bored and I noticed that it tried to probe outside of the printable area.

In my config I have the following values:

`  // Set the boundaries for probing (where the probe can reach).
  #define LEFT_PROBE_BED_POSITION 10
  #define RIGHT_PROBE_BED_POSITION 185
  #define FRONT_PROBE_BED_POSITION 20
  #define BACK_PROBE_BED_POSITION 155`

So to me that says the maximum distance backwards the code can probe is 155 Y
and the maximum forward is 20 Y

Now then when I do a home my probe travels to the middle of the bed, because I have safe Z turned on. What im wondering is if the UBL thinks im at 0,0 when im not, in actual fact im at 100,100.

Essentially the behavior I noticed was that during the probing the bed was trying to move too far forward (ie past the maximum of 0 on the Y which in actually fact the maximum should be 20 as per above code).
So my hypothesis is that if I do G28 which does a safe home, then do G28 X0 Y0 and THEN do G29 P1 will it obey the laws of boundaries.
We shall find out soon. I shall make a coffee and observe and report back.

PS( if you do G28 with safe homing then do G26 it tries to print outside of the bed)

Roxy-3D, 2017-10-12:
No...   Sorry...   Those options are only for the normal Grid Based Leveling and Bi-Linear leveling.
Those values have no meaning inside the UBL system.
```cpp
#if ENABLED(AUTO_BED_LEVELING_LINEAR) || ENABLED(AUTO_BED_LEVELING_BILINEAR)

  // Set the number of grid points per dimension.
  #define GRID_MAX_POINTS_X 3
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X

  // Set the boundaries for probing (where the probe can reach).
  #define LEFT_PROBE_BED_POSITION 39
  #define RIGHT_PROBE_BED_POSITION 170
  #define FRONT_PROBE_BED_POSITION 10
  #define BACK_PROBE_BED_POSITION 170
```

Festivejelly, 2017-10-12:
Okay that makes sense, but lets look at my UBL detailed report:

`1:30:04.001: UBL object count: 1
1:30:04.006: planner.z_fade_height : 10.0000
1:30:04.007: zprobe_zoffset: -1.4499999
1:30:04.009: UBL_MESH_MIN_X 1=1
1:30:04.009: UBL_MESH_MIN_Y 1=1
1:30:04.038: UBL_MESH_MAX_X (195 - (1))=194
1:30:04.039: UBL_MESH_MAX_Y (200 - (1))=199
1:30:04.068: GRID_MAX_POINTS_X 10
1:30:04.068: GRID_MAX_POINTS_Y 10
1:30:04.096: MESH_X_DIST 21.44
1:30:04.096: MESH_Y_DIST 22.00
1:30:04.383: X-Axis Mesh Points at: 1.000 22.444 43.889 65.333 86.778 108.222 129.667 151.111 172.556 194.000
1:30:04.641: Y-Axis Mesh Points at: 1.000 23.000 45.000 67.000 89.000 111.000 133.000 155.000 177.000 199.000`

Lets focus on these:

1:30:04.038: UBL_MESH_MAX_X (195 - (1))=194
1:30:04.039: UBL_MESH_MAX_Y (200 - (1))=199

Some interesting numbers...

Now lets look at my offsets for my probe:

`#define X_PROBE_OFFSET_FROM_EXTRUDER 0  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -42  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER 0   // Z offset: -below +above  [the nozzle]`

Hrm ok so thats 200 - 42 which I make out to be 158

In the Configuration_adv.h I notice theres this bit of code:

`  #define UBL_MESH_MAX_Y (Y_BED_SIZE - (UBL_MESH_INSET))`

So this doesnt seem to be taking into account my offset?

Festivejelly, 2017-10-12:
I get that I could use UBL_MESH_INSET but what if i'm only restricted on the Y? What if I only wanted to put a restriction on that 1 axis?

Roxy-3D, 2017-10-12:
By default the inset is applied symmetrically to all four edges.  But in configuration_adv.h you can adjust the amount of inset on a per edge basis if you so choose.

Festivejelly, 2017-10-12:
Okay I can look at that.

Out of interest what is the purpose of this:

`//===========================================================================
  //========================= Unified Bed Leveling ============================
  //===========================================================================

  #define UBL_MESH_INSET 1          // Mesh inset margin on print area
  #define GRID_MAX_POINTS_X 10      // Don't use more than 15 points per axis, implementation limited.
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X

  #define UBL_PROBE_PT_1_X 39       // Probing points for 3-Point leveling of the mesh
  #define UBL_PROBE_PT_1_Y 155
  #define UBL_PROBE_PT_2_X 39
  #define UBL_PROBE_PT_2_Y 20
  #define UBL_PROBE_PT_3_X 180
  #define UBL_PROBE_PT_3_Y 20`

If its ignored... then why is it there?

Roxy-3D, 2017-10-12:
If what is ignored?      UBL_MESH_INSET affects how much of the bed is used.   You can use it to  keep the nozzle away from the edges.

Festivejelly, 2017-10-12:
No i mean the probe points, sorry, I wasnt clear. I understand the inset.

Why are the probing points defined for UBL if they arnt used?

Roxy-3D, 2017-10-12:
They are used if you use the G29 J option to tilt the bed.   But that option is really only in the system for backward compatibility.   The original bed leveling system was 3-Point.   So it needs to be included just for completeness.

Festivejelly, 2017-10-12:
Makes sense. Okay so its quite possible that issue was affecting my bed probing so the mesh wasnt showing an accurate result. im doing another mesh with some hard coded changes to see if it makes a difference to the compensation issue i was facing.

Roxy-3D, 2017-10-12:
And really...  The right mindset is you just need the initial probing to be accurate enough you can run a G26.   At that point...  you edit any mesh point that is questionable.     And from then on, you just home the Z axis and use the edited mesh for correction.

Festivejelly, 2017-10-12:
this seems sensible but then it begs the question whats the point in having a probe if you have to manually do it? I thought the idea of UBL is that it compensated during the print. On some printers it going to be damn near impossible to get a proper levelled bed. I still think that its not compensating enough when it knows the bed is higher or lower. Im not sure how to prove it other than showing lots of G26 prints.  I wonder if theres a way I can get it to spit out a log when its printing so I can see by how much its trying to compensate by.

AnHardt, 2017-10-13:
Try to print with and without bed leveling.
I would bed leveling is not under but over compensating for the reason described above. Because the probe does not see the same height as the nozzle.
Changing the leadsrew in a similar to Prusa i3 system can hardly help with your problem.
More likely sources for the error include
Distorted frame
Too long/short/broken/distorted  lower/upper smooth rod holders
Additional/omitted washers
Lose screws
Distorted z sleds
Bowed z smoothe rods
...

Roxy-3D, 2017-10-13:
> this seems sensible but then it begs the question whats the point in having a probe if you have to manually do it? I thought the idea of UBL is that it compensated during the print. 

It does.   Let's stop this discussion right here...   Let's turn off all bed level correction and start our stop watch.   I bet its weeks before you can get a print like this with no bed leveling:   https://github.com/MarlinFirmware/Marlin/issues/7618#issuecomment-328297556   

> On some printers it going to be damn near impossible to get a proper levelled bed. 

Yes.   The more physically level and flat the bed is, the better...     UBL does not currently correct for skewed X & Y axis.   But that isn't a common problem.  What was this guy's last thoughts?   

https://github.com/MarlinFirmware/Marlin/issues/7618#issuecomment-328309188

>  I wonder if theres a way I can get it to spit out a log when its printing so I can see by how much its trying to compensate by.

Sure...  With lots of programming effort that can be done.   But how about you look at the mesh point map (G29 T) and just see what the nozzle has to do to track the bed.   And as it turns out, you can do a G29 T 1 to get a .CSV file that can be cut and pasted into Excel and plotted in 3D to show you the surface of the bed.    Maybe that will do what you need?

> I still think that its not compensating enough when it knows the bed is higher or lower. Im not sure how to prove it other than showing lots of G26 prints.

Let's back up...   Let's have you turn off all bed leveling and see a picture of a calibration cube printed at the center of your bed.

Festivejelly, 2017-10-17:
I've printed many calibration cubes just fine. The issue at hand here is, and I can reproduce this, and have done many many times. The z simply isnt being compensated. The probe is telling me the bed is higher on the left than it is on the right (shows up as squish in the g26). I turned debug on and checked the values it was probing and it was pretty damn accurate. I even attached a dial gauge to the printer to measure it myself.
What I notice also during G26 is that it doesnt compensate at all. Is it meant to? What I see is the lead screws turning between each point for the hop but the Z compensation never seems to happen.

Here is a report of my bed as of today:

22:18:01.070: -0.045 0.003 -0.002 -0.005 -0.012 -0.048 -0.090 -0.130 -0.202 -0.202
22:18:01.074:
22:18:01.229: 0.000 0.040 0.000 0.030 0.028 -0.045 -0.090 -0.132 -0.170 -0.177
22:18:01.229:
22:18:01.384: 0.030 0.070 0.045 0.025 -0.007 -0.045 -0.085 -0.130 -0.173 -0.165
22:18:01.385:
22:18:01.545: 0.085 0.043 0.077 0.040 0.033 -0.010 -0.060 -0.122 -0.170 -0.197
22:18:01.545:
22:18:01.700: 0.082 0.100 0.043 0.038 0.030 0.018 -0.058 -0.092 -0.165 -0.185
22:18:01.700:
22:18:01.861: 0.063 0.077 0.077 0.040 -0.005 -0.007 -0.050 -0.085 -0.132 -0.160
22:18:01.861:
22:18:02.081: 0.060 0.043 0.043 0.028 0.025 -0.010 -0.048 -0.087 -0.135 -0.132
22:18:02.081:
22:18:02.237: [ 0.038] 0.028 0.038 0.000 -0.007 -0.010 -0.043 -0.085 -0.125 -0.130
22:18:02.266: (10,10) (190,10)
22:18:02.291: (0,0) (9,0)

The right is a fair bit lower. But well within the range for the steppers to be able to compensate. Yet it doesnt. And as i've said above ive measured this myself with a dial gauge and its about right. I really cant see what the issue is here. The printer is square, I replaced the lead screws just to be sure.

I'm stumped. I guess the next thing for me to try is to get rid of the BL touch an try an alternative sensor.

Roxy-3D, 2017-10-17:
> What I notice also during G26 is that it doesnt compensate at all. Is it meant to? What I see is the lead screws turning between each point for the hop but the Z compensation never seems to happen.

Yes.  G26 most definitely is supposed to compensate as it draws the mesh validation pattern.  Do you have UBL activated?   G29 W will tell you if it is active.   G29 A will activate it.

> And really... The right mindset is you just need the initial probing to be accurate enough you can run a G26. At that point... you edit any mesh point that is questionable. And from then on, you just home the Z axis and use the edited mesh for correction.

Please just use the existing G26 pattern to raise and lower mesh points.   Inductive sensors vary more than BL-Touch probes.  But BL-Touch probes do vary in what they report.    Please see if you can fine tune a mesh.   2 or 3 iterations of doing a G29 followed by a G29 P4 R of 'bad areas' should get everything correct.    

Once you know (via G26's) that you have an accurate mesh....  You can check the probe's accuracy at different places on the bed.

Festivejelly, 2017-10-17:
Yeah Its active. I just did a custom test print and watched the leadscrews and I could see them actually moving as it went along both Y and X axis. Good so i know its doing something. however it doesnt appear to be doing it enough.

On the left side its a bit too squished. Same as in the G26 pattern and on the right side its a bit too high. Again same as on the G26 pattern.

So im left with the conclusion that either Marlin isnt compensating enough OR the BL touch isnt returning an accurate Z height for marlin to offset.

My problem is I dont know what the bad areas are. Are the bad areas on the left And right? because they are too low or too high? What do i change them to? I guess I could do some trial and error and raise the ones on the left by a teeny amount and vice versa for the right.

I wonder if the position of the probe is causing a problem. its offset Y-42 and X+2 maybe It just cant get to the important parts of the bed. I know the corners can usually have much much different readings.

Interestingly the closer it gets to the centre of the bed the better the print looks heh.

Roxy-3D, 2017-10-17:
On this image...

[image]

The left side is too squished.   Raise the very left edge mesh points .1 mm      And the right side has the nozzle too far from the bed.  Lower those mesh points .05 mm and try again.    

Once we get the mesh looking uniform (and with the 'right' amount of squish) we can try to figure out why the BL-Touch is not probing accurately.

Festivejelly, 2017-10-17:
will do, thanks for the tips. As you can see I have some silicone underneath the glass, so I've removed that plate and put a new one on that doesnt have silicone. I'm wondering if its deforming the glass. Currently doing a G29 P1 to get a fresh mesh. Then ill do another G26 and see what that yields.

There were 2 sets of probing points on the Y that it couldnt reach because of the -42 offset so i auto populated those with G29 P3. Will be interesting to see if the bed looks much different now.

Roxy-3D, 2017-10-17:
OK!   And if your first mesh validation pattern doesn't look the same...   The middle looks good in that picture.   Raise and lower mesh points to get the whole thing to look like the middle.    The left edge (of that picture) looks like the mesh points could be .2 mm too high.   But it is best to just move .1 mm or .05 mm at a time.   You will see improvement very quickly.

AnHardt, 2017-10-17:
Right/Left? If it gets worse you probably have a "left handed coordinate system".

Festivejelly, 2017-10-17:
Okay so new report:

0:19:24.385: Bed Topography Report:
0:19:24.410: (0,9) (9,9)
0:19:24.439: (10,190) (190,190)
0:19:24.664: . . . . . . . . . .
0:19:24.664:
0:19:24.824: . . . . . . . . . .
0:19:24.825:
0:19:24.980: -0.200 -0.165 -0.087 -0.055 [-0.020] -0.012 0.013 -0.012 -0.048 -0.058
0:19:24.980:
0:19:25.205: -0.165 -0.127 -0.058 -0.058 -0.015 0.018 -0.012 -0.045 -0.058 -0.058
0:19:25.205:
0:19:25.361: -0.130 -0.090 -0.050 -0.015 -0.020 -0.012 0.015 -0.053 -0.090 -0.092
0:19:25.361:
0:19:25.520: -0.110 -0.085 -0.063 -0.053 -0.015 -0.017 -0.050 -0.053 -0.100 -0.102
0:19:25.520:
0:19:25.676: -0.085 -0.060 -0.048 -0.053 -0.050 -0.015 -0.030 -0.063 -0.130 -0.127
0:19:25.676:
0:19:25.835: -0.077 -0.055 -0.048 -0.048 -0.053 -0.020 -0.055 -0.065 -0.095 -0.130
0:19:25.835:
0:19:25.992: -0.058 -0.055 -0.012 -0.055 -0.017 -0.048 -0.048 -0.087 -0.132 -0.135
0:19:25.992:
0:19:26.151: -0.058 -0.060 -0.015 -0.012 -0.053 -0.022 -0.055 -0.097 -0.135 -0.140
0:19:26.175: (10,10) (190,10)
0:19:26.204: (0,0) (9,0)

Looks to me like the back left could do with some tweaking downwards.

G29 W here:

0:28:38.186: Unified Bed Leveling System v1.01 active.
0:28:38.234: Mesh 1 Loaded.
0:28:38.288: UBL object count: 1
0:28:38.288: planner.z_fade_height : 10.0000
0:28:38.292: zprobe_zoffset: -1.4499999
0:28:38.292: UBL_MESH_MIN_X (0 + 10)=10
0:28:38.296: UBL_MESH_MIN_Y (0 + 10)=10
0:28:38.324: UBL_MESH_MAX_X (200 - (10))=190
0:28:38.328: UBL_MESH_MAX_Y (200 - (10))=190
0:28:38.353: GRID_MAX_POINTS_X 10
0:28:38.353: GRID_MAX_POINTS_Y 10
0:28:38.382: MESH_X_DIST 20.00
0:28:38.382: MESH_Y_DIST 20.00
0:28:38.669: X-Axis Mesh Points at: 10.000 30.000 50.000 70.000 90.000 110.000 130.000 150.000 170.000 190.000
0:28:38.927: Y-Axis Mesh Points at: 10.000 30.000 50.000 70.000 90.000 110.000 130.000 150.000 170.000 190.000
0:28:38.984: ubl_state_at_invocation :0
0:28:38.984: ubl_state_recursion_chk :0
0:28:39.037: Meshes go from 0x0250 to 0x0F7F
0:28:39.091: sizeof(ubl) : 1
0:28:39.091: z_value[][] size: 400
0:28:39.119: EEPROM free for UBL: 0x0D2F
0:28:39.172: EEPROM can hold 8 meshes.
0:28:39.201: Unified Bed Leveling sanity checks passed.

@AnHardt (left,right,vertical) thats the co ordinate system am I not correct? X = left to right , Y = front to back Z = up and down. 0,0 = 0X and 0Y aka front left and 0,200 = back left.

Festivejelly, 2017-10-17:
Out of curiosity what does the [ ] around the numbers mean? I assume its where it thinks the probe is?

Roxy-3D, 2017-10-18:
> Out of curiosity what does the [ ] around the numbers mean? I assume its where it thinks the probe is?

In ubl.cpp there is a declaration for the magic array that holds the mesh matrix, regardless of size: 
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/feature/bedlevel/ubl/ubl.cpp#L69-L71

  float unified_bed_leveling::z_values[][];  with nothing inside the brackets means we don't care about the dimensions of the matrix.

Festivejelly, 2017-10-18:
Here is my results:

[image]

[image]

Couldn't fit it all in one image. So there looks like theres some consistency issues with the bl touch but nothing it seems above 0.05

Weird stuff. Brain hurts so time for bed but I intend to tackle this more tomorrow and get to the bottom of it! :)

Roxy-3D, 2017-10-18:
So...   Tell it to do a G29 T 1 to make a .csv file to put into Excell.
It will plot the shape of the mesh.   If G26 is dialed in and looking nice,
that is what your bed's surface looks like.

Roxy-3D, 2017-10-18:
Oooops....  You obviously have it in Excell...  Or another spreadsheet...   Tell it to 3d-Plot your bed.   And it might be obvious where your BL-Touch has a hard time making a good, trusted measurement.

Festivejelly, 2017-10-18:
It was google sheets, i'll see if I can get hold of a 3d plot graph. Google sheets doesnt have one. :-/ I dont have excel on my machine.

Festivejelly, 2017-10-18:
-0.190,-0.117,-0.080,-0.037,-0.007,0.035,0.043,0.000,0.000,-0.002
-0.157,-0.115,-0.085,-0.045,-0.005,0.035,0.035,-0.010,-0.007,-0.040
-0.125,-0.075,-0.077,-0.040,-0.002,0.033,0.003,-0.005,-0.053,-0.055
-0.095,-0.075,-0.045,-0.048,-0.015,-0.017,0.023,-0.037,-0.087,-0.097
-0.095,-0.080,-0.072,-0.037,-0.007,0.028,-0.012,-0.050,-0.092,-0.125
-0.085,-0.070,-0.045,-0.080,-0.012,-0.010,-0.015,-0.053,-0.090,-0.117
-0.082,-0.043,-0.040,-0.048,-0.012,-0.040,-0.040,-0.082,-0.120,-0.125
-0.070,-0.070,-0.040,-0.043,-0.040,-0.020,-0.053,-0.095,-0.122,-0.120

Thats my latest one. Now if I could just figured out how to get that in a bloody graph

Roxy-3D, 2017-10-18:
[image]

Festivejelly, 2017-10-18:
Oooo quite the dip there in that 1 corner. I should fix that.

Festivejelly, 2017-10-22:
So ive been doing a LOT of testing and producing a lot of charts. Using excel. Do you know how I can change the steps on the Z axis? so it shows a bigger range than -0.2 to 0.05? 

What I'm going to do is post the charts on a blog post. I want people to see what may look like a flat bed certainly isnt.

[image]
Looks a bit drastic because of the range on the Z. ideally it want it to run from -1mm to +1mm

Festivejelly, 2017-10-22:
Figured it out.

With my bed screwed right down this is what it looks like. 
[image]

When I measure from the frame to the bed it says its even on both sides. Which says to me my sensor may be inaccurate. I'm going to try the BL touch clone to see if that yields similar results. Maybe even a capacative sensor.

Roxy-3D, 2017-10-22:
Also...   Something else we found out in the early days of Mesh Bed Leveling:    Flat pieces of glass very often have an edge that drops off or rises up dramatically.   You can't see it.   But rotating the glass and re-probing shows the edge in a new location.   The height change of that edge is often times enough to cause no adhesion without the mesh bed leveling system active.

Festivejelly, 2017-10-22:
Sorry to nag but just to clarify im looking at this the correct way:

19:31:38.351: -0.327,-0.255,-0.175,-0.100,-0.020,0.048,0.067,0.100,0.095,0.107
19:31:38.503: -0.337,-0.265,-0.180,-0.107,-0.028,0.020,0.067,0.102,0.092,0.145
19:31:38.659: -0.370,-0.293,-0.185,-0.137,-0.033,0.015,0.095,0.092,0.130,0.143
19:31:38.814: -0.345,-0.290,-0.220,-0.145,-0.060,0.015,0.053,0.053,0.092,0.105
19:31:38.969: -0.370,-0.295,-0.217,-0.143,-0.072,-0.025,0.050,0.085,0.087,0.102
19:31:39.126: -0.340,-0.267,-0.220,-0.150,-0.070,-0.023,0.018,0.050,0.055,0.097
19:31:39.277: -0.332,-0.265,-0.187,-0.150,-0.110,-0.035,0.008,0.013,0.045,0.050
19:31:39.498: -0.298,-0.255,-0.180,-0.110,-0.070,-0.028,0.013,0.010,0.013,0.058

With this matrix am I correct in thinking that the bottom left of this table is X 0, Y 0?

Roxy-3D, 2017-10-22:
Yes.   The code that displays the map is trying to put the (X,Y) = (0,0) origin at the lower left.  The code is a little bit complicated because it displays both Map Type 0 and Map Type 1 maps.   If you do a map type 0 you will see the nozzle position on the map also and be easier to understand.
```cpp
    for (uint8_t j = UBL_MESH_NUM_Y_POINTS - 1; j >= 0; j--) {
      for (uint8_t i = 0; i < UBL_MESH_NUM_X_POINTS; i++) {
        const bool is_current = i == current_xi && j == current_yi;

        // is the nozzle here? then mark the number
        if (map0) SERIAL_CHAR(is_current ? '[' : ' ');

        const float f = z_values[i][j];
        if (isnan(f)) {
          serialprintPGM(map0 ? PSTR("   .  ") : PSTR("NAN"));
        }
        else {
          // if we don't do this, the columns won't line up nicely
          if (map0 && f >= 0.0) SERIAL_CHAR(' ');
          SERIAL_PROTOCOL_F(f, 3);
          idle();
        }
        if (!map0 && i < UBL_MESH_NUM_X_POINTS - 1) SERIAL_CHAR(',');

        #if TX_BUFFER_SIZE > 0
          MYSERIAL.flushTX();
        #endif
        safe_delay(15);
        if (map0) {
          SERIAL_CHAR(is_current ? ']' : ' ');
          SERIAL_CHAR(' ');
        }
      }
      SERIAL_EOL;
      if (j && map0) { // we want the (0,0) up tight against the block of numbers
        SERIAL_CHAR(' ');
        SERIAL_EOL;
      }
    }
```

Festivejelly, 2017-10-22:
This is very peculiar. Because whats happening is the lower side of my bed (where you can see the dip in the graph) is being squished into the bed. Yet the higher part of the bed (towards the right) the nozzle is ending up too high and the filament is not adhering because of it. I would have expect it to be the opposite way around.

I just cant see to get my head around why its not working. If the printer know the bed is uneven... then why isnt it compensating. I'm starting to run out of ideas on what to do now.

Roxy-3D, 2017-10-22:
My recommendation is do a fresh G26, post the picture...   And edit the mesh.   You should see improvement after the edit...

Festivejelly, 2017-10-22:
So I did a video to try and show you what my problem is:

https://www.youtube.com/watch?v=5yuoRT5uhCc

Roxy-3D, 2017-10-22:
It was hard to see...   But it didn't look like the Z-Axis adjusted (or moved) when you made the left to right shift in the nozzle.   Is UBL active?    If the nozzle height is greater than the Fade Height (10 mm usually) you won't see movement even with UBL active.   But your nozzle is clearly below that limit.

So the most likely explanation is UBL is not active.   But I couldn't see that for sure by looking at the Z-Axis.

Let's do a G26.     And while doing it...   please verify you see the Z-Axis doing corrections.   Please post the picture of the Mesh Validation Pattern and we will edit one small area of the pattern and see if we can get better tracking.

AnHardt, 2017-10-22:
Repeat the same test as in the video, but this time with deployed probe - in a height where it will not trigger.
Is the difference in height the same as with the nozzle?

Festivejelly, 2017-10-22:
@Roxy-3D Doing a G29 W shows that it was indeed active. I did try doing a test print but it wouldnt stick at all so I gave the bed a good clean with acetone and ill try again.

@AnHardt I'm not sure what you mean here. Set the probe to be deployed... and measure the hight underneath like a did with the nozzle?

Roxy-3D, 2017-10-22:
> Doing a G29 W shows that it was indeed active. 

OK!  Good to know!

> I did try doing a test print but it wouldnt stick at all so I gave the bed a good clean with acetone and ill try again.   

OK...   I'm almost done for the day...  But I'll keep watching for a picture to show up.   And with any luck at all...  We will be able to edit some part of it and 'see' results on the very next G26.     At which point, it will be easier for you to edit the whole thing and make things perfect.

AnHardt, 2017-10-22:
@Festivejelly 
Right.

Festivejelly, 2017-10-23:
Right so the first thing I did was change my build platform so it was easier to see.
Then I warmed the bed to 60c did a home (which centres to the middle of the bed) and then I did a G29 P1.

This then created this mesh (I used G29 P3 to fill in the missing values)

2:39:40.293: -0.250,-0.190,-0.117,-0.082,-0.007,0.042,0.045,0.033,0.070,0.118
2:39:40.448: -0.260,-0.192,-0.120,-0.082,-0.007,0.040,0.040,0.033,0.070,0.118
2:39:40.665: -0.270,-0.195,-0.122,-0.082,-0.007,0.038,0.035,0.033,0.070,0.118
2:39:40.821: -0.280,-0.197,-0.125,-0.048,-0.007,0.035,0.030,0.065,0.108,0.150
2:39:40.972: -0.277,-0.197,-0.125,-0.095,-0.015,0.023,0.060,0.070,0.105,0.118
2:39:41.128: -0.272,-0.195,-0.122,-0.090,-0.045,0.023,0.063,0.070,0.070,0.118
2:39:41.283: -0.242,-0.165,-0.120,-0.055,-0.012,0.060,0.063,0.070,0.072,0.118
2:39:41.440: -0.202,-0.160,-0.085,-0.053,-0.010,0.030,0.067,0.070,0.070,0.115
2:39:41.595: -0.165,-0.120,-0.045,-0.017,-0.010,0.063,0.067,0.067,0.072,0.108
2:39:41.813: -0.125,-0.087,-0.045,-0.007,0.063,0.067,0.100,0.070,0.110,0.113

I then saved the mesh and did G29 W to see if UBL was active. It was not so as AnHardt suggested I deployed the probe to 0.2mm above the bed in the centre.
As I moved the hotend to the left, I noticed more space between the probe and the bed (as you'd expect as per mesh). As I moved right I noticed less space between the bed and the probe.

Next I did G29 A to activate the bed leveling and performed the same task. I noticed the hotend lowered slighty as I went left (okay good) and as I went right it raised a little. Okay so thats how I was expecting it to behave. Awesome. I even measured the distance between the probe at the bed and it stayed constant 0.2mm (I used a feeler gauge).

So i decided to try a G26.

I did a G26 and I got the same bloody results. Too squished on the left too high on the right. I want to pull my hair out. Its like its over compensating. I did some repeatability tests on the sensor and it was coming back with a low standard deviation and was accurately reporting the same result back on every side of the bed. So i'm cautiously ruling out the sensor.

Here is a result of the G26.:

[image]
[image]
[image]

Festivejelly, 2017-10-23:
Its like when I move... it compensates fine, but when I print its like its flipping the mesh so Left becomes right and vice versa,

Roxy-3D, 2017-10-23:
OK...  Let's continue with the plan.    Move the nozzle to the left front corner where it says Print Rite.

And give it a G29 P4 R      Lower the first 3 mesh points .15mm  and lower the next 3 or 4 mesh points by .05 mm.

Then...  Lets move the nozzle to the far right side where you had no adheasion.  Do another G29 P4 R and raise 4 or 5 mesh points by .1 mm.   Save the mesh.

Clean the bed and do another G28 and G26....

I suspect things are going to come out much better.

AnHardt, 2017-10-23:
Thank you for the test.
Overcompensating! :-)

Here an OpenScad visualisation of what could go on.
Not really mathematically correct but you should get the mechanism.

```
// x_rotation.scad

// Animated demonstration for nozzle
// and probe seeing not the same height
// caused by rototion of the x-sled,
// around the x-axis,
// when moving in x-dirextion.
// 
// Load in OpenScad.
// Start animation with about 10 FPS
// and 50 steps.
// Observe from all sides, then take
// o look directly along the x-axis.
// See nozle (red) and probe (blue)
// not flying at the same height.

z_till = 10; // misalignment of the z-towers in °.
// when 0 machine is perfectly adjusted.

nozzle_dist = 20; // from center of rotaton in mm.
// When 0 the effect to the nozzles height is minimized.

probe_dist = 40; // from center of rotaton in mm.
// When same as nozzle_dist the probe will see the same height as the nozzle.
// probe_dist > nozzle_dist --> overcompensation
// probe_dist < nozzle_dist --> undercompensation

rod_len = 300;
rod_dia = 8;
x_rod_dist2 = 25;
cone_size = 10;

module rod(){
  translate([0,0,-rod_len/2])
  cylinder(d= rod_dia, h= rod_len);    
}

module z_rod(){
  rod();
}

module x_rod() {
  rotate([0,90,0]) 
  rod();    
};

module nozzle() {
  color("red")
  translate([0,nozzle_dist,0])
  cylinder(d1=0, d2=cone_size, h=cone_size);
}
module probe() {
  color("blue")
  translate([0,probe_dist,0])
  cylinder(d1 = 0, d2 = cone_size, h = cone_size);
}

rotate([-z_till,0,0]) translate([-rod_len/2,0,0]) z_rod();
rotate([z_till,0,0]) translate([+rod_len/2,0,0]) z_rod();

x_end_shift = x_rod_dist2 * tan(z_till);
echo("x_end_shift",x_end_shift);
x_rot = atan2(x_end_shift, rod_len/2);
echo("x_rot",x_rot);

rotate([0,0,x_rot]) translate([0,0,-x_rod_dist2]) x_rod();
rotate([0,0,-x_rot]) translate([0,0,+x_rod_dist2]) x_rod();

x = $t * rod_len - rod_len/2;
x_sled_rot = 2*z_till * $t - z_till; 
echo("x_sled_rot",x_sled_rot);
rotate([x_sled_rot,0,0]) translate([x,0,0]) nozzle();
rotate([x_sled_rot,0,0]) translate([x,0,0]) probe();
```

[7 more replies]

closed 2017-10-23 by Festivejelly