MarlinFirmware/Marlin #7929: UBL / ENABLE_LEVELING_FADE_HEIGHT causes Z change on Tool Change

dulfe, 2017-10-10:
**- Description:**

When `UBL` is enabled, `ENABLE_LEVELING_FADE_HEIGHT` is defined and Z is less than 10..., every time the Tool Changes, **Z raises a little bit**, changing from T0 to T1 or T1 to T0 keeps adding to Z until it gets to 10 (default for leveling fade).

Changing limit to 5 (using M420 Z5) with current Z less than 5 does the same. If changing Z to 6 (G1 Z6) then changing tools works as expected and it does not change Z.

I am using version 1.1.5 version from 1.1.x branch.

**- Expected behavior:**

Keep Z as is.

**- Actual behavior:**

Changes Z on tool change until leveling fade height gets reached.

- Steps to reproduce:
G28; Home XYZ
G29 L0; Load mesh
G29 A; Active mesh
G1 X100 Y0 Z0.2 F6000; Move a little to the right
T1; Change to tool 1 - This causes the issue, Z moves up
T0; Change to tool 0 - Same here
G1 Z11; Move up over fade limit
T1; Change to tool 1 - Z DOES NOT move - as expected
T0; Change to tool 0 - same

Changing M420 Z5 will reduce the "area affected" to 0 to 5mm

**Configuration files:** [Config.zip](https://github.com/MarlinFirmware/Marlin/files/1373584/Config.zip)

**Video showing the issue:** [Video](https://1drv.ms/v/s!As5KDgCTBt6bkPhokyET7t0alf6yrg)
That video shows what it does every time I change tools (T1 -> T0 / T0 -> T1)

**Leveling Debug Info when Z is less than 10:**

```>>>G28 Z
SENDING:G28 Z
>>> gcode_G28
Machine Type: Cartesian
Probe: BLTOUCH
Probe Offset X:22 Y:80 Z:-2.20 (Right-Back & Below Nozzle)
Auto Bed Leveling: UBL (enabled)
UBL Adjustment Z0.00
echo:Active Extruder: 0
  current_position=(100.00, 0.00, 10.00) : setup_for_endstop_or_probe_move
> endstops.enable(true)
>>> homeaxis(Z)
  current_position=(100.00, 0.00, 10.00) : set_probe_deployed
deploy: 1
do_probe_raise(8.00)
>>> do_blocking_move_to(100.00, 0.00, 10.20)
<<< do_blocking_move_to
>>> do_blocking_move_to(100.00, 0.00, 10.20)
<<< do_blocking_move_to
Home 1 Fast:
>>> do_homing_move(Z, -450.00, 0.00)
set_bltouch_deployed(1)
  current_position=(100.00, 0.00, 0.00) : sync_plan_position
echo:busy: processing
set_bltouch_deployed(0)
<<< do_homing_move(Z)
Move Away:
>>> do_homing_move(Z, 5.00, 0.00)
  current_position=(100.00, 0.00, 0.00) : sync_plan_position
echo:busy: processing
<<< do_homing_move(Z)
Home 2 Slow:
>>> do_homing_move(Z, -10.00, 1.50)
set_bltouch_deployed(1)
[... 354 lines cut ...]
>>>T0
SENDING:T0
>>> gcode_T(0)
  current_position=(100.00, 0.00, 10.00) : BEFORE
Offset Tool XY by { -43.60, 0.00 }
For X axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = -22.00
 soft_endstop_max = 250.00
For Y axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = -5.00
 soft_endstop_max = 295.00
  current_position=(56.40, 0.00, 10.00) : Sync After Toolchange
  current_position=(56.40, 0.00, 10.00) : sync_plan_position
  destination=(100.00, 0.00, 10.00) : Move back
echo:Active Extruder: 0
  current_position=(100.00, 0.00, 10.00) : AFTER
<<< gcode_T
```

Thanks!
Daniel.

Roxy-3D, 2017-10-11:
It's on the list to get resolved...

thinkyhead, 2017-10-13:
I'll have to look very closely at this one! @dulfe Can you also check to see if the same thing happens with Automatic Bed Leveling (BILINEAR)? It uses essentially the same maths for adjusting Z in the fade zone. Tool Change is probably one of those things where fade has weird implications.

Roxy-3D, 2017-10-13:
> Tool Change is probably one of those things where fade has weird implications.

I don't know this...  But my suspicion is there might be a mis-match between the moving in and out of a corrected mode.

thinkyhead, 2017-10-13:
The log doesn't show anything unusual, but that in itself is helpful. I note from the video that the move up occurs before the move sideways on the tool change. That helps narrow it down. I'll have a better idea what's going on after a bit more investigation.

thinkyhead, 2017-10-13:
One more thing to test, @dulfe — Try `T1 S1` and `T0 S1` with a small X movement afterward to see if there's any change in Z due to these tool changes. The `S1` suppresses the movement of the new tool to the current position. This will tell us whether the Z move is due to a movement performed during the tool change, or whether it is due to Z being left at a bad value following the tool change.

thinkyhead, 2017-10-13:
Finally…… Be sure to test version `1.1.6` or the `bugfix-1.1.x` branch on the narrow possibility that we fixed this issue along the way.

dulfe, 2017-10-13:
I will check all that out in few hours and report... thank you!

oldmcg, 2017-10-13:
Last time I checked, UBL for cartesian does not call apply_leveling/unapply_leveling because UBL for cartesian does not define PLANNER_LEVELING in conditionals_post.h.  

I had a change for this pending before the 32-bit freeze but did not want to change UBL cartesian because I couldn't test it.  The same change would likely make G2/G3 arc work for UBL cartesian which I suspect is broken without PLANNER_LEVELING.

@dulfe, if you #define M114_DETAIL in configuration.h, you will get a lot more information from M114 command about current position with respect to internal leveling layers.  You can run M114 command before and after repeated tool change and see if something in Z axis is collecting error.

dulfe, 2017-10-13:
@thinkyhead I tested `T1 S1` and `T0 S1` and they did not moved `Z`, as soon as I moved `X` a little bit, the `Z` axis changed.

Executing `M420 S1 Z0` and moving `Z` to 60....`T1` and `T0` still raising the Z axis.

Roxy-3D, 2017-10-13:
> EDIT: Forgot to mention, I think with PLANNER_LEVELING #defined for UBL cartesian, inside ubl_motion.cpp where cartesian UBL calls buffer_line, it should be changed to _buffer_line so it bypasses the planner leveling since UBL cartesian line_to has already applied the leveling.

Yes.   I think what you are saying is correct.   But this issue may be rectified with https://github.com/MarlinFirmware/Marlin/pull/7966

dulfe, 2017-10-13:
@thinkyhead I just tested Bilenear and it does not have that problem.

dulfe, 2017-10-13:
@oldmcg If I add `#define M114_DETAIL` to `Configuration.h` I got compilation errors:

```
Marlin_main.cpp: In function void report_current_position_detail()
 
Marlin_main.cpp: 8092:13: error: 'class Planner' has no member named 'apply_leveling
   planner.apply_leveling(leveled)
 
Marlin_main.cpp: 8097:13: error: 'class Planner' has no member named 'unapply_leveling
   planner.unapply_leveling(unleveled)
Error compiling project sources
```

oldmcg, 2017-10-13:
@dulfe, sorry, guess the code has changed since I added M114_DETAIL.

dulfe, 2017-10-13:
@thinkyhead I just tested version `1.1.6` and branch `bugfix-1.1.x` and both have the same issue.

thinkyhead, 2017-10-13:
Thanks, @dulfe — All that info will help. I do have some changes pending that localize the Z fade code a bit more, but as far as I can tell there won't be any change in behavior. Still, as soon as I merge those changes into the `bugfix-1.1.x` branch I'll let you know so you can give the new code a try.

thinkyhead, 2017-10-14:
Alright, so… I guess you're the canary in the coal mine on this patch! 🐦 Again, I don't know if it will have any effect on your issue, but there were enough changes that it potentially could. It might be worse! Anyway, please try the current `bugfix-1.1.x` branch when you have a chance and let us know how it works w/r/t this issue. Thanks!

dulfe, 2017-10-14:
I will put on my helmet and try it out tonight (the wife has 'plans' for today)., thanks!

chris-bo, 2017-10-14:
I have the same behavior in bugfix-2.0.x.
Those are probably the related/the same #7604 and #7469

dulfe, 2017-10-14:
@thinkyhead no dice... I tested bugfixc-1.1.x published today @ 18:13 (0976d222) and it has the same issue.

Roxy-3D, 2017-10-15:
One thing that I'm convinced would 'fix' the problem is totally remove all Z-Height correction from the tool change command.   The next destination[] (with the new tool enabled) will end up at the 'right' height.

But I think we are trying to always keep the nozzle 'current'.   I'm not sure that is that important because the various nozzle's should always be at very similar heights.   But if we can't find the 'correct' fix...  We can just switch nozzles...  And at the end of the first move the nozzle will be at the right 'corrected' height.

thinkyhead, 2017-10-27:
> @thinkyhead no dice... I tested bugfix-1.1.x published today @ 18:13 (0976d22) and it has the same issue.

Hmm, but did you test it with the patch here?

thinkyhead, 2017-10-27:
> The next destination[] (with the new tool enabled) will end up at the 'right' height.

Mmm, not if positions are out of sync. It's possible there may be issues in Z correction based on enabled leveling plus tool offsets. It can't be dropped, as the tool change does various things which differ for dual-x versus dual-nozzle versus single-nozzle. Meticulous testing and debugging will yield best results.

My next guess is that the `unapply_leveling` and `apply_leveling` methods don't have perfect parity. But I'll be methodical in approaching this…

Roxy-3D, 2017-10-27:
> My next guess is that the unapply_leveling and apply_leveling methods don't have perfect parity. 

That is my guess.   

>> The next destination[] (with the new tool enabled) will end up at the 'right' height.
> Mmm, not if positions are out of sync, and not all tool-changes do a move. Some just update the position to where the other nozzle is.

If NO correction and NO move is done when the tools change...   The new tool will end up at the right height.   The calculated correction is applied to the current tool's destination position.

putertdi, 2018-02-10:
Is this still an open issue?  I've read through the history here but I can't tell if this has been fixed or not.

(note: I'm not very familiar with git hub bug tracking so if I'm missing something obvious I apologize).

dulfe, 2018-02-14:
@putertdi As you mentioned, it seems resolved in 1.1.8

closed 2018-02-14 by dulfe