MarlinFirmware/Marlin #7930: [FR] File printing display in sub-menu

JerryDurand, 2017-10-10:
Occasionally when selecting a file from the SD card, when I click the button it selects an adjacent file.  I know the Information Display shows the filename for a short time until Bed Heating pops up, but I may or may not notice that.

It would be nice if there was some way to check the file name, maybe when clicking to the Main Menu the file name is shown as one of the entries (but you can't click on it since that is meaningless).

thinkyhead, 2018-02-09:
Ultimately we would like to improve the status message so that messages like "Bed Heating" and "Heating Done" will expire and restore the name of the file being printed, or alternate with the name of the file being printed. Particularly, the "Heating Done" message doesn't need to linger.

sjasonsmith, 2020-11-11:
This is 2.5 years old and I do not believe it is still relevant.
When I print from SD I am asked to confirm the file prior to beginning a print. I am using a RepRapDiscount Full Graphics display.

I'm closing this out, please reply if you believe this was in error.

closed 2020-11-11 by sjasonsmith