MarlinFirmware/Marlin #7931: bf-2.0.x - LCD compile error

Allted, 2017-10-10:
### Compile Error
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
```
avr-gcc: error: CreateProcess: No such file or directory

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

```

------------------------------------------------------------
### Bug Report
- Description: When selecting REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER in configuration.h Arduino IDE won't compile no matter what board I select.
- Expected behaviour: Compile
- Actual behaviour: Error
- Steps to reproduce: un-comment #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
  - Verify

No other changes made to the config files.
------------------------------------------------------------

AnHardt, 2017-10-10:
u8glib added?

Allted, 2017-10-10:
Yes. It worked previously but I lost track as some of the pushes caused file errors from some of the devs using platform.io.

fiveangle, 2017-10-11:
Surely dupe of of #7707 (problem with Arduino IDE)

Look in here for updated arduino-builder.exe binary to resolve: https://github.com/arduino/arduino-builder/pull/249

Allted, 2017-10-11:
I just tried it, no different. 
Side note-that fix shows up as (maybe it is) a virus in windows 10, once allowing it verified a compile without the LCD and the fail with. I also tried another screen with the same lib (REPRAPWORLD_GRAPHICAL_LCD) ...same error, other lcds (#define REPRAP_DISCOUNT_SMART_CONTROLLER) are fine.

fiveangle, 2017-10-11:
You'll see my report of the same on the Arduino-builder tracker I liknked you to above  They had a different debug version posted earlier that did work for me but haven't been able to re-test it (need to hot up a VM since it too has the AV warning).  Regardless, as Anhardt mentioned: w/o the info requested on issue open (Configs, full error messages) it's tough to help :)

bobc, 2017-10-11:
I spent some time looking into this, it is the same underlying cause as #7707 but a different incarnation. There is another limit within avr-gcc. It creates processes internally, and it appears the line gets truncated (maybe related to ARG_MAX or something like that?). This then causes Windows to error on a mangled file name.

Searching on google suggests that "avr-gcc: error: CreateProcess: No such file or directory" is frequently related to line length too long, although there are several other causes. If the line length happens to be  in the region ~31,000 to 32,767 you will get this error and not "line too long".

By testing, I found that a line length of 31426 is OK, and 31748 fails, so the limit is somewhere in there. The fix applied in arduino/arduino-builder#249 needs to have a lower threshold before kicking in, e.g. 31000.

Allted, 2017-10-11:
fiveangle- No need for configs is there, I un-commented one line? And that is the entire error message from the Arduino IDE. It compiles fine in platform.io. 

bobc- So you are thinking this is somewhere between those numbers. In case it was close I moved it directly into my home drive's root folder, no luck.

I'm not trying to be a pain and I am in no rush, I am just trying to help out. 

Unfortunately I only know enough programming to be dangerous and not enough to be truly helpful. I will do what I can, If you have any suggestions let me know. I have been following along daily and keeping a Pull Request current #7648 with a new 2.0 build, but as of late these new issues are popping up and going away so I was hoping this was an easy one as well, I guess not.

fiveangle, 2017-10-11:
> fiveangle- No need for configs is there, I un-commented one line? And that is the entire error message from the Arduino IDE. It compiles fine in platform.io.

How are we supposed to know this ?  If we had a dime for everytime someone's configs had an obvious error in it, we'll...  but if you want to make it harder, so be it ;)

> If you have any suggestions let me know.

Submit the issue to Arduino, since they are the ones with the problem they need to hear about it.

Allted, 2017-10-11:
fiveangle - It is the first sentence of this bug report, I am sorry if you think I am being difficult. I figured it was actually easier to delete two "//" and confirm it than to load my config.

 Please don't waste your time on me close the bug report and I will wait it out.

I thought was here to lend a hand, sorry to have ruined your day.

closed 2017-10-11 by Allted

bobc, 2017-10-11:
@Allted there are several workarounds, none of them very nice. The simplest is probably to edit Arduino preferences.txt and add a line "build.path=c:\tmp" or similar. That means all sketches compile to the same location though which may lead to conflicts.

The best solution is probably to abandon Arduino and use platform.io. They use the @file option on gcc, which Arduino appear to have rejected in favor of a temporary hack. I have more confidence that platform.io guys know what they are doing when it comes to large scale software development.

Allted, 2017-10-11:
Thanks bobc. That does work Thank you. I tried with a few different version of the firmware and no problems so far. I can also get it to work with platform.io as well, I was just hoping to help to get it to work with the Arduino IDE . Please let me know if there is anything I can do to help but it looks like we will just have to wait for an arduino update? 

As 2.0 matures I am sure this will get ugly quick as more and more try and "update". A majority of my users have a very hard time with arduino as it is, trying to get them on to some other platform is probably not going to happen.

fiveangle, 2017-10-11:
@Allted - sorry you took offence, but my day is going fine :) .  I've obviously not made my point so sorry for that.

@bobc - I've been pushing on the DevIoT plugin author that allows very easy use of PlatformIO within Sublime Text 3 (granted, honorware), but it's been just one small bug after anther.  Once it is solid, I'll publish a video and/or doc of a quick setup of compiling Marlin under PIO.  I agree with @Allted that people already have a tough time with Arduino IDE so any alternative needs to be at least as simple.  Atom+PIO doesn't even come close.