MarlinFirmware/Marlin #7932: [FR] Pin Writes gCode

AcrimoniousMirth, 2017-10-10:
Hi all, 
       First off thanks to the awesome dev team for the work done to make this incredible software. Been using it for a long time now and love seeing the new updates roll in.
       My request is perhaps a bit selfish but I’m looking for a sort of reverse M42. I’m working on an integrated design for my university final project that I can’t say too much about but will be making opensource after graduating.
      Basically I need to easily send a command from a separate microcontroller to move to a specific X, Y and Z.
       I could use “Park position” by figures if mention this as it could be useful. What I suggest is: 
When not printing, if pin x written high, execute script.
       This in my case would park the nozzle and bed precisely where they need to be but could be used for other things such as button-settable heating presets and so on. If this feature already exists, my apologies, I haven’t seen it! But if it doesn’t I think it should be decently easy to implement (oops, my naivety is showing).
Thanks and keep being awesome!

terryb58, 2017-10-12:
Can't you just send a G27 instead of having to mess with pins?

thinkyhead, 2017-11-15:
An i2c interface would probably be more powerful and easier to implement. I've done just this sort of thing using the TWIBus class included with Marlin. See https://github.com/MarlinFirmware/Marlin/issues/4776#issuecomment-246262879 for the details.

boelle, 2019-02-19:
@AcrimoniousMirth is this one still relavant?

boelle, 2019-06-22:
wrong label

InsanityAutomation, 2019-06-22:
This may be applicable to the discussion on input to trigger macro dealing with some hardware on the Dagoma printers. Basically another use case for the same function. Id tag it 2.1.

thisiskeithb, 2022-07-09:
Added in https://github.com/MarlinFirmware/Marlin/pull/18389

closed 2022-07-09 by thisiskeithb