MarlinFirmware/Marlin #7933: [BR]Bed does not drop before G28 nor between probe points in G29 P1

blistovmhz, 2017-10-11:
---------------------------------------------------------
### Bug Report
- Description: Bed does not drop during G28, nor between mesh points during a G29 P1.
- Expected behaviour: Bed should drop before homing X or Y during a G28, and should drop before moving to each G29 mesh point before probing.
- Actual behaviour: 
1. Bed does not drop at all before homing.
2. Bed drops at G29 initiation, travel to first probe point, bed raises until z_probe is triggered, then moves directly to the next mesh point without lowering the bed first. If z_probe is still triggered at the next mesh point, it simply moves on to the next. If the z_probe is NOT already triggered, bed will raise until trigger, and then move to the next point. This results in the mesh only compensating for the lowest possible point on the bed.
- Steps to reproduce:
 Try G28? I'm not sure how to configure the bed to drop before X/Y home in the most recent versions.
For UBL, enable UBL and run a G29 P1. I am using dual Z axis optical endstops set just before the nozzle hits the bed, as well as an IR Z-probe set 1.4mm above the nozzle. M851 Z-1.4.

I have wiped EEPROM and tried several recent releases.

RAMPS 1.4
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/1373925/configs.zip)

blistovmhz, 2017-10-11:
https://www.youtube.com/watch?v=wqp9mb6opW4

Sorry. Firmware versions 1.1.5 and 1.1.6.

blistovmhz, 2017-10-11:
Nevermind. In all my screwing about, I'd forgot that at some point I was using Z_MAX as my Z_probe. At some point I had to swap pins in pins_RAMPS.h to get both zprobe and dual z endstops working, and forgot that i'd left zmax inverted, so it was always triggered.
I guess G28 ignores Z_HOMING_HEIGHT if Z_MAX is triggered, and G29 is able to initiate, drop the bed even when Z_MAX is triggered because SOFTWARE_ENDSTOPS are disabled, but DURING the G29 probe, it takes Z_MAX seriously again.

Inverting Z_MAX (it's not used) solved both problems. I would close this, but just hoping someone feels there should be an error thrown when this occurs.

fiveangle, 2017-10-16:
If SOFTWARE_ENDSTOPS were enabled and it allowed you to continue, I'd agree, but not sure how to improve, since it is certaily possible for system to be at max z when attempting to initiate a home.  How would you change it ?

closed 2018-02-09 by thinkyhead