MarlinFirmware/Marlin #7934: [BR] Dual Z axis steppers - Z2 stepper is disabled immedately after move.

blistovmhz, 2017-10-11:
- Description: Z2 stepper is disabled immediately after a Z move, resulting in the bed wandering off axis until the next Z move is commanded.
- Expected behaviour: Z2 should remain on like Z
- Actual behaviour: Z2 is disabled. EG: G28, and the second it completes, I try to spin both Z axis screws. Z1 holds. Z2 is floppy.
- Steps to reproduce:
Enable Z_DUAL_ENDSTOPS
G28 or any other Z move. You'll find the Z2 stepper is free while Z1 is not.
In my setup, Z2 is connected to E1 on my RAMPS 1.4 board. This used to work correctly with a much older version of Marlin.
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/1373946/configs.zip)

Marlin 1.1.5 and 1.1.6

blistovmhz, 2017-10-11:
This is related to #7933 (resolved). I'm not sure why.
Z_MAX was mistakenly triggered at all times (it was not in use, and no sensor is using it). For whatever reason, inverting Z_MAX so that it is normally open, causes both Z steppers to remain enabled. As soon as Z_MAX is triggered, Z2 is disabled, regardless of configured timeouts. Not sure if bug or feature?

fiveangle, 2017-10-16:
Curious if setting this false keeps 2nd Z stepper active all the time, or still disables immediately after move ?

`#define DISABLE_INACTIVE_Z true  // set to false if the nozzle will fall down on your printed part when print has finished.`

thinkyhead, 2018-02-09:
Presuming solved.

closed 2018-02-09 by thinkyhead