MarlinFirmware/Marlin #7938: Compile error  USE_CONTROLLER_FAN

android444, 2017-10-11:
### Compile Error
When I compile with `USE_CONTROLLER_FAN` I get an error:
```
In file included from k:\Temp\arduino_build_136131\sketch\src/inc/MarlinConfig.h:39:0,

                 from H:\drukarka 3d\Github\Marlin\Marlin\Marlin.ino:31:

k:\Temp\arduino_build_136131\sketch\src/inc/SanityCheck.h:893:6: error: #error "USE_CONTROLLER_FAN requires a CONTROLLER_FAN_PIN. Define in Configuration_adv.h."

     #error "USE_CONTROLLER_FAN requires a CONTROLLER_FAN_PIN. Define in Configuration_adv.h."

      ^

exit status 1
Błąd kompilacji dla płytki Arduino/Genuino Mega or Mega 2560.
```

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/1375764/Configuration_adv.zip)

oakdesign, 2017-10-11:
which board are you using and where have you wired your controller fan to?

android444, 2017-10-11:
Default configuration file 'configuration.h'
Changes only in 'configuration_adv.h'

oakdesign, 2017-10-11:
Default for which board? If for RAMPS in EFB Extruder Fan Bed Setup the FAN_PIN1 is set to -1 so you need the Signal pin number you have wired your Controller fan to

android444, 2017-10-11:
This is badly done.
Pin should be selected in xxx_pins.h

Radiu, 2017-10-11:
sorry to jump in but im having same kind of problems it seems to me that the FAN_Pin sections is not well documented and share same name on multiple sections
for what i see the  USE_CONTROLLER_FAN  is for the fan turn on automatically whenever any stepper is enabled

the part from exruder fans i never got  to work out this section

#define E0_AUTO_FAN_PIN -1
#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED   255  // == full speed

my extruder is set on the E0 slot and im using ramps motherboard EFB, when i try set a pin  #define E0_AUTO_FAN_PIN 9 always get error, by default it should work sense fan is conected on D9 of ramps
maybe im missing something no idea, but i never got my fan to start working when goes above AUTO_FAN_TEMPERATURE  50, only got to work by using the comand M106 S50
its a litle confusing, dont get it why doesnt work sense is already set on pin.h 

#if ENABLED(IS_RAMPS_EFB)                      // Hotend, Fan, Bed
  #define FAN_PIN        RAMPS_D9_PIN
  #define HEATER_BED_PIN RAMPS_D8_PIN

AnHardt, 2017-10-11:
In Marlin we have 3 kinds of fans
* Part cooling fan. For most the most useful one. Needs to be PWMd to be adjustable. The one mend in 'EFB'.
* Extruder fan. To cool the cold end of an hotend. Makes most sense when a hotend is hot. Could be permanently connected to 12V id ware and noise is no problem.
* Controller fan. Fan for cooling the mainboard - especially the stepper drives. If the steppers are not enabled they probably no not need cooling.

oakdesign, 2017-10-11:
Ok this my working setup for 
E0_AUTO_FAN_PIN and 
USE_CONTROLLER_FAN
I have both fans wired through a IRF520 Mini Mosfet. From the every IRF520 I have s single Signal wire to one of the Signal Pins on the Servo Pins D4 D5 D6 
so for 
#define E0_AUTO_FAN_PIN 4
#define CONTROLLER_FAN_PIN 5 
and D6 I'm using for LED case light
#define CASE_LIGHT_PIN 6

On the RAMPS Pins D2-D11 and D44-46 are PWM cabable

closed 2017-11-15 by thinkyhead