MarlinFirmware/Marlin #7939: ultratronics compile error

silversat, 2017-10-11:
### Compile Error
When I compile for ULTRATRONICS board with `REPRAPWORLD_GRAPHICAL_LCD` I get an error:
```
In file included from sketch\src\lcd\ultralcd_impl_DOGM.h:50:0,

                 from sketch\src\lcd\ultralcd.cpp:55:

sketch\src\lcd\dogm/ultralcd_st7920_u8glib_rrd.h: In function 'uint8_t u8g_dev_rrd_st7920_128x64_fn(u8g_t*, u8g_dev_t*, uint8_t, void*)':

sketch\src\lcd\dogm/ultralcd_st7920_u8glib_rrd.h:156:22: error: 'ST7920_SET_CMD' was not declared in this scope

       ST7920_SET_CMD();

                      ^

sketch\src\lcd\dogm/ultralcd_st7920_u8glib_rrd.h:157:29: error: 'ST7920_WRITE_BYTE' was not declared in this scope

       ST7920_WRITE_BYTE(0x08);       //display off, cursor+blink off

                             ^

sketch\src\lcd\dogm/ultralcd_st7920_u8glib_rrd.h:164:24: error: 'ST7920_SET_DAT' was not declared in this scope

         ST7920_SET_DAT();

                        ^

sketch\src\lcd\dogm/ultralcd_st7920_u8glib_rrd.h:193:24: error: 'ST7920_SET_CMD' was not declared in this scope

         ST7920_SET_CMD();

                        ^

sketch\src\lcd\dogm/ultralcd_st7920_u8glib_rrd.h:195:37: error: 'ST7920_WRITE_BYTE' was not declared in this scope

           ST7920_WRITE_BYTE(0x80 | y);       //y

                                     ^

sketch\src\lcd\dogm/ultralcd_st7920_u8glib_rrd.h:199:44: error: 'ST7920_WRITE_BYTE' was not declared in this scope

           ST7920_WRITE_BYTE(0x80 | (y - 32)); //y

                                            ^

sketch\src\lcd\dogm/ultralcd_st7920_u8glib_rrd.h:202:24: error: 'ST7920_SET_DAT' was not declared in this scope

         ST7920_SET_DAT();

                        ^

sketch\src\lcd\dogm/ultralcd_st7920_u8glib_rrd.h:203:54: error: 'ST7920_WRITE_BYTES' was not declared in this scope

         ST7920_WRITE_BYTES(ptr, (LCD_PIXEL_WIDTH) / 8); //ptr is incremented inside of macro

                                                      ^

'

sketch\src\lcd\dogm/ultralcd_st7920_u8glib_rrd.h:224: confused by earlier errors, bailing out

exit status 1
Errore durante la compilazione per la scheda Arduino Due (Programming Port).

```

psavva, 2017-10-12:
Please post your configuration.h and configuration_adv.h files

silversat, 2017-10-12:
sorry... the issue involved another forked fw.

closed 2017-10-12 by silversat