MarlinFirmware/Marlin #7942: Z-Axis too fast and Z-coordinates wrong in 1.1.6 and bugfix_1.1.x

mahowi, 2017-10-11:
### Bug Report
- **Description:** I have used Marlin 1.1.5 without problems on my Renkforce RF100. But with 1.1.6 or bugfix_1.1.x I get problems with Z-axis control.
- **Expected behaviour:** speed and control of the Z-axis should be the same as in 1.1.5
- **Actual behaviour:** Z-axis is moving way too fast, Z-coordinates are not correct. Homing works although the axis moves too fast (2-3x faster than in 1.1.5). But if I move to e.g. Z=20, the bed really moves to about 100! Changes for my printer in `Configuration.h` and `Configuration_adv.h` are equal to 1.1.5.
- **Steps to reproduce:**
  - adapt changes in `Configuration.h` and `Configuration_adv.h` from 1.1.5 to 1.1.6 or bugfix_1.1.x
  - compile firmware and transfer to printer
  - home and move Z-axis

[`Configuration.h`, `Configuration_adv.h` zipped](https://github.com/MarlinFirmware/Marlin/files/1376672/config.zip)

fiveangle, 2017-10-11:
After you upgraded, did you perform a M502, M500, M501 and confirm your feedrates are same as set in your config ?

If yes, you confirm you then revert to 1.1.5, run M502, M500, M501 and it shows the same feedrates as under 1.1.6 but Z axis moves normally ?

 ```
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 78.82,78.82,800,106 }

/**
 * Default Max Feed Rate (mm/s)
 * Override with M203
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_MAX_FEEDRATE          { 300, 300, 20, 25 }
```

mahowi, 2017-10-11:
No, I didn't perform these commands. Usually I only print from SD-card. I will connect the printer to my laptop and try to perform the M-codes.

fiveangle, 2017-10-12:
I tried to look through the diffs and am at a lost as to how a Z-only bug like this has crept in somewhere.  And you are running a standard RAMPS config so the microstepping on your Z axis is not software-configuratble, correct ? (assume yes since that's what your configs show).

My only suggestion would be to start bisecting the commits between 1.1.5 and 1.1.6 to narrow down what commit it was and it should then be obvious.  If you know how to use git, you can use the [bisect feature](http://effectif.com/git/simple-approach-to-finding-bugs-with-git-bisect), and if not, you can just grab somewhere in the [middle of this diff list](https://github.com/fiveangle/Marlin/compare/bf1.1.5...fiveangle:bf1.1.6), click "Browse Files..." and then "Download" to download the zip of the source tree at that commit and test it.  If the issue is gone, grab somehwere in the middle of the newer half of commits from the last one and do the same.  If the issue is still there, instead, grab somewhere in the older half of commits from the last one.  Wash, Rinse, Repeat. 

There are 93 commits so should be able to narrow down the exact commit in no more than log<sub>2</sub> 93, or 7 iterations, but if you can narrow it down to even a dozen commits (vs 93), it might be easy to spot.

thinkyhead, 2017-10-12:
To help narrow dow the issue, a quick question: Does this problem show up if you move X and Z together, or Y and Z together, or even E and Z together in single moves? For example, what kind of changes in speed do you see if you do each of these commands in turn…?
```gcode
G91
G1 F9999
G1 X10 Z20
G1 X10 Z10
G1 X10 Z5
G1 X10 Z2
G1 X10 Z1
G90
```
And also, when you move Z by itself, does it move the same speed and distance _going up_ as it does _going down_?
```gcode
G91
G1 Z10 F999 ; Move up at top speed
G1 Z-10 ; Down at top speed
G90
```

mahowi, 2017-10-12:
@thinkyhead Z moves the same speed up and down, also the distance is the same. The problem shows up every time Z axis is moved, also when moved together with other axis.

@fiveangle I will try to find out which commit changed the behaviour.

fiveangle, 2017-10-12:
> Z-axis is moving way too fast, Z-coordinates are not correct. Homing works although the axis moves too fast (2-3x faster than in 1.1.5). But if I move to e.g. Z=20, the bed really moves to about 100!

> Z moves the same speed up and down, also the distance is the same. The problem shows up every time Z axis is moved

Not sure how to reconcile these statements.

mahowi, 2017-10-12:
I mean, Z axis is moving too fast, but it doesn't depend on direction (up or down) or if it moves together with another axis.
And the distance is the same although it's wrong. So if I try to move Z to -20 it goes to -80, if I want to move to -10 it goes to -40.

mahowi, 2017-10-12:
Now it's getting even more strange. :confused:
I recompiled and flashed 1.1.6 again without changes in configuration and it works as expected. The three times I flashed it before the printer showed the above behaviour. I also tried Marlin from bugfix_1.1.x branch some days ago with exactly the same problems.

I will try some other versions now to find out what leads to this behaviour.

mahowi, 2017-10-13:
As I found out with the help of another forum, it is a problem with the printer not Marlin.

When flashing the firmware with EEPROM enabled it doesn't save the correct parameters in EEPROM. So it works now as expected because I have written the parameters with `M500`. Seems to be a problem with all Renkforce printers.

closed 2017-10-13 by mahowi

fiveangle, 2017-10-13:
_[placing here for posterity as the answer above is very misleading]_

@mahowi - this is not a "problem" with the printer, but how computers work.  There is no complex EEPROM content validation code included in Marlin as it would be a large resource drain - both human and computing - to manage on these modest MCUs, without much of a return.

As we [mentioned above](https://github.com/MarlinFirmware/Marlin/issues/7942#issuecomment-335931918), and you [claim you tested](https://github.com/mahowi), you must issue `M502; M500` upon major upgrade from very old Marlin in order to "reset" the EEPROM contents so they are consistent with what the running version of Marlin expects, else you will get unexpected results.  There's basic versioning included in Marlin now, but even so, when an upgrade is acting weirdly, the first action is to "M502; M500" to ensure it isn't an EEPROM issue.

I suspect when you "tested" you didn't actually check the running speed after that (only reported the M501 output above, which was correct), and so when you "[upgraded again](https://github.com/mahowi)" later, the EEPROM had likely already been normalized.

In short this is working exactly as designed.  Glad you figured it out and it wasn't a bug ! :smile:

thinkyhead, 2017-10-14:
Sorry for any confusion. Recent versions of Marlin should notice when the EEPROM contents don't match the current version and simply reset to defaults. But once in a while you get weird anomalies like this. If you see this again, please let us know!

fiveangle, 2017-10-14:
"Recent versions of Marlin should notice when the EEPROM contents don't match the current version and simply reset to defaults."

It didn't in this case, but I suspect like due to an intermediary upgrade along the way that made the EEPROM content appear "compatible" to current Marlin release, but in fact had latent issues.  If true, there's really know what to cover every case like that without autocratically resetting EEPROM every time (which throws the baby out with the bathwater :)

JerryDurand, 2017-10-14:
What I do with eeproms is have a signature in them.  Simplest is a byte 
or two for revision.  Every time I change the eeprom structure in my 
code, I ++ the revision.  So the first thing the code does is check the 
revision, if wrong, init the EEPROM to defaults.

I've also used __DATE2__ and __TIME__ as the signature when I have lots 
of room, just have to make sure they're in a header that will always be 
touched.  I use main.h to keep the list of all changes so that's always 
modified.

#define EEPROM_Signature  __DATE2__ __TIME__

fiveangle, 2017-10-14:
Yes, Marlin has signatures now  But none of that helps if there was a bug along the way on the current revision you have or if someone downgrades to an earlier code w/o checks, does some eeprom manipulation, then upgrades back :)

mahowi, 2017-10-16:
Thanks for the explanations. The original firmware of the printer was based on version "1.0.3 dev". When I upgraded first to 1.1.3 then 1.1.5 I didn't have any problems. So I didn't know I have to reset the EEPROM. 

Perhaps you could mention this in the "Getting Started" part of the homepage.