MarlinFirmware/Marlin #7950: SD init fail on Marlin 1.1.5

Qdrfglio, 2017-10-12:
Hi I'm new to the GitHub and I hope that my question is not already covered somewhere but I really need some help.

So I build Prusa i3 reworked 1,5 some time ago and from the very beggining sd card didn't work. I thought that the problem was in RepRap discount Smart Controller so I bought new RepRap Full Graphic Small Controller. Unfortunately my Sd card still didn't work. LCD shows that card is inserted or removed but in menu I see "no card" repetier shows "echo: SD init fail" I decided then to update marlin to 1.1.5 rewrite config.h to my printer and still same output. I have already tried shorter cables for display I have sd support in config.h and tried half speed quarter speed and eight speed unfortunately nothing work. Sd card is in fat32 8gb. After M21 comment in repetier there is SD init fail answer from printer.

I hope someone could help me with that.

thinkyhead, 2017-10-12:
Before proceeding, try doing a "low level" format on the SD card. Marlin sometimes has issues with cards that take a longer time to respond. If that doesn't help, there is a minor hack I've produced that can work around a very slow SD card, so I can point you towards that. In my case, the hack was needed to deal with a mis-wired LCD that had the SDDETECT pin wired to the WRITEABLE connector on the SD unit. If the slider on the SD card was in the PROTECT position, the SD init would fail.

Anyway, test teh low level format first and see if it helps…

Qdrfglio, 2017-10-12:
Thank you! Unfortunately low level format didn't help. The card is new and it's 4th class card so I don't think speed could be a problem. I also tried faster 16gb card from my dslr camera with same result. I never locked the slider neither on the sd cards. What could I try now?

Qdrfglio, 2017-10-13:
In the meanwhile I search for other problems these diplays have so I checked if I have 5v. Everythings fine there. I also tried older versions of SD library with no results. How could I check pin assigments for SD? I search throught pins_RAMPS.h but there is no pin assigments for Full Graphic Smart Controller so I'm not sure if in that case Marlin use default one for that board or if there is other place I could look for them?

closed 2018-02-06 by thinkyhead