MarlinFirmware/Marlin #7954: Pin naming scheme for LPC

alexxy, 2017-10-12:
Lets discuss pin naming scheme here =)

My proposal is to set naming convention like this

P1.2 -> D102 -> 102
P2.48 -> D248 -> 248 

and so on. So in this case we can keep simple integer numbers for pins and have their names in sync with arm ones =)

Roxy-3D, 2017-10-12:
Welcome to the club...    https://github.com/MarlinFirmware/Marlin/issues/7076#issuecomment-334877071

bobc, 2017-10-12:
Currently the pin number in Marlin is defined as int8_t, so the suggested sparse numbering "PNN" allows only pins up to P1.27, about half of the IO pins on LPC176x. OTOH if we allocate pins contiguously from D0, we could easily accommodate the 70 pins of LPC176x without code change outside pinmap_xxx.h

However, 127 pins may be restrictive for some of the high end 32 bit chips, e.g. STM32F429 has up to 168 IO pins. If we extend the size to int16_t across the board, that might have a significant (negative) impact on AVR. 

The pin type could be a HAL type, e.g. HAL_PIN_TYPE, which could be defined as 8 or 16 bit for AVR/32 bit respectively. That then allows way more pins than we will ever need (don't quote me!) even using a sparse naming. It might be efficient code wise to use 7 bits for the port, and 8 bits for the bit number, although that is less friendly for the user.

I'd suggest anyone who is sufficiently motivated to create a PR, and test it on at least AVR and Re-ARM, and evaluate the impact in terms of CPU cycles and memory usage. Then we can decide if it is something that is worthwhile to do.

It seems unlikely any printer currently needs more than 127 IO pins, so it doesn't seem like an issue that needs solving immediately. We can continue to use the Arduino style soft pin numbering which has worked for many years and no one has complained about.

Roxy-3D, 2017-10-12:
> It might be efficient code wise to use 7 bits for the port, and 8 bits for the bit number, although that is less friendly for the user.

One problem with the suggested encoding (in https://github.com/MarlinFirmware/Marlin/issues/7076#issuecomment-334877071) is every access would require a divide by 100 to get the port and a modulo operation to get the pin number.   That is very inefficient.    Using a set number of bits for the port and a set number of bits for the pin numbers as you suggest would allow for a fast mask operation.   I would suggest 8 bits and 8 bits just because the mask operation could return the value and store it with no shifts into an 8 bit entity.

p3p, 2017-10-12:
> We can continue to use the Arduino style soft pin numbering which has worked for many years and no one has complained about.

I agree, I'm not sure why it is important as long as we have clear documentation, I will change the pinmapping to be sequential and generally clean up removing reference to Re-ARM, but as it stands I don't think I will be working on changing the Marlin pin naming convention.

closed 2017-11-15 by thinkyhead