MarlinFirmware/Marlin #7957: Case Light on-off bug

mcmatrix, 2017-10-12:
### Bug Report
- Description: Marlin release 1.1.6. When case light pin is hardware pwm then turning case light off (via menu -> off or M355 S0) wont work.
- Expected behaviour: Case light should turn off
- Actual behaviour: It wont turn off
- Steps to reproduce:
  - Enable case lighting on pin 7 (MKS BASE 1.4)
  - Set brightness to 127
  - Via menu turn light on 
  - Try to turn light off without changing brightness

```cpp
#define MOSFET_D_PIN 7

#define CASE_LIGHT_ENABLE
#if ENABLED(CASE_LIGHT_ENABLE)
  #define CASE_LIGHT_PIN MOSFET_D_PIN         // Override the default pin if needed
  #define INVERT_CASE_LIGHT false             // Set true if Case Light is ON when pin is LOW
  #define CASE_LIGHT_DEFAULT_ON false         // Set default power-up state on
  #define CASE_LIGHT_DEFAULT_BRIGHTNESS 127   // Set default power-up brightness (0-255, requires PWM pin)
  #define MENU_ITEM_CASE_LIGHT                // Add a Case Light option to the LCD main menu
#endif
```

## My workaround:
Original Marlin_main.cpp
```cpp
#if HAS_CASE_LIGHT
  #ifndef INVERT_CASE_LIGHT
    #define INVERT_CASE_LIGHT false
  #endif
  uint8_t case_light_brightness;  // LCD routine wants INT
  bool case_light_on;

  void update_case_light() {
    pinMode(CASE_LIGHT_PIN, OUTPUT); // digitalWrite doesn't set the port mode
    if (case_light_on) {
      if (USEABLE_HARDWARE_PWM(CASE_LIGHT_PIN)) {
        analogWrite(CASE_LIGHT_PIN, INVERT_CASE_LIGHT ? 255 - case_light_brightness : case_light_brightness);
      }
      else WRITE(CASE_LIGHT_PIN, INVERT_CASE_LIGHT ? LOW : HIGH);
    }
    else WRITE(CASE_LIGHT_PIN, INVERT_CASE_LIGHT ? HIGH : LOW);
  }
#endif // HAS_CASE_LIGHT
```

With workaround
```cpp
#if HAS_CASE_LIGHT
  #ifndef INVERT_CASE_LIGHT
    #define INVERT_CASE_LIGHT false
  #endif
  uint8_t case_light_brightness;  // LCD routine wants INT
  bool case_light_on;

  void update_case_light() {
    pinMode(CASE_LIGHT_PIN, OUTPUT); // digitalWrite doesn't set the port mode
    if (case_light_on) {
      if (USEABLE_HARDWARE_PWM(CASE_LIGHT_PIN)) {
        analogWrite(CASE_LIGHT_PIN, INVERT_CASE_LIGHT ? 255 - case_light_brightness : case_light_brightness);
      }
      else WRITE(CASE_LIGHT_PIN, INVERT_CASE_LIGHT ? LOW : HIGH);
    }
    else {
      if (USEABLE_HARDWARE_PWM(CASE_LIGHT_PIN)) {
        analogWrite(CASE_LIGHT_PIN, INVERT_CASE_LIGHT ? 255 : 0);
      }
      else WRITE(CASE_LIGHT_PIN, INVERT_CASE_LIGHT ? HIGH : LOW);
    }
  }
#endif // HAS_CASE_LIGHT
```
------------------------------------------------------------
Not sure if this is a right way to fix it but with workaround I'm able to turn it off.

mcmatrix, 2017-10-12:
digitalWrite(CASE_LIGHT_PIN, LOW) will also turn it off but not that fastio macro WRITE

jmdearras, 2017-11-02:
I came up with the same fix independently and submitted it.  Great minds think alike!  It was accepted into the bugfix 1.1.0 fork.
Merged #8205.  https://github.com/MarlinFirmware/Marlin/pull/8205

closed 2017-11-02 by mcmatrix